h2zero/NimBLE-Arduino #132: Server and client coexistance

doudar, 2020-09-21:
I'm working on porting an existing project that subscribes to services from two different BLE Clients and then rebroadcasts the two services on one BLE client. Both the NimBLE Client and the Server work well separately however if I try to run both the client and server at the same time I get a random kernel panic on the esp32. I've double checked that I'm not initializing the BLEdevice twice. 

Interestingly it doesn't crash if i start the server before the client (although not stable) but it causes a kernel panic immediately on the following line if a client previously is started and device connected.

NimBLEServer *pServer = BLEDevice::createServer(); 

I was working on this late last night. I'll try again this morning first by increasing the NimBLE stack size. 

Anyone else had luck with client and server coexistance with NimBLE on arduino ESP32 yet? 

Thanks again to h2zero! So far this library looks awesome!

elguiri, 2020-09-21:
Yep, I have a similar project and it works without problems. You'll need two clients though, one per remote server!

doudar, 2020-09-21:
Thanks! Good to know! I discovered I have some issues with my Client setup that I think are the root of the issue.

closed 2020-09-21 by doudar