h2zero/NimBLE-Arduino #134: Get and set values of types other than std::string

ntwallace, 2020-10-06:
Hi @h2zero, is there a way to get or set values of characteristics of a type other than std::string that is not immediately obvious? From a quick peek it seems like all `ESP_GATTS_READ_EVT` values end up as a string.

Ideally I would like to directly get bools, ints, and int arrays from my client. Would that require a rewrite of BLECharacteristic to check for this?

At the moment in all my server characteristic callback functions I'm just parsing strings with library functions, but it's pretty inefficient.
```
std::string data = characteristic->getValue();

std::string rString = data.substr(0,3);
std::string gString = data.substr(3,3);
std::string bString = data.substr(6,3);

uint8_t r = std::atoi(rString.c_str());
uint8_t g = std::atoi(gString.c_str());
uint8_t b = std::atoi(bString.c_str());
```

h2zero, 2020-10-06:
Yes, there is a typecast template defined so you can cast to anything you wish.

Used like: `uint8_t u8 = pChar->getValue<uint8_t>()`

ntwallace, 2020-10-07:
Wondeful, thank you

closed 2020-10-07 by ntwallace