h2zero/NimBLE-Arduino #135: What should I do if the advertising size exceeds 31 bytes ?

wakwak-koba, 2020-10-09:
If I use 128-bit service class UUIDs, the payload size will be large and there will be little room to store other items.
Now when the payload exceeds 31 bytes, it will abort as follows:
```
CRIT NimBLEAdvertising: "error setting advertisement data; rc=4, "
```
If I use uuid128, specify Appearance, the local name can only use up to 4 characters.
What should I do?

h2zero, 2020-10-09:
You'll need to move the excess data to the scan response data. Keep in mind each individual data type has 2 extra bytes added for the type and length indicators.

wakwak-koba, 2020-10-09:
In `NimBLEAdvertising::start`, it calculates the payload length and decide whether to rsp before calling ble_gap_adv_rsp_set_fields, but I feel that it is because I only consider the size of the address book and local name.

I'm trying to fix `NimBLEAdvertising::start` to take into account the length of other items, is this approach correct?

h2zero, 2020-10-09:
Using the `NimBLEAdvertising::addServiceUUID` parameter was only intended for basic uses, when the more "advanced" uses are needed the `NimBLEAdvertisementData` class should be used with the corresponding `NimBLEAdvertising::setAdvertisementData` and `NimBLEAdvertising::setScanResponseData` calls. 

If I was to write this library without basing it on the original I would not have implemented the `addserviceUUID` method (it was a pain to get working) and just had the `NimBLEAdvertisementData` class. It was left in to make things easier for transition.

wakwak-koba, 2020-10-09:
I understood the situation.
However, if a user uses uuid128, they will most likely end up in a situation where they must use the RSP.

So my suggestion is that it's a pain to find the payload for all items.
As a temporary measure, why not always advertise the device name in RSP when `m_scanResp == true` ?

It doesn't matter if `m_scanResp == true` and uuid128.

h2zero, 2020-10-09:
It should do that already. If there isn't room in the ADV packet for the UUID and the device name NimBLEAdvertising::start moves the name into the RSP data (or at least it should). If that is not working then there is a bug I have not seen. The only time that would not be the case is when using custom data. 

Can you share a code snippet to reproduce this?

wakwak-koba, 2020-10-09:
```c++
#include <NimBLEDevice.h>

void setup() {
  BLEDevice::init("12345");
  /*
   * Conditions to abort
   * OK "1234"      // RSP not required
   * NG "12345"     // have to use RSP but not trying to use it
   * NG "123456"    // have to use RSP but not trying to use it
   * NG "1234567"   // have to use RSP but not trying to use it
   * NG "12345678"  // have to use RSP but not trying to use it
   * OK "123456789" // use RSP
   */
  
  auto pServer = BLEDevice::createServer();
  auto pService = pServer->createService("550e8400-e29b-41d4-a716-446655440000"); // dummy
  pService->start();
    
  auto pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(pService->getUUID());
  pAdvertising->setAppearance(0x00);
  pAdvertising->setScanResponse(true);
  pAdvertising->start();
}

void loop() {}
```

After setting the priority to be used, how about storing it in `m_advData` as follows ?

1. flags
2. appearance
3. tx_pwr_lvl
4. BLE_UUID (s)
5. name

So what about interrupting when it exceeds 31 bytes and moving only the name to the `m_advData` when it exceeds?

h2zero, 2020-10-10:
Good catch with the appearance bytes not being calculated in the payload, thanks! 

I left a comment on your PR and I will note it here as well. If we prioritize tx_pwr_lvl over the name, we will be limiting the name length by another 3 bytes which may cause issues for those not using scan data and already near the edge of full advertisement payload.

If we were to change it in this way there should be an option to disable tx_pwr_lvl advertising.

PR #136 (2020-10-21): Fixed incomplete payload size calculation
I tried to fix #135 in my own way.
closed 2020-10-21 by h2zero