h2zero/NimBLE-Arduino #138: compiling errors in platformio

ccssw, 2020-10-28:
I installed the library like this in my `platformio.ini`
`
lib_deps =    
   h2zero/NimBLE-Arduino
`

When I compile I get:
`{
	"resource": "/home/x/.pio/libdeps/esp32_dev/NimBLE-Arduino/src/NimBLEDevice.h",
	"owner": "cpp",
	"severity": 8,
	"message": "conflicting declaration 'typedef int (* gap_event_handler)(struct ble_gap_event*, void*)'",
	"startLineNumber": 84,
	"startColumn": 65,
	"endLineNumber": 84,
	"endColumn": 65
}`

and
`{
	"resource": "/home/x/.pio/libdeps/esp32_dev/NimBLE-Arduino/src/NimBLERemoteCharacteristic.h",
	"owner": "cpp",
	"severity": 8,
	"message": "conflicting declaration 'typedef class std::function<void(NimBLERemoteCharacteristic*, unsigned char*, unsigned int, bool)> notify_callback'",
	"startLineNumber": 34,
	"startColumn": 80,
	"endLineNumber": 34,
	"endColumn": 80
}`

h2zero, 2020-10-28:
I'm not sure what would be causing this, I don't use platformio currently, but many users here do. I will see if I can find the error when I get home tonight. Hopefully someone else will have a solution for you before then though.

h2zero, 2020-10-29:
I just checked with platformio and I cannot reproduce this error.

My `platformio.ini` was the following:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = h2zero/NimBLE-Arduino
```
I compiled a basic ardruino sketch with no issue. Would you be able to provide any more details so I can attempt further to reproduce?

ccssw, 2020-10-29:
Thank you for your answer @h2zero . I tried a new basic project and it works. If the problem is in my project my `platformio.ini` is:

```
[env:noukee]
board_build.partitions = /$PROJECT_DIR/custompart.csv
platform = espressif32
board = esp32dev
framework = arduino
build_flags =
    -DCORE_DEBUG_LEVEL=4
    -DBOARD_HAS_PSRAM 
    -mfix-esp32-psram-cache-issue
lib_deps =    
   h2zero/NimBLE-Arduino
   https://github.com/tzapu/WiFiManager.git#293f705
   https://github.com/knolleary/pubsubclient.git#master
   https://github.com/bblanchon/ArduinoJson.git#6.x

```

h2zero, 2020-10-29:
I don't see any issues there, must be something in your code. If you are using those functions in your code, check the signatures where you are defining them to be sure they match the declarations.

ccssw, 2020-11-02:
Solved! The problem was I had a `#include <BLEUtils.h> `  in another part of my code.

h2zero, 2020-11-02:
Great! I'm Glad you found the issue, thanks for the update.

closed 2020-11-02 by ccssw