h2zero/NimBLE-Arduino #139: Guru Mediation Error after connect

noukee, 2020-10-30:
When I connect to device I get a "Guru Mediation Error"

```
[I][blecentral.cpp:151] onResult(): BLE Advertised Device found: Name: Nuki_14F0BFDD, Address: 54:d2:72:f0:bf:dd, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6614f0bfddc5
[I][blecentral.cpp:154] onResult(): Found Our Service
D NimBLEScan: ">> stop()"
[I][blecentral.cpp:235] scanEndedCB(): Scan Ended
D NimBLEScan: "<< stop()"
[I][blecentral.cpp:240] connectToServer(): [BLE] Created client
[I][blecentral.cpp:273] connectToServer(): New client created
D NimBLEClient: ">> connect(54:d2:72:f0:bf:dd)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=200"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
[I][blecentral.cpp:171] onConnect(): Connected
[D][main.cpp:21] event_handle(): new received event
[I][main.cpp:36] event_handle(): Service is connected
D NimBLEClient: "<< connect()"
[I][blecentral.cpp:302] connectToServer(): Connected to: 54:d2:72:f0:bf:dd
D NimBLEClient: ">> getRssi()"
[I][blecentral.cpp:303] connectToServer(): RSSI: -53
[I][blecentral.cpp:309] connectToServer(): #1
D NimBLEClient: "Got Client event "
D NimBLEClient: "Peer requesting to update connection parameters"
D NimBLEClient: "MinInterval: 12, MaxInterval: 12, Latency: 0, Timeout: 600"
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d55dd  PS      : 0x00060630  A0      : 0x800db734  A1      : 0x3ffc8770  
A2      : 0x014019bc  A3      : 0x3ffcb768  A4      : 0x3ffc885c  A5      : 0x3f4035c5  
A6      : 0x3f403314  A7      : 0x00000258  A8      : 0x800d55d5  A9      : 0x3ffc8720  
A10     : 0x3ffbec08  A11     : 0x3ffcb768  A12     : 0x3ffc88c4  A13     : 0x0000001f  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x014019cc  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d55dd:0x3ffc8770 0x400db731:0x3ffc8810 0x400db77b:0x3ffc8830 0x400dbcb5:0x3ffc8850 0x400e3693:0x3ffc88b0 0x400e35d3:0x3ffc88f0 0x400e20b1:0x3ffc8920 0x400dfed5:0x3ffc8950 0x400dfee7:0x3ffc8970 0x400e8016:0x3ffc8990 0x400d6162:0x3ffc89b0 0x4008f955:0x3ffc89d0
```

Decoding 15 results
```
0x400d55dd: NimBLEClient::handleGapEvent(ble_gap_event*, void*) at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/NimBLEClient.cpp line 676
0x400d55dd: NimBLEClient::handleGapEvent(ble_gap_event*, void*) at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/NimBLEClient.cpp line 676
0x400db731: ble_gap_call_event_cb at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 5709
0x400db77b: ble_gap_call_conn_event_cb at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 5709
0x400dbcb5: ble_gap_rx_l2cap_update_req at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 5709
0x400e3693: ble_l2cap_sig_update_req_rx at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/host/src/ble_l2cap_sig.c line 383
0x400e35d3: ble_l2cap_sig_rx at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/host/src/ble_l2cap_sig.c line 1146
0x400e20b1: ble_hs_hci_evt_acl_process at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 959
0x400dfed5: ble_hs_process_rx_data_queue at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 109
0x400dfee7: ble_hs_event_rx_data at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 109
0x400e8016: ble_npl_event_run at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c line 130
:  (inlined by) nimble_port_run at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c line 81
0x400d6162: NimBLEDevice::host_task(void*) at /Users/fjsualop/Development/PlatformIO/Nuki NimBLE/.pio/libdeps/esp-wrover-kit/NimBLE-Arduino/src/NimBLEDevice.cpp line 592 (discriminator 1)
0x4008f955: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)
```

h2zero, 2020-10-30:
Thanks for reporting, seems the error occurs in [this code block](https://github.com/h2zero/NimBLE-Arduino/blob/f48e05c8183cab356d1d074cb2f01361939d7119/src/NimBLEClient.cpp#L811).
I have not seen any issues with this before, I will look into it tonight.

noukee, 2020-10-31:
@h2zero I was looking into and the error occurs in [this line](https://github.com/h2zero/NimBLE-Arduino/blob/f48e05c8183cab356d1d074cb2f01361939d7119/src/NimBLEClient.cpp#L823)

I think that my code is similar to the example "NimBLE_Client.ino":
```
class ClientCallbacks: public NimBLEClientCallbacks //derived_class
{
    void onConnect(NimBLEClient* );
    void onDisconnect(NimBLEClient* );
    bool onConnParamsUpdateRequest(NimBLEClient*, const ble_gap_upd_params*);
};
/** Create a single global instance of the callback class to be used by all clients */
static ClientCallbacks clientCB;

void ClientCallbacks::onConnect(NimBLEClient* pClient)
{
  log_i("Connected");

  /** After connection we should change the parameters if we don't need fast response times.
  *  These settings are 150ms interval, 0 latency, 450ms timout. 
  *  Timeout should be a multiple of the interval, minimum is 100ms.
  *  I find a multiple of 3-5 * the interval works best for quick response/reconnect.
  *  Min interval: 120 * 1.25ms = 150, Max interval: 120 * 1.25ms = 150, 0 latency, 60 * 10ms = 600ms timeout 
  */
  pClient->updateConnParams(120,120,0,60);

  if (doPair) call_back(ble_event_pair_on);
  else call_back(ble_event_srv_on);
}

void ClientCallbacks::onDisconnect(NimBLEClient* pClient)
{
  log_i("onDisconnect  %s", pClient->getPeerAddress().toString().c_str());
  //log_i(" Disconnected - Starting scan");
  //NimBLEDevice::getScan()->start(scanTime, scanEndedCB);
    
  if (doPair) call_back(ble_event_pair_off);
  else call_back(ble_event_srv_off);
}

bool ClientCallbacks::onConnParamsUpdateRequest(NimBLEClient* pClient, const ble_gap_upd_params* params) 
{
  if(params->itvl_min < 24) { /** 1.25ms units */
    return false;
  } else if(params->itvl_max > 40) { /** 1.25ms units */
    return false;
  } else if(params->latency > 2) { /** Number of intervals allowed to skip */
    return false;
  } else if(params->supervision_timeout > 100) { /** 10ms units */
    return false;
  }
  return true;
}

static esp_err_t connectToServer() 
{
    log_i("[BLE] Created client");
    NimBLEClient* pClient = nullptr;

    /** Check if we have a client we should reuse first **/
    if(NimBLEDevice::getClientListSize()) 
    {
        /** Special case when we already know this device, we send false as the 
         *  second argument in connect() to prevent refreshing the service database.
         *  This saves considerable time and power.
         */
        pClient = NimBLEDevice::getClientByPeerAddress(advDevice->getAddress());
        if(pClient){
            if(!pClient->connect(advDevice, false)) {
                log_e("Reconnect failed");
                return false;
[... 71 lines cut ...]
                /** Disconnect if subscribe failed */ 
                pClient->disconnect();
                return false;
            }
        }
        else if(pChr->canIndicate()) {
            const uint8_t indicationOn[] = {0x2,0x0};
            pChr->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn,2,true);
            /** Send false as first argument to subscribe to indications instead of notifications */
            //if(!pChr->registerForNotify(notifyCB, false)) {
            if(!pChr->subscribe(false, notifyCallback)) {
                /** Disconnect if subscribe failed */ 
                pClient->disconnect();
                return false;
            }
        }    
    }
    else {
        log_e("DEAD service not found.");
    }
    
    log_i("Done with this device!");
    return true;
} // end connectToServer
```

h2zero, 2020-10-31:
Thanks for the extra info, I haven't been able to reproduce this issue yet.

It looks like your code is rejecting the requested connection parameters, if you remove the code in the `onConnParamsUpdateRequest()` callback and just `return true;` does the issue still occur?

noukee, 2020-10-31:
If a comment my code onConnParamsUpdateRequest:
```
/***
bool ClientCallbacks::onConnParamsUpdateRequest(NimBLEClient* pClient, const ble_gap_upd_params* params) 
{
  if(params->itvl_min < 24) { // 1.25ms units
    return false;
  } else if(params->itvl_max > 40) { // 1.25ms units
    return false;
  } else if(params->latency > 2) { // Number of intervals allowed to skip
    return false;
  } else if(params->supervision_timeout > 100) { // 10ms units
    return false;
  }
  return true;
}***/

It keeps failing on the same line ¿?
```

h2zero, 2020-10-31:
I'm still unable to reproduce this. I've used your code as shown and my own with a peripheral that changes the connection parameters repeatedly. 

I did look more closely at the log you posted and `LoadProhibited` error according to espressif is described:
> This CPU exception happens when application attempts to read from or write to an invalid memory location. The address which was written/read is found in EXCVADDR register in the register dump. If this address is zero, it usually means that application attempted to dereference a NULL pointer. If this address is close to zero, it usually means that application attempted to access member of a structure, but the pointer to the structure was NULL. If this address is something else (garbage value, not in 0x3fxxxxxx - 0x6xxxxxxx range), it likely means that the pointer used to access the data was either not initialized or was corrupted.

Looking at the `EXCVADDR` in your log it would seem the latter part of the description would apply. What the cause for this may be I am not sure. Is your code running in a task?, it might need to have it's stack increased if so.

noukee, 2020-10-31:
Muy code is running un a task, I fix the issue declaring NimBLERemoteCharacteristic* pChr out from connectToServer función.

Maybe the issue was  to exceed task stack, but now It's working with this change.

Thank you a lot.

h2zero, 2020-10-31:
Awesome, glad to hear you found the fix!

closed 2020-10-31 by noukee