h2zero/NimBLE-Arduino #140: I NimBLEClient: "disconnect; reason=574, " no error description

fightforlife, 2020-10-31:
I am currently trying to connect a ESP32 as Client to my bluetooth thermostats.
In around 2 out of 10 tries the code below is working. But most of the time I get the disconnect reason 574.
I could not find any description for the reason 574.
Did someone have the same error before?   

simple code
```
 //Write Value
    NimBLEClient* pClient;
    NimBLEDevice::init("");
    NimBLEDevice::setPower(ESP_PWR_LVL_P9, ESP_BLE_PWR_TYPE_CONN_HDL0);
    NimBLEDevice::setSecurityAuth(false, false, false); 
    pClient = NimBLEDevice::createClient();
    pClient->setConnectTimeout(ConnectTimeout);
    //pClient->setConnectionParams(40,200,0,5000,100,80);
    Serial.println("start connecting");
    if (pClient->connect(NimBLEAddress(Address), false)) {
      Serial.println("connection successfull");
      NimBLERemoteService* writeService = pClient->getService(NimBLEUUID(ServiceUUID));
      NimBLERemoteCharacteristic* writeCharacteristics = writeService->getCharacteristic(NimBLEUUID(CharacteristicUUID));
      writeCharacteristics->writeValue((const uint8_t *)Value, ValueLength, WriteResponse);
      pClient->disconnect();
      doc["Response"]["success"] = true;
      Serial.println("write successfull");
    }
    else {
      doc["Response"]["success"] = false;
    }
    NimBLEDevice::deleteClient(pClient);
```

debug log
```
D NimBLEDevice: ">> setPower: 7 (type: 0)"
D NimBLEDevice: "<< setPower"
D NimBLEDevice: "Setting bonding: 0, mitm: 0, sc: 0"
start connecting
D NimBLEClient: ">> connect(00:1a:22:0b:07:65)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=574, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"

```

h2zero, 2020-10-31:
Hi error code 0x574 translates to `BLE_ERR_CONN_ESTABLISHMENT` 0x23E in the nimble return codes. 

This is what it indicates:

> LL initiated a connection but the connection has failed to be established. Controller did not receive any packets from remote end.

Perhaps there is excessive radio noise or you're close to the limit of receiving range.

You could try adjusting the connection parameters to compensate.

fightforlife, 2020-11-01:
When I get a successfull connection the logs looks as follows:
```
D NimBLEDevice: ">> setPower: 7 (type: 0)"
D NimBLEDevice: "<< setPower"
D NimBLEDevice: "Setting bonding: 0, mitm: 0, sc: 0"
start connecting
D NimBLEClient: ">> connect(00:1a:22:0b:07:86)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Peer requesting to update connection parameters"
D NimBLEClient: "MinInterval: 40, MaxInterval: 200, Latency: 0, Timeout: 5000"
D NimBLEClientCallbacks: "onConnParamsUpdateRequest: default"
D NimBLEClient: "Accepted peer params"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D NimBLEClientCallbacks: "onConnect: default"
D NimBLEClient: "<< connect()"
connection successfull
...
```

I tried to set the requested connection parameters before connecting to the device:

`pClient->setConnectionParams(40,200,0,5000,100,80);`

But I get the following error. Most likely I need to convert some of these values.

```
D NimBLEDevice: "Setting bonding: 0, mitm: 0, sc: 0"
assertion "rc == 0 && "Invalid Connection parameters"" failed: file "C:\Users\Frederik\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEClient.cpp", line 311, function: void NimBLEClient::setConnectionParams(uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t)
abort() was called at PC 0x4013d36b on core 1

Backtrace: 0x4009186c:0x3ffcc230 0x40091a9d:0x3ffcc250 0x4013d36b:0x3ffcc270 0x400df2a3:0x3ffcc2a0 0x400d4eaa:0x3ffcc2c0 0x4018fcdf:0x3ffcc8b0 0x400dba0e:0x3ffcc8d0 0x400d4556:0x3ffcc910 0x400f4e6d:0x3ffccb60 0x4008dc91:0x3ffccb80

Rebooting...
ets Jun  8 2016 00:22:57
```

fightforlife, 2020-11-01:
I found out that the Timeout has to be value between 10 and 3200. It is a little bit more stable now. But still not perfect. I am trying some different positions.

h2zero, 2020-11-01:
The first 2 parameters are multiples of 0.625ms, they dictate how often a packet is sent to maintain the connection. 

The third lets the peripheral skip the amount of connection packets you specify, usually 0.

The fourth is a multiple of 10ms that dictates how long to wait before considering the connection dropped if no packet is received, I usually keep this around 100-200 (1-2 seconds).

The last 2 parameters I would suggest not specifying, they are the scan parameters used when calling `connect()` as it will scan to find the device you're connecting to, the default values work well.

tonton81, 2020-12-30:
this may or may not be revelant to you but....

on my BLE devices I am working on, you need at least ONE valid write to a characteristic, after that the peripheral will stop kicking you after it's own timeout. When I didn't do the write initially after setting up the connection, the peripheral would just drop the connection every 20-30 seconds, now it stays connected after the first write even if i stop writing to it so long as the BLE connection holds :)

h2zero, 2021-01-08:
@fightforlife have you been able to resolve this?

fightforlife, 2021-01-08:
Hi @h2zero,
I think this is just a problem with connection stability on the side of the eQ3 thermostats.
I changed my code to account for the parameters that are requested by the thermostat before conncting: `setConnectionParams(64, 320, 0, 500)` 
and the thermostat requests `D NimBLEClient: "MinInterval: 40, MaxInterval: 200, Latency: 0, Timeout: 5000"`
But this only helped marginally. I also added a loop to try multiple times, which helps.
Most of the time it works in 1 out of 5 tries.
I also noticed that the MTU is set to 23: `I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"`
But in the documentation it says `This should be larger than 23 and lower or equal to BLE_ATT_MTU_MAX = 527.` Could this be a problem?

Is there somewhere a guide how to configure the Connection Parameters to get the most stable connection?
Or maybe it would be better to not accept the recommended connection parameters from the thermostat?

Code snippet
```
    response["success"] = false;
    NimBLEClient *eq3Client;
    eq3Client = NimBLEDevice::createClient(NimBLEAddress(Address));
    eq3Client->setConnectionParams(64, 320, 0, 500);
    eq3Client->begin();
    //eq3Client->setConnectTimeout(ConnectTimeout);
    for (int i = 0; i < retryCount; i++)
    {
      if (eq3Client->connect(NimBLEAddress(Address), false))
      {
        if (eq3Client->getService(NimBLEUUID(ServiceUUID))->getCharacteristic(NimBLEUUID(CharacteristicUUIDResponse))->subscribe(true, EQ3_Notify_Callback, false))
        {
          if (eq3Client->setValue(NimBLEUUID(ServiceUUID), NimBLEUUID(CharacteristicUUIDRequest), asciiValue, true))
          {
            response["success"] = true;
            delay(1000);
            break;
          }
        }
      }
    }
    eq3Client->disconnect();
    NimBLEDevice::deleteClient(eq3Client)
```

Log ( 3 meters away)
```
2021-01-08T08:05:30+0100T: MQTT RX: gateway/eq3/status/request
2021-01-08T08:05:30+0100V: MQTT RX Payload: {"id":"5b296817.202f88","Address":"00:1a:22:0b:07:46","retryCount":5}
D NimBLEClient: ">> connect(00:1a:22:0b:07:46)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=574, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> connect(00:1a:22:0b:07:46)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=574, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> connect(00:1a:22:0b:07:46)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Peer requesting to update connection parameters"
D NimBLEClient: "MinInterval: 40, MaxInterval: 200, Latency: 0, Timeout: 5000"
D NimBLEClientCallbacks: "onConnParamsUpdateRequest: default"
D NimBLEClient: "Accepted peer params"
D NimBLEClient: "Got Client event "
[... 34 lines cut ...]
D NimBLEClient: "Got Client event "
D NimBLEClient: "Notify Recieved for handle: 1057"
D NimBLEClient: "checking service 3e135142-654f-9090-134a-a6ff5bb77046 for handle: 1057"
D NimBLEClient: "Got Notification for characteristic Characteristic: uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a, handle: 1057 0x0421, props:  0x1a
Descriptor: uuid: 0x2902, handle: 1072"
D NimBLEClient: "Invoking callback for notification on characteristic Characteristic: uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a, handle: 1057 0x0421, props:  0x1a
Descriptor: uuid: 0x2902, handle: 1072"
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: ">> disconnect()"
E NimBLEClient: "ble_gap_terminate failed: rc=2 "
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"
```

h2zero, 2021-01-08:
Hi @fightforlife thanks for getting back. 

>I changed my code to account for the parameters that are requested by the thermostat before conncting: setConnectionParams(64, 320, 0, 500)
and the thermostat requests D NimBLEClient: "MinInterval: 40, MaxInterval: 200, Latency: 0, Timeout: 5000"

Minor issue here, my fault. The documentation is incorrect, I wrote 0.625ms units for the intervals when they are 1.25ms units. Sorry about that, I must have had the scan interval settings in mind, I will be updating the docs soon.

That is 1 part of the issue, the second part is the parameters the peer is requesting in the log, well lets just say they violate the spec and would be invalid (5000 * 10ms > 32 second max allowed). So we can assume it is asking for true values here, not unit values. This is problematic because if we accept them we will be setting the parameters to those values but in units. So first, you should reject those parameters. 

Next we need to calculate what they should be for connecting:
40 / 1.25 = 32 min interval
200 / 1.25 = 160 max interval
You already had the correct timeout at 500.

Hopefully that will resolve some of those connection timeouts. Those rc=574 messages come right after the connection event which means the device wasn't responding to channel negotiations, so the connection never fully established.

The MTU message is the device reporting it's MTU, which in this case is spec minimum, there is nothing we can do to change it as the lowest value between the client and the server will be the MTU used. It's simply not capable of more, but this does not affect the connection.

gittehupe, 2021-01-26:
Hi @fightforlife ,

were you able to fully resolve this issue or to reach some kind of optimum? I am facing the same issue with my eQ3 thermostats.
Unfortunately, also the values suggested by @h2zero did not really help (I tried `pclient->updateConnParams(32, 160, 0, 500)`).

h2zero, 2021-01-26:
Hi @gittehupe, I'm curious about this as well. I believe the implementation you want is `pclient->setConnectionParams(32, 160, 0, 500);` which is called before `connect()` instead of the `updateConnParams()` method. 

Also make sure you have implemented the `onConnParamsUpdateRequest()` callback in `NimBLEClientCallbacks` such that it returns `false`, since that device sends invalid parameters we need to reject them.

Let me know if that works.

fightforlife, 2021-01-27:
Hi together, I switched from my own little programm to using Tasmota on the ESP32.
bitsimonh is currently creating pull requests to include a new BLE driver into the main tasmota branch and he also created an EQ3 module.  https://github.com/btsimonh/Tasmota/tree/EQ3_BLE_ESP32_XDRV53-2

Together with the new Nimble-Arduino release and the new esp-idf changes this is working much better than my own programm.
You still need to implement a small retry mechanism for when Tasmota is busy or it failed to connect (e.g. in Node-Red). 
For me this is a much better expierience, since Tasmota also supports reading the Xiaomi sensors over BLE.

gittehupe, 2021-01-27:
Thanks for your hint. I just gave it a short try. Unfortunately, this didn't help. Everything rather got worse, but also the test circumstances were changed. I have now a distance of maybe 5-6m (and a sofa) between the ESP32 and the two eQ3 thermostats. Formerly, there were only 30-50 cm.

Which optimal behaviour can be expected at all? Should it always connect without any issues (if the distance is not too high)?

Is what we see a typical behaviour if the distance gets (too?) high? (In that case an increase of the transmission power might possibly help.)

h2zero, 2021-01-27:
Can you post a log of the issue with debug on? It is expected to have failed connections, I see them quite often as well, sometimes it takes one attempt, sometimes multiple. It's all dependent on the timing of sending the connect and the device listening for it at the same time.
Sources of 2.4ghz noise such as wifi near to the devices will interfere with this process as well.

gittehupe, 2021-01-28:
Still, I would like to do soon further tests focussing on NimBLEClient alone...

h2zero, 2021-01-28:
Looks normal to me, I usually use a loop mechanism with a counter to retry connecting if it fails.
```
while (!pClient->connect() && --counter) {
    delay(10);
}
```

Or some other variant, other times i will just restart the scan and try to connect again when I see the device advertising so you know it's listening.

gittehupe, 2021-01-28:
So, if this is totally normal that logs like
`20:56:47.716 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
20:56:47.716 -> E NimBLEClient: "Connection failed; status=574 "`
occur from time to time I can happily proceed :-)

Do you think that tweaking with NimBLEDevice::setPower(...); could reduce the number of failed connects?

h2zero, 2021-01-28:
Yes, perfectly normal, just like a missed phone call.

Increasing the txpower might help, hard to say as we don't know the reason for the failure. Feel free to try though, certainly won't hurt anything.

closed 2022-01-19 by h2zero