h2zero/NimBLE-Arduino #142: Ble scan Wi-Fi SSL mqtt

tcpipchip, 2020-11-11:
Using the kolbahn ble scan + Wi-Fi + mqtt, works nice!
Using the kolbahn ble scan + Wi-Fi + ssl mqtt, crashs - heap memory

If I use Nimble ble scan + Wi-Fi + ssl mqtt, Will solve the heap memory problem ?

h2zero, 2020-11-11:
I can't make any promises but the answer is most likely yes. 

This library will free up 30k - 100k of ram over the bluedriod BLE library (depends how you use it).

tcpipchip, 2020-11-11:
Tomorrow I will start the conversion ! I did a test of your code and the scan response looks good and see if SSL mqtt works!

tcpipchip, 2020-11-11:
getting good progress!

tcpipchip, 2020-11-11:
looks that is working good until now
But, why shows 
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
During SCAN ?

tcpipchip, 2020-11-11:
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
.
.

tcpipchip, 2020-11-11:
And how can i get the MAC address of the BEACON/EDDYSTONE ?

h2zero, 2020-11-11:
That error occurs because the data from the advertisement was corrupt on incomplete. I've never actually seen that error before though.

Could you try changing [this line](https://github.com/h2zero/NimBLE-Arduino/blob/ba141e9da8f9c790a5ecc450e98208a2e9fffc63/src/NimBLEAdvertisedDevice.cpp#L301) to: 
```
NIMBLE_LOGE(LOG_TAG, "Gap Event Parse ERROR, rc=%d", rc);
```
That will allow me to see the error code returned and hopefully answer you better.

tcpipchip, 2020-11-11:
Hi
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR, rc=10"
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR, rc=10"
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR, rc=10"
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR, rc=10"

tcpipchip, 2020-11-11:
btw, the beacon/eddystone are coming correctly!

h2zero, 2020-11-11:
Thanks, that error code is `BLE_HS_EBADDATA`, which unfortunately only tells us that one of the data sets in the advertisement packet has an incorrect length. 

The function that sets this code is [here](https://github.com/h2zero/NimBLE-Arduino/blob/5bc570a6c2a9dd4c1e045d3a58534395ccb44fd4/src/nimble/host/src/ble_hs_adv.c#L543) there are many conditions in it that will return that code so you may need to isolate it on your end to find what data is incorrect. 
This seems to be a fault in the advertisement of device you're scanning. What is the device?

tcpipchip, 2020-11-11:
[image]

Well, i am scanning many, one of i them i developted and maybe i am implemented with Fault when mounted the Payload.

But, the most important, is that the beacon/eddystone that i am scanning looks does not have problem :+1: 

Btw, the good news is that no crash when make a SSL MQTT connection and send the first TOPIC message.

Now i will test if crash during the SCANNING

Thanks

tcpipchip, 2020-11-11:
yea, i have to wait for the SCANNING finish and then send to MQTT, otherwise  crash!

Nice, your LIB solved the problem of SSL MQTT memory :)

tcpipchip, 2020-11-11:
is your LIB (ble scan) "no blocking" ?

h2zero, 2020-11-11:
There is a non-blocking scan call. If you call `NimBLEScan::start(duration, nullptr, false);` it will not block.

>yea, i have to wait for the SCANNING finish and then send to MQTT, otherwise crash!

This is an issue that was recently brought up that I'm trying to investigate in 1technophile/OpenMQTTGateway#794

Could you share the log of your crash? This may help identify the issue.

tcpipchip, 2020-11-11:
i will do tomorrow for you!

tcpipchip, 2020-11-12:
good morning
this is the error if publish during the ble scanning (into CALL BACK)

Found EfentoSensor: 28:2c:02:40:3d:9b 6c020205090100000cdf000300010206446f00470e4b0000df52
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)

tcpipchip, 2020-11-12:
if i wait the scanning finish, works nice :)

h2zero, 2020-11-12:
Great, thanks for that! 

Seems to be the same error as reported in the other issue. 
This is also a known problem in bluedroid as reported over a year ago at arduino-esp32#3449, not sure what can be done about it.

Glad it's working for you!

chegewara, 2020-11-12:
When ble is scanning then 100% radio time is used by ble, so wifi cant be used.

tcpipchip, 2020-11-12:
Yes, i tested if stopped the ble scanning, then i use wifi!

h2zero, 2020-11-12:
> When ble is scanning then 100% radio time is used by ble, so wifi cant be used.

@chegewara do you know if this is also true if scan is set with an interval higher than the scan window. I haven't tried to do this but I would think radio would be accessible between the end of the scan window and the start of the next at the interval?

chegewara, 2020-11-12:
Yes, if you set interval higher than window then wifi works, but this needs to be done experimentally. IIRC there is one example somewhere that has values i found with wireshark on windows PC, but all depends on app requirements.

h2zero, 2020-11-12:
Thanks, I don't have a need to do this but it's nice to know for reference 😄

tcpipchip, 2020-12-04:
https://nina-b302-wifi-mqtt.blogspot.com/2020/12/utilizando-u-blox-nina-w102-como-ble.html

Working nice the Nimble

This is the blog

h2zero, 2020-12-04:
Very nice, glad it's working for you!

h2zero, 2020-12-29:
Closing this as resolved.

closed 2020-12-29 by h2zero