h2zero/NimBLE-Arduino #143: Multiple connect / enable, receive notification (Question, not an issue)

m-menon, 2020-11-13:
Is something like below possible with NimBLE-Arduino? The below code is kinda clean (and lean). I was wondering if there is an equivalent example (BLENotifier class) using NimBLE-Arduino. Thanks.

This is the source: [Multiple iTags with ESP32](https://www.fernandok.com/2018/11/multiplos-itags-com-esp32.html)
Got the comments in Portuguese translated to english :)

```
#include <BLEDevice.h>

// Customer id. It is incremented every time a new customer is connected
static uint16_t appId = 3;
// UUID of the service we want from iTag
static const BLEUUID serviceUUID("0000ffe0-0000-1000-8000-00805f9b34fb");
// UUID of the service characteristic (iTag button)
static const BLEUUID characteristicUUID("0000ffe1-0000-1000-8000-00805f9b34fb");

// Class responsible for controlling each iTag
class ITag: public BLENotifier {
  public: int pinNumber; // pin number linked to this iTag
  int pinStatus = HIGH; // pin status (HIGH or LOW)
  BLEClient * client; // iTag connection client
  std::string address; // address of iTag to which you will connect

  ITag(std::string addrs, int pNum) {
    // We link the values ​​of the address, pin number, put it as an output
    address = addrs;
    pinNumber = pNum;
    pinMode(pinNumber, OUTPUT);
    digitalWrite(pinNumber, pinStatus);
  }

  void connect() {
    // Check if there was a customer before and disconnect if so
    if (client! = NULL) {
      client -> disconnect();
      delete client;
    }

    // We create the client with a new id and connect to iTag
    client = BLEDevice::createClient(appId++);
    BLEAddress bleAddress(address);
    boolean connected = client -> connect(bleAddress);

    // If the connection was successful
    if (connected) {
      // We obtain the service and feature of the iTag button and link the onData function
      // to answer the button press
      BLERemoteService * remoteService = client -> getService(serviceUUID);
      BLERemoteCharacteristic * remoteCharacteristic = remoteService -> getCharacteristic(characteristicUUID);
      remoteCharacteristic -> registerForNotify(this);
    }
  }

  // Function called every time the iTag button is pressed
  void onData(BLERemoteCharacteristic * pBLERemoteCharacteristic, uint8_t * pData, size_t length, bool isNotify) {
    // We inverted the current state of the pin (from HIGH to LOW or from LOW to HIGH)
    // and send the new state to the pin output
    pinStatus = !pinStatus;
    digitalWrite(pinNumber, pinStatus);
  }
};

// Interval between each scan
#define SCAN_INTERVAL 3000
// Quantity of iTag we have (change according to the amount of iTags you have)
#define ITAG_COUNT 3

// Definition of the iTags that we will use
// For each iTag we link his address (obtained through a scan)
// and the pin whose status will be changed when the iTag button is pressed
ITags ITag[ITAG_COUNT] = {
  ITag("ff: ff: c2: 07: ab: 16", 25),
  ITag("ff: ff: c2: 07: 9d: b6", 26),
  ITag("ff: ff: c2: 07: 8f: bf", 27)
};

// Variable that will save the scan
BLEScan * pBLEScan;

// When the last scan occurred
uint32_t lastScanTime = 0;

void setup() {
  Serial.begin(115200);

  // We started BLE
  BLEDevice::init("");

  // We keep the object responsible for the scan
  pBLEScan = BLEDevice::getScan();
  pBLEScan -> setActiveScan(true);
}

void loop() {
  // Time in milliseconds since boot
  uint32_t now = millis();

  // If it's time to scan
  if (now - lastScanTime > SCAN_INTERVAL) {
    // Marks when the last scan occurred and starts the scan
    lastScanTime = now;
    scan();
  }
}

void scan() {
  // Performs the scan for 2 seconds
  BLEScanResults results = pBLEScan -> start(2);
  pBLEScan -> stop();

  // For each device found by the scan
  for (int i = 0; i < results.getCount(); i++) {
    // We keep the reference for the device and show it on the serial monitor
    BLEAdvertisedDevice advertisedDevice = results.getDevice(i);
    Serial.println("advertisedDevice:" + String(advertisedDevice.toString().C_str()));

    // For every iTag we have
    for (int j = 0; j < ITAG_COUNT; j++) {
      // If the scanned device is one of our iTags
      if (advertisedDevice.getAddress().toString() == iTags[j].address) {
        // We order to connect
        iTags[j].connect();
      }
    }
  }
}
```

h2zero, 2020-11-13:
Yes, this will work fine with minor changes to use NimBLE. 

Looks like the only changes necessary are the header include and removing the `appId` parameter from the `createClient` call.

m-menon, 2020-11-13:
Thanks. Checked a short while ago and it wouldn't compile on Arduino or on PlatformIO.

I ran into these errors

This line generates the following errors `class ITag: public BLENotifier {`

`expected class-name before '{' token`
`not a class or struct name`

h2zero, 2020-11-13:
Ahh yes, I didn't see that. You could recreate that class in the code or just create a notify callback function.

m-menon, 2020-11-14:
Yup, gona try recreating something similar. Thanks.

closed 2020-11-14 by m-menon

joelccrouch, 2021-12-08:
My ESP32 crashes if I try to use ArduinoOTA with NimBLE library.  If I comment out NimBLE, then OTA works fine.