h2zero/NimBLE-Arduino #145: Question: How to implement very fast value changes?

antonsperling, 2020-11-20:
Hello All,
hopefully someone can help me with the behavior i'm struggling with currently.
What i do:
Basically i use the the "NimBLE_Server" example on an ESP32 to change the value of a characteristic periodically. Let's say every 50ms. After invocation of `setValue()` i use `notifiy(true)`.
Notifications seem to occur as suggested, as i see the callbacks to `onNotify` and `onStatus` at the desired frequency. But not every Notification causes a new value to be receipt. That means i get the same Value about 10 times, before it changes on the Receiver.
I have tested it with my ionic-cordova-app and with nRF Connect (Andoid 9.0 on a Samsung S8).
```int loopcount=0;
void loop() {
  /** Do your thing here, this just spams notifications to all connected clients */
  loopcount++;
    if(pServer->getConnectedCount()) {
        NimBLEService* pSvc = pServer->getServiceByUUID(MOTO_SERVICE);
        if(pSvc) {
            NimBLECharacteristic* pChr = pSvc->getCharacteristic(MOTO_CHARACTERISTIC);
            if(pChr) {
                String strVal = String(loopcount);
                int len = strVal.length();
                char cVal[len];
                strVal.toCharArray(cVal, len);
                pChr->setValue(cVal);
                pChr->notify(true);
            }
        }
    }
    
  delay(50);
}
```
In my real-world-app the value would be a CSV-Datarow with about 80 bytes.
Am i doing something wrong?
Is the a limit how often a value can be changed and notified?

h2zero, 2020-11-20:
I don't see anything wrong with what you're doing here. Have you tried this with a different client device? Android has been known to drop notification packets occasionally.

If you have another esp32 you could try making it the client and check the results.

antonsperling, 2020-11-20:
Thanks for your advice. I will give this a try and report the results tomorrow.

antonsperling, 2020-11-21:
I'm sad to see the same behavior on a second ESP32.
I used the example above to send just a loopcounter. On the second ESP32 i used the "NIMBLE_client" example to receive these notifications.

Below the output on my Serial debug.
Every value is beeing receipt 10 times. So that is the same what i have seen before on my Android Device.

Is it possible, that this is an issue with my ESP32 Hardware?

```
11:54:11.416 -> 16524 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.449 -> 16569 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.516 -> 16629 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.549 -> 16674 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.615 -> 16734 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.649 -> 16779 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.715 -> 16824 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.748 -> 16884 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.815 -> 16929 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.848 -> 16974 Notification from 24:0a:c4:5a:05:2a, Value = 172
11:54:11.915 -> 17034 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:11.948 -> 17079 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.015 -> 17124 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.048 -> 17184 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.115 -> 17229 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.181 -> 17289 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.214 -> 17334 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.248 -> 17379 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.314 -> 17439 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.347 -> 17484 Notification from 24:0a:c4:5a:05:2a, Value = 173
11:54:12.414 -> 17544 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.481 -> 17589 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.514 -> 17634 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.580 -> 17694 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.613 -> 17739 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.680 -> 17784 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.713 -> 17844 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.780 -> 17889 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.813 -> 17949 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.880 -> 17994 Notification from 24:0a:c4:5a:05:2a, Value = 174
11:54:12.913 -> 18039 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:12.979 -> 18099 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.013 -> 18144 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.079 -> 18189 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.113 -> 18249 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.179 -> 18294 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.212 -> 18339 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.279 -> 18399 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.312 -> 18444 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.412 -> 18519 Notification from 24:0a:c4:5a:05:2a, Value = 175
11:54:13.412 -> 18549 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.478 -> 18594 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.545 -> 18654 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.578 -> 18699 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.611 -> 18744 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.678 -> 18804 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.745 -> 18849 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.779 -> 18894 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.845 -> 18954 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.878 -> 18999 Notification from 24:0a:c4:5a:05:2a, Value = 176
11:54:13.945 -> 19059 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:13.978 -> 19104 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.045 -> 19149 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.078 -> 19209 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.145 -> 19254 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.178 -> 19299 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.245 -> 19359 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.312 -> 19419 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.345 -> 19464 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.378 -> 19509 Notification from 24:0a:c4:5a:05:2a, Value = 177
11:54:14.445 -> 19554 Notification from 24:0a:c4:5a:05:2a, Value = 178
[... 20 lines cut ...]
11:54:15.510 -> 20619 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.543 -> 20664 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.610 -> 20724 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.643 -> 20769 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.677 -> 20814 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.743 -> 20874 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.810 -> 20919 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.876 -> 20994 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.910 -> 21024 Notification from 24:0a:c4:5a:05:2a, Value = 180
11:54:15.943 -> 21069 Notification from 24:0a:c4:5a:05:2a, Value = 181
[... cut ...]
11:54:16.343 -> 21474 Notification from 24:0a:c4:5a:05:2a, Value = 181
11:54:16.409 -> 21534 Notification from 24:0a:c4:5a:05:2a, Value = 181
11:54:16.443 -> 21579 Notification from 24:0a:c4:5a:05:2a, Value = 182
11:54:16.509 -> 21624 Notification from 24:0a:c4:5a:05:2a, Value = 182
11:54:16.576 -> 21684 Notification from 24:0a:c4:5a:05:2a, Value = 182
11:54:16.609 -> 21729 Notification from 24:0a:c4:5a:05:2a, Value = 182
11:54:16.675 -> 21789 Notification from 24:0a:c4:5a:05:2a, Value = 182
```

antonsperling, 2020-11-21:
I've just raised the wait time to 1000ms on my sender device. Same here. So this is not because of "fast changing values", but every notification seems to be processed 10 times.
Here output with raised delay time:
```
12:04:24.313 -> 49749 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:25.279 -> 50739 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:26.310 -> 51744 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:27.274 -> 52734 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:28.306 -> 53739 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:29.305 -> 54744 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:30.302 -> 55749 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:31.300 -> 56739 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:32.299 -> 57744 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:33.296 -> 58749 Notification from 24:0a:c4:5a:05:2a, Value = 2
12:04:34.294 -> 59739 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:35.291 -> 60744 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:36.289 -> 61749 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:37.287 -> 62739 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:38.317 -> 63759 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:39.316 -> 64749 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:40.314 -> 65755 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:41.312 -> 66745 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:42.308 -> 67750 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:43.306 -> 68755 Notification from 24:0a:c4:5a:05:2a, Value = 3
12:04:44.302 -> 69745 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:45.299 -> 70750 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:46.296 -> 71755 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:47.292 -> 72745 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:48.289 -> 73750 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:49.319 -> 74755 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:50.284 -> 75745 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:51.315 -> 76750 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:52.313 -> 77755 Notification from 24:0a:c4:5a:05:2a, Value = 4
12:04:53.312 -> 78760 Notification from 24:0a:c4:5a:05:2a, Value = 4
```

antonsperling, 2020-11-21:
Arrrggg, don't ever let a Java Programmer do stuff with Microcontrolers and C/C++.
The Problem on this Example was `int len = strVal.length()` which caused the last byte to be cut off.

closed 2020-11-21 by antonsperling

h2zero, 2020-11-21:
Glad you found your issue!