h2zero/NimBLE-Arduino #146: Periodically change advertising between iBeacon and ServiceUUID

lknop, 2020-11-24:
Due to limitations of the BT API in iOS, I need one ESP32 BLE device to advertise both as an iBeacon and as a BLEServer with defined BLEServices.

Preliminary tests suggest that it should be possible but I struggle with the API in `NimBLEAdvertising` - once I call `setAdvertisementData` I am unable to use `addServiceUUID` since the `m_customAdvData` is already set to true and there is no API to reset the payload.

Please point me in the right direction if it is possible with this stack and with the current API - my current understanding is that the stack will allow it but the API does not. 

A dummy idea to circumvent it without rewriting most of the logic from `NimBLEAdvertising::start` would be to:
1. call `addServiceUUID`
2. start advertising
3. get payload and store in a variable a
4. stop advertising 
5. setup beacon advertisements with `setAdvertisementData`
6. get payload and store in variable b
7. call ble_gap_adv_set_data directly, alternating between values a and b

Does it make any sense? Is there a better way to do it?

And last question - would it be welcome to have a method in `NimBLEAdvertising` to reset it to defaults?

chegewara, 2020-11-24:
You can use `setAdvertisementData` for both, iBeacon and regular service UUID IIRC.

Yes, it would be nice to have option to set if you want to use custom advertising or standard, if its not already there.

lknop, 2020-11-24:
You are right, I thought I would need to calculate the advertising packet and spend some time debugging it, but I can just start advertising, do a BLE scan using another device and copy the raw value. This will be much easier than my idea above. 

If the switching between the two dataset proves to be working I will create a PR to allow resetting the payload together with the `m_customAdvData` flag.

chegewara, 2020-11-24:
It is working. I have app where i am doing exactly this, just with old BLE library.

lknop, 2020-11-24:
Thanks, this is great news. Can you share if the beacon is properly recognized by iOS devices? I think I saw some of your comments when googling about that. From my tests iOS is extremely picky when it comes to BT LE (and not only that, but that's a different story...) and I am wondering if the advertisement packet replacement may pose any problems here.

chegewara, 2020-11-24:
Well, yes and no. iOS does not recognize iBeacon IIRC, but custom app on iPhone does recognize it as device that is looking for. Its the matter if you want that all apps from store can find and recognize iBeacon in your device or just you want your own app find it.

h2zero, 2020-11-24:
There is not enough room in the advertising packet for the iBeacon advertising and service uuid's. 

What you could do however, is set the iBeacon advertising and put the service uuids in scan response.

h2zero, 2021-01-08:
@lknop did you get this sorted?

lknop, 2021-01-08:
Well yes and no. When there are services in the scan response, iOS does not recognize the beacon properly - I think it cannot be advertised as connectable.

I was able to switch periodically between ibeacon advertisements and service advertisements, but this had a large impact on performance. I divided the time to give ~25% to beacon and 75% to service advertising. Sometimes this caused the beacon to not be recognized for 20-30 seconds. And the service I use has to wait for an external event before replying to the mobile device, so the connection has to stay active for several (5-10) seconds and this also caused problems with connection being dropped or not possible at all.

My impression is that if the beacon detection was not time-critical and the service could reply and disconnect within the timeslot assigned to the service advertising this could work quite well, but in my use case this was not possible.

One more remark - the switching of the advertisements required some off time to work properly, I did not investigate if it was something in my code, the stack or the hardware. The actual algorithm that was optimal for me looked like this:
- 500 ms beacon
- 80 ms off
- 1800 ms service
- 80 ms off

I think the issue can be closed, I decided to use external hardware for the beacons which worked reliably and I keep only the service code on the esp32.

h2zero, 2021-01-08:
Thanks for the update, the off time requirement is curious, I'm not sure why that would be. I will look into this further as NimBLE provides for multiple advertisement data sets that can be switched between. I have not yet used them or created an API for it, I think that would make a good future enhancement.

h2zero, 2022-04-05:
Closing as resolved.

closed 2022-04-05 by h2zero