h2zero/NimBLE-Arduino #147: How can I get which device was disconnected?

JesusAlexlara, 2020-11-24:
I try yo get which device was disconnected, NimBLEServerCallbacks only has "void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc)" on the other hand, for disconnection I don't have something similar.
Is there any way to know which device was disconnected?

tanks

h2zero, 2020-11-25:
Thank you for bringing this up. Currently there is no way to do this through the API but it could be added similarly to the `onConnect()` callback. I will try add this functionality tomorrow.

h2zero, 2020-11-26:
Hi @JesusAlexlara here is a patch for this if you could please test it and let me know if it works for you. I will push the change to master if it's satisfactory.

[disconnect.txt](https://github.com/h2zero/NimBLE-Arduino/files/5600088/disconnect.txt)

doudar, 2020-12-02:
I'm running into a similar issue where knowing the device that disconnected would be helpful. Thanks for your awesome contribution @h2zero !

h2zero, 2020-12-02:
@doudar You're welcome. Did you try the patch I posted above? If it's working for you I will merge it into master.

doudar, 2020-12-04:
I'll see if I can get some time to test it out. Forking right now. It might be a couple days before I can report back as I'll need to modify my code as well and then test for stability.

JesusAlexlara, 2020-12-04:
My program has been running for several days and everything has worked correctly.

h2zero, 2020-12-05:
Thanks @JesusAlexlara, I will merge the patch ASAP.

doudar, 2020-12-05:
I've got about 3 hours of testing in so far. No issues yet either.

h2zero, 2020-12-29:
Closing as resolved in commit 9074bba.

closed 2020-12-29 by h2zero