h2zero/NimBLE-Arduino #148: max characters to write in a characteristic

ccssw, 2020-11-25:
I found that I can only write up to 216 characters in a characteristic, otherwise the callback is not run. Is it possible to enlarge it?

h2zero, 2020-11-26:
That doesn't seem right, I'm able to write from a client device and receive them on the server. What is the MTU size?

ccssw, 2020-11-26:
How can I get the MTU size?

ccssw, 2020-11-26:
When I try to set the MTU with `NimBLEDevice::setMTU(BLE_ATT_MTU_MAX)` size I get:

`
D NimBLEDevice: ">> setLocalMTU: 527"
assertion "mu->handle" failed: file ".pio/libdeps/noukee/NimBLE-Arduino/src/porting/npl/freertos/src/npl_os_freertos.c", line 185, function: npl_freertos_mutex_pend
`

h2zero, 2020-11-27:
That's a very strange error. I have not had this issue nor seen any reports regarding this previously. I will do what I can to reproduce.

As for your initial issue, what is your setup are you connecting with a phone to the esp32 or another device? If from a phone you may need to use long write if the data size is greater than MTU-3.

h2zero, 2020-12-01:
Ok, so the limit you see is due most likely to the phone MTU. You cannot change it as the spec states that the lowest of the 2 MTU's is the one that is used.

In that situation you will need to use a long write procedure.

ccssw, 2020-12-01:
> what is your setup are you connecting with a phone to the esp32 or another device?

Yes, it's a smart phone.

h2zero, 2020-12-29:
@ccssw any updates on this?

ccssw, 2021-01-11:
Requesting the MTU to 517 fixed the issue for now.

chegewara, 2021-01-11:
Max MTU is 517 IIRC.

h2zero, 2021-02-01:
>Max MTU is 517 IIRC.

Max for NimBLE is 527 to accomodate signed writes, which we don't do (yet) so 517 works 😄 

Going to close this as it seems to be resolved.

commit 71aa0205 (2024-06-02): Initialize ble_gatt_chr_def[]. Fixes #148 (#150)

IDF 5.2 introduced a new member, cpfd, to the
ble_gatt_chr_def struct. It needs to be initialized
to nullptr in order to avoid accessing uninitialized
memory. By initializing the whole struct, we get
everything initialized in a backward-compatible way.
closed 2021-02-01 by h2zero