h2zero/NimBLE-Arduino #149: Connection parameters for a single client; setConnectionParams / updateConnParams

m-menon, 2020-11-25:
Just wondering if you have ever needed to work out the optimal settings for a single client? I am referring to 
`setConnectionParams` (Set initial connection parameters) and `updateConnParams` (Change parameters after connection)

[Example](https://github.com/h2zero/NimBLE-Arduino/blob/cdb7d7839c1406c71cd8a158d28c11f2a970ded1/examples/NimBLE_Client/NimBLE_Client.ino#L171)

h2zero, 2020-11-26:
The device you are connecting to has a random address that changes every X minutes. You would need to bond with it in order to resolve the address and use it's ID address instead. I've been working on this locally and hope to provide a solution soon.

Optimal connection parameters depends on the application. If connecting to a battery powered device I keep the interval long to save power. If you need more data throughput you should keep it shorter. Mostly a matter of the situation and preference.

m-menon, 2020-11-26:
@h2zero thanks for the pointers. I will try different values and see how it goes.

closed 2020-11-26 by m-menon