h2zero/NimBLE-Arduino #150: Eddysone UID beacon

AdamMiltonBarker, 2020-11-26:
Hi guys, haven't had much luck creating an Eddystone UID beacon for ESP32 with this library, is it supported, if not can you suggest a library that could be used? TIA

h2zero, 2020-11-27:
Hi @AdamMiltonBarker eddystone uid is supported the same way as the other beacons. There is no direct api for this however. 

If you post your code I can help you with it.

AdamMiltonBarker, 2020-11-27:
OK thanks will keep working on it. Having a bit of a nightmare lol. Yesterday using a project based on the URL example the accuracy was great, I was able to position myself pretty accurately using an Android scanner, looks like it was a freak, today the accuracy is terrible. Had spent a lot of time working on RPI beacons for indoor positioning and came to the conclusion beacons are simply no good for this task, thought my luck was in yesterday, sadly it seems not.

Thanks for the offer, I am working right now but will upload code tomorrow. Speak soon.

h2zero, 2020-12-29:
@AdamMiltonBarker any updates on this?

h2zero, 2022-04-05:
Closing as stale.

closed 2022-04-05 by h2zero