h2zero/NimBLE-Arduino #151: Pairing callback not invoked

skypoldo, 2020-11-29:
Hi! I'm implementing a BLE client connecting to a BLE server requiring pairing to write some characteristics (not very expert in BLE, though...). I'm working on NimBLE_Client example, and I successfully discovered the service and listed the characteristics I need.

What I don't see is the onPassKeyRequest callback invoked, and the characteristics result to be not writeable, so I presume that the security is not correctly configured. BLE server uses 123456 as passkey, and works fine using a BLE app on my phone.

I used the following:
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY); // Should use the passkey
  NimBLEDevice::setSecurityAuth(false, false, true);  // Default

I also added pClient->secureConnection() after connecting the client to the server (it returns true) as I read something about it in a previous issue.

The onAuthenticationComplete() callback finds the following: 

  desc->sec_state.authenticated = 0
  desc->sec_state.bonded = 1
  desc->sec_state.encrypted = 1

As some characteristics can be notified, they are successfully subscribed, but I can't read the result as I need to send a command to start the notifications.

Project is developed in Visual Studio Code, platform is espressif32, framework arduino.

Any suggestion will be appreciated!

skypoldo, 2020-12-04:
Ok, found out the problem... In NimBLE_client example, the test on the number of clients is not working correctly.

I function connectToServer() the test

if(NimBLEDevice::getClientListSize())

is not satisfied as the number of clients returned is 1 instead of 0, even if the program has just been executed. So following the logic, the following instruction is executed:

pClient = NimBLEDevice::getDisconnectedClient();

but no client has been created, and therefore no callback is assigned to the client. Changing the test to:

if(NimBLEDevice::getClientListSize() > 1)

solved my issue.

I'm wondering why is the function getClientListSize() is not returning 0. Any idea?

h2zero, 2020-12-04:
Sorry, I forgot to get back to this one. Interesting findings, I will need to look into this today. I'm glad you found a solution for now, thanks for reporting!

skypoldo, 2020-12-04:
No problem! Please let me know what you'll find and if it needs to be corrected.

Thank you!

skypoldo, 2020-12-05:
Shame on me! As I moved the declaration of the client handle global, I declared it together with the creation with:

NimBLEClient* pClient = new NimBLEClient();

And that's where the non empty list came from...

Sorry about this, all working as expected.

h2zero, 2020-12-05:
Awesome, thanks for the update.

h2zero, 2020-12-29:
Closing as resolved.

closed 2020-12-29 by h2zero