h2zero/NimBLE-Arduino #153: setting default mtu / conn paramaters

foofel, 2020-12-05:
Hey,

I am using this library to send hx711 (load sensor) data via bluetooth and it works very well, as the hx711 can read up to 80hz i need to decrease the connection time, i am currently doing this in the 
```void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {```
function but is this the correct way? Is it possible to specify my desired default params so they are automatically set on the connection startup? Also i need a larger mtu, i currently handle this on the client side but this forces the client to know the mtu size, is it possible to set the default mtu on the server side?

Best

h2zero, 2020-12-06:
The connection parameters are controlled by the client, the server can request parameters and the client can reject or accept them, no guarantees. The MTU can be set after calling `NimBLEDevice:init()` by calling 'NimBLEdevice::setMTU()` but the lowest value between the client and the server will be selected.

h2zero, 2020-12-29:
@foofel any updates on this?

h2zero, 2021-02-01:
Closing as stale, please re-open if you have further issues.

closed 2021-02-01 by h2zero

foofel, 2022-12-07:
Oh sorry didn't see your comment (little late now), it works perfectly as you describe with the client requesting the mtu. also setting the connection parameters work. thanks for the great library! :)

jojo0329, 2023-07-12:
@h2zero ,Hey,I'm having the same problem, I need to set my mtu to 1024. however the client set it and it can only be set to 255，although i calling 'NimBLEdevice::setMTU()`！

h2zero, 2023-07-12:
The MTU can be a maximum of 512 as per BLE specifications. Both the client and the server need to support this value and if not the lowest value supported between the devices will be used.

jojo0329, 2023-07-16:
> 

> The MTU can be a maximum of 512 as per BLE specifications. Both the client and the server need to support this value and if not the lowest value supported between the devices will be used.
Thanks for the reply, I did in fact pay attention to that. Even though I've supported 1024mtu in my code, I set 1024 in the app and it still doesn't modify the value of mtu, it becomes 255 instead
`BLEDevice::init("BZ");
// ADDED to support larger MTU
    NimBLEDevice::setMTU(1024);//test
// Create the BLE Server`
[image]

h2zero, 2023-07-17:
The controller of each device will reject such values that is why you're not seeing it change. Try it with 512 and see if it changes to that.