h2zero/NimBLE-Arduino #154: Pairing via passkey doesn't appear to work

dom96, 2020-12-06:
So I am running some code based on the NimBLE_Server example with the `NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);` enabled. Connecting to the device via nRF connect doesn't trigger a passkey prompt though.

Is this expected? I have tried setting the passkey via `setSecurityPasskey` directly as well as the `onPassKeyRequest` doesn't get called. Is there something else I need to enable to get this to work?

h2zero, 2020-12-07:
Are you calling this before or after `NimBLEDevice::init()`?

dom96, 2020-12-07:
After. I literally am using https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Server/NimBLE_Server.ino#L159 with that line uncommented.

h2zero, 2020-12-07:
Thanks for getting back. It was working fine for me last time I tested, maybe something changed since. I'll test shortly and see if I can reproduce.

h2zero, 2020-12-08:
Hi @dom96, I have tested this against the latest master using nRFConnect and seems to be working fine for me. You should note that the passkey is not requested until an attempt is made to read or write a characteristic that has the `NIMBLE_PROPERTY::WRITE_ENC` or `NIMBLE_PROPERTY::READ_ENC` flagged in it's creation (you can see this in the `BEEF` characteristic in the example code). 

If you wish to secure the connection immediately when a client connects you will have to do so manually via the `onConnect()` callback. 

The code to do that would look like this:
```
void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
       NimBLEDevice::startSecurity(desc->conn_handle);
    };
```

Let me know if this works for you.

dom96, 2020-12-09:
Ahh, I was missing the `_ENC` suffix. It looks like the BLE library I am using for my Android app doesn't deal with this unfortunately, I guess the OS is supposed to show the pairing dialog but I just get rejections. In any case, I think I'll just end up implementing my own pairing scheme and hope it won't cause any significant security problems.

Thanks for the help!

closed 2020-12-09 by dom96

h2zero, 2020-12-10:
Interesting that the library doesn't handle it. I haven't made any android apps so I'm not familiar with it's BLE implementation. 

Let me know if you have any more issues.