h2zero/NimBLE-Arduino #155: Hang at disconnect during notify

btsimonh, 2020-12-08:
I'm trying to use NimBLE-arduino in Tasmota, and having an issue with an unrecoverable BLE hang.

Basically, I have:
a single client.

I then
1/ stop a scan
2/ connect by MAC
3/ get a service
4/ get a characteristic (for read/write use)
5/ get a characteristic for notify.
6/ subscribe for notify
7/ write to the read/write characteristic.

All Initialisation and operation of BLE is performed from a single Task (separate from the main task in Tasmota). 

this works a lot of the time.
But often, I see logs like the below.

basically, after the notification comes in, a little later (~1s) I disconnect.
But the disconnect does not seem to be successful, and client->isConnected() remains true.

After this, I can't do anything with BLE, it seems broken.  I can't even ::deinit and ::init - my only option is reboot.

sometimes it is associated with an earlier occurence of the log:

lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)

Any advice as to how to avoid this, or further diagnose?

I have tried with 
ESP32 Arduino 1.0.4 (IDF 3.2) Release
and latest git.
behaviour seems similar.

in the log below '....' indicates where I've remove http related sections, and '>>' indicates comments I have added.

br,
Simon

```
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
22:50:52 SL4-26 runCurrentOperation: new currentOperation
22:50:52 SL3-27 Scan ended
22:50:52 SL2-28 MI32: scancomplete
22:50:52 SL1-29 scancompleteCallbacks 0 0
22:50:52 MQT: tele/tasmota_esp32/BLE = {"active":{"5FD2D49DE1D7":{"n":""},"2D82E35C7469":{"n":""},"743E6283EA71":{"n":""},"1D1151D65BFF":{"n":""},"001A22092C9A":{"n":"CC-RT-M-BLE"},"001A22092CDB":{"n":"CC-RT-M-BLE"},"001A22092FB7":{"n":"CC-RT-M-BLE"}}}
D NimBLEClient: ">> connect(00:1a:22:09:2f:b7)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
22:50:53 SL1-29 attempt connect 00:1a:22:09:2f:b7
[V][WebServer.cpp:291] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: m=1
.....
[V][Parsing.cpp:231] _parseRequest():  Arguments: m=1
[V][WebServer.cpp:291] handleClient(): New client
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: m=1
.....
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.212/?
D NimBLECl[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
ie[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
nt: "Got Client[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9,en-GB;q=0.8,fr;q=0.7
 e[V][Parsing.cpp:247] _parseArguments(): args: m=1
ve[V][Parsing.cpp:265] _parseArguments(): args count: 1
n[V][Parsing.cpp:273] _parseArguments(): pos 0 =@1 &@-1
t [V][Parsing.cpp:284] _parseArguments(): arg 0 key: m value: 1
[V][Parsing.cpp:291] _parseArguments(): args count: 1
"
[V][Parsing.cpp:230] _parseRequest(): Request: /
[V][Parsing.cpp:231] _parseRequest():  Arguments: m=1
D NimBLEClient: "Peer requesting to update connection parameters"
>> what does this mean?

D NimBLEClient: "MinInterval: 40, MaxInterval: 200, Latency: 0, Timeout: 5000"
D NimBLEClient: "Rejected peer params"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"
>> delayed log.....
22:50:58 SL1-29 onConnParamsUpdateRequest 00:1a:22:09:2f:b7
D NimBLEClient: "<< connect()"
D NimBLEClient: ">> getService: uuid: 3e135142-654f-9090-134a-a6ff5bb77046"
D NimBLEClient: ">> retrieveServices"
22:50:58 SL2-28 onConnect 00:1a:22:09:2f:b7
22:50:58 SL1-29 connected 00:1a:22:09:2f:b7 - will getservice
D NimBLEClient: "Service Discovered >> status: 0 handle: 1024"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 3e135142-654f-9090-134a-a6ff5bb77046"
22:50:58 SL1-29 got service
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 1041"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 3e135142-654f-9090-134a-a6ff5bb77046"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 1057"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> setNotify(): Characteristic: uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a, handle: 1057 0x0421, props:  0x1a, 01"
D NimBLERemoteCharacteristic: ">> getDescript22:50:59 SL1-29 got notify characteristic
or: uuid: 0x2902"
D NimBLERemoteCharacteristic: ">> retrieveDescriptors() for characteristic: d0e8434d-cd29-0996-af41-6c90f4e0eb2a"
D NimBLERemoteCharacteristic: "Descriptor Discovered >> status: 0 handle: 1072"
D NimBLERemoteCharacteristic: "Descriptor Found"
D NimBLERemoteCharacteristic: "<< Descriptor Discovered. status: 0"
D NimBLERemoteCharacteristic: "<< setNotify()"
D NimBLERemoteDescriptor: ">> Descriptor writeValue: Descriptor: uuid: 0x2902, handle: 1072"
D NimBLERemoteCharacteristic: ">> writeValue(), length: 1"
22:50:59 SL2-28 subscribe for notify
22:50:59 SL1-29 got read/write characteristic
I NimBLERemoteCharacteristic: "Write complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "<< writeValue, rc: 0"
22:50:59 SL1-29 write characteristic
D NimBLEClient: "Got Client event "
D NimBLEClient: "Notify Recieved for handle: 1057"
D NimBLEClient: "checking service 3e135142-654f-9090-134a-a6ff5bb77046 for handle: 1057"
D NimBLEClient: "Got Notification for characteristic Characteristic: uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a, handle: 1057 0x0421, props:  0x1a
Descriptor: uuid: 0x2902, handle: 1072"
D NimBLEClient: "Invoking callback for notification on characteristic Characteristic: uuid: d0e8434d-cd29-0996-af41-6c90f4e0eb2a, handle: 1057 0x0421, props:  0x1a
Descriptor: uuid: 0x2902, handle: 1072"
22:50:59 SL1-29 Notified length: 6
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
22:51:00 SL2-28 notify operation complete

>> delayed log - probably cause the dicsonnect above...

22:51:00 SL1-29 runTaskDoneOperation: disconnecting connected client
[V][WebServer.cpp:291] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: m=1
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
.....
[V][Parsing.cpp:231] _parseRequest():  Arguments: m=1
>> log indicates client->isConnected() returns true! ????
22:51:01 SL1-29 wait disconnect 1
22:51:01 MQT: tele/tasmota_esp32/SENSOR = {"Time":"2020-12-08T22:51:01","IBEACON":{"MAC":"001A22092C9A","NAME":"CC-RT-M-BLE","RSSI":0,"STATE":"OFF"}}
D NimBLEClient: ">> disconnect()"
E NimBLEClient: "ble_gap_terminate failed: rc=2 "
D NimBLEClient: "<< disconnect()"
22:51:02 SL2-28 wait disconnect 2
>> cause of disconnect above..  but is_connected() is still true.....
22:51:02 SL1-29 retry disconnect 3
22:51:02 WIF: Checking connection...
22:51:02 SL1-29 wait disconnect 3
22:51:03 SL1-29 wait disconnect 4
22:51:03 SL0-30 wait disconnect 5
22:51:03 SL1-29 BLE Failed, restarting Tasmota
[V][WebServer.cpp:291] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: m=1
....
[V][Parsing.cpp:231] _parseRequest():  Arguments: m=1
22:51:04 BLE Failure! Restarting in 5s
D NimBLEClient: ">> disconnect()"
E NimBLEClient: "ble_gap_terminate failed: rc=2 "
D NimBLEClient: "<< disconnect()"
D NimBLEClient: ">> disconnect()"
E NimBLEClient: "ble_gap_terminate failed: rc=2 "
D NimBLEClient: "<< disconnect()"

```

btsimonh, 2020-12-08:
hmm...  looks like we are missing 'D NimBLEClient: "Got Client event "' for that disconnect.  I.e. BLE does not give us a disconnect?

Also, just tested with write only, and it still fails (no notify requested).

h2zero, 2020-12-09:
Hi @btsimonh, yes it seems the event is not being triggered. I'll see if I can reproduce this. Thanks for reporting.

h2zero, 2020-12-09:
When you call disconnect do you perform any delay? Ideally you should wait until the disconnect completes which will trigger the callback.

btsimonh, 2020-12-09:
the disconnect code is below.

It feels like it is a lot worse with heavier Wifi traffic.
Is there a wifi config I could be missing which enables time switching between BLE and Wifi on the radio?  I can't find anything in the ESP documentation about how wifi and BLE can co-exist; just lot's of speculation about a shared radio.

```
    if ((*ppClient)->isConnected()){
      BLE_ESP32::SafeAddLog_P(LOG_LEVEL_DEBUG,PSTR("runTaskDoneOperation: disconnecting connected client"));
      (*ppClient)->disconnect();
      // wait for 1/2 second after disconnect
      int waits = 0;
      do {
        vTaskDelay(500/ portTICK_PERIOD_MS);
        if (waits) BLE_ESP32::SafeAddLog_P(LOG_LEVEL_ERROR,PSTR("wait disconnect %d"), waits);
        waits++;
        if (waits == 3){
          // try again?
          if (waits) BLE_ESP32::SafeAddLog_P(LOG_LEVEL_ERROR,PSTR("retry disconnect %d"), waits);
          (*ppClient)->disconnect();
        }
        if (waits == 6){
          // try again?
          BLE_ESP32::SafeAddLog_P(LOG_LEVEL_ERROR,PSTR("BLE Failed, restarting Tasmota"));
          BLEStop = 1;
          BLERestartTasmota = 5;
          return;
        }
      } while ((*ppClient)->isConnected());
```

btsimonh, 2020-12-09:
just looking through the code, one thing I notice is use of 1ms delays, e.g.:
```
        if(rc == BLE_HS_EBUSY) {
            vTaskDelay(1 / portTICK_PERIOD_MS);
        }
```
from here https://github.com/espressif/arduino-esp32/issues/2981 it seems this will not give up the timeslice.
Suggest if the timeslice should be given up, that a minimum delay would be vTaskDelay(1); with a comment for reasoning.

(I did get an idle task watchdog restart **once**, naming 'btController' as the task in progress....).

BUT, I changed all of them in NimBLE to vTaskDelay(1), and the problem persists, so this is not the cause.

btsimonh, 2020-12-09:
even more detailed log here:
https://gist.github.com/btsimonh/d9b4bc7a12e8ab81c4adb9b999d4b425
in this log, I stall at the point of failure by staying in the above while loop with an additional vTaskDelay of 5s per loop.
I've included quite some time after the failure - there is nothing more from BLE low level after it stops -i.e. looks like it never got the disconnect packet?

p.s. I added the two changes in esp_nimble vs. the version in arduino-nimble - but no improvement. (e.g. a PR addressing a stack overflow due to recursion).

btsimonh, 2020-12-09:
raised an issue here:
https://github.com/apache/mynewt-nimble/issues/890

I think there should be some timeout on not receiving the terminate confirmation - but I don't know if this should be in mynewt or in arduino-nimble.  Is there some way to force the connection to abort, so that we could put a timeout inside or outside of arduino-nimble?

btsimonh, 2020-12-09:
I've added the ability to reset the BLE stack through NimBLE device as a test - it ran through it's first iteration a few minutes ago; let's see how many resets it can survive.  May be a good belt-and-braces thing to add anyway....

h2zero, 2020-12-09:
I've done some testing and have not been able to reproduce this issue. Looking at the log it seems the disconnect is occurring but the callback is not triggered. This comes when the controller returns the event to the host, which it seems does not happen.

I see you are performing some wifi operations in the time that BLE is working. This could be the source of the issue. Since the controller is closed source we don't have access to debug at that point sadly. I suggest for testing to remove the wifi code and run BLE only to check if the issue still occurs.

btsimonh, 2020-12-09:
hmm..  
I'm also convinced that wifi makes it worse, but I can't easily stop it using wifi - the writes are triggered over MQTT.... and Tasmota is quite a complex project.
I'm also talking to possibly particularly bad BLE devices - EQ3 radiator valves based on dubious hardware.
BLE transactions are so complex and so variable, I'm not surprised that you could not reproduce.
But at the moment, I can't see anything in mynewt or NimBLE designed to cope with a disconnect timeout.  So I think this should be added.  At what level, i'm not sure.

I will fork the repo, and push a branch with my current changes, which are so far successful in restoring operation - maybe they will be a good discussion point.

btsimonh, 2020-12-09:
hi @h2zero,
these are the changes I made to enable reset of the BLE stack when stuck....
https://github.com/h2zero/NimBLE-Arduino/compare/master...btsimonh:btsimonh-test
The reset restores operation, but I'm seeing crashes too...  I'll monitor better tomorrow (too many logs to catch the crash...)

h2zero, 2020-12-09:
I don't think it should be necessary to reset the host. There is something else happening that shouldn't be and would be better if we could find that.

The disconnect code you shared, is that in your notification callback handler? That could be a source of trouble, delays used in the callback will stuff things up.

btsimonh, 2020-12-10:
no, I'm not even asking for notify for this testing..  only write.  And the notify callback has been stripped to the bone, just noting that it happened and capturing the data (did this after I found that the notify cb could occur BEFORE the write returned - even though the write is necessary to trigger the notify!!! - so I must subscribe and then write, but be able to catch the notify response before the write is officially over....  two days ago I thought the issue could be related to the write/notify timing; maybe it is but at a lower level? - but now I'm testing with write only....

(
some stupid thoughts around this.... I observed notify hitting me before the write function completed in at least two of the previous failures.  Now I am only writing to remove this as a failure mode.  BUT.  is it **possible** that the device ALWAYS sends a notify packet when it gets a write - or has somehow remembered that I requested notify ONCE, and so is sending one?  And what if this packet REALLY arrives during the write packet sequence, not after, just sometimes?  I know nothing about BLE packet sequences, so it's just a thought.
-- in the log in the gist above... my request command is (line 582):
Data "m:001A22092FB7 s:3e135142-654f-9090-134a-a6ff5bb77046 c:3fa4585a-ce4a-3bad-db4b-b8df8179ea09 w:03 u:581046 go"
-- I am NOT requesting notify (would be n:xxxxx)
but...  at line 572, we DO receive a notification at a low level.
BUT..  the log looks no different to a successful transaction, until it just does not get the disconnect callback to NimbleClient.
log containing success: https://gist.github.com/btsimonh/dd96853159594216d66addebf611b3a7
)

Ref reset() - I wanted to test if it was a hardware thing or if the stack could proceed...  I still think it's a good addition to arduino-nimble, as in the failed situation, deleteClient() refuses to delete because it believes it's connected, so you have no way out.  I didn't try yet just deleting and recreating the dead client - I am guessing that the connected handle needs disposing of somehow in the myNewt, and could not work out the best way to do that, hence the reset which should do it (in a sledgehammer kind of way) .... and deinit(), init() similarly did not do the job in a previous test probably because it could not delete the client?.

I pushed a branch with my current test code:
https://github.com/btsimonh/Tasmota/blob/Tasmota/BLE_TEMP/tasmota/xdrv_47_BLE_ESP32.ino

ALL the BLE code is run from a single task (I was running some from main thead, but moved it all into a single task because i was concerned it could be a cross-thread thing, but it seems not).

[BLEOperationTask ](https://github.com/btsimonh/Tasmota/blob/afd80acdc22273aeb3d7b16871e0d4887c7483f4/tasmota/xdrv_47_BLE_ESP32.ino#L1210) is the task.  

It runs runCurrentOperation once per second at most.  If no operation is in progress, it starts a 20s scan.
BLESensorCallback is the client callback class, and only logs.

when an operation is 'complete' (for write, this just means 'when we finished writing'), the next runCurrentOperation (1s later) will trigger runTaskDoneOperation, which removes the current operation, and disconnects if the client is not disconnected.  This is now the only normal disconnect (I don't leave the task, which is the only other disconnect() in the code).

Testing specifics:
I usually have 2-3 devices n range which I'm trying to talk to, plus another 3-4 which are providing adverts only.
Example:
If I see 3 devices in range of the required type, I will cycle through them, issuing a write command and waiting for a result.
'normally' at least one of the three is not close enough for a reliable connection, so I see a failure of some sort, either a failure to connect, or a failure of some part of the write process.  in this case, I get a fail result, disconnect, and move on.

A new write is triggered when the external device has received the Success/fail, so normally there is a 2-5 seconds of scan before the next request.

What I will try to do today if I get time:
I'm currently driving Tasmota via MQTT.  I may be able to disconnect the MQTT, and not connect a webpage to Tasmota, and this would reduce network traffic significantly (down to wifi maintenance packets only), and I will attempt to create a Node-Red flow which drives it over serial, and also captures the logs looking for failures and reboots.  My biggest problem at the moment is that VSCode/PIO only has so much scrollback stored, so I have to sit and watch for a failure to catch, and then manipulate VSCode rather quickly before that failure's logs are lost! :).

btsimonh, 2020-12-10:
with only one EQ3 visible (and at fairly close range), it's run faultlessly for ages.  Even with the network traffic.
I've also INCREASED the network traffic by logging to MQTT, and it's still running nicely.
I moved the ESP32 so that it got a sniff of a second device, and it seemed ok (a sniff to see it, but always failure to connect).
When I introduced a 3rd device (very close range, new batteries), it failed quite quickly afterwards.  (the failure was against the device which it had been talking to quite well - (229 successful writes vs 21 failure to connect).

btsimonh, 2020-12-10:
but also a new type of failure, logged here:
https://gist.github.com/btsimonh/1ec22e96051a98efe8c4202543782727
after line 2699, BLE logs next a disconnect at 3481 (90 seconds), and then nothing.
This seems to be stuck in the client->connect() call...

at line 2681 in the log, this seems missing:
```
D NimBLEClient: "Connection established"
GATT procedure initiated: exchange mtu
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0x00 0x01 
txed att command: mtu req; conn=0 mtu=256
```

It is as though these lines in client happened:
```
        case BLE_GAP_EVENT_CONNECT: {
            if(!client->m_waitingToConnect)
                return 0;
```

Maybe BLE_GAP_EVENT_CONNECT came in before you set m_waitingToConnect = true?  if it did, it's stuck in the ulTaskNotifyTake() :(.

I'll rebuild with the m_waitingToConnect = true before the ble subsystem call.... and add some more diags around this point and the disconnect point.

h2zero, 2020-12-10:
I think I have an answer for this. I should have noticed this earlier but in your log [here](https://gist.github.com/btsimonh/d9b4bc7a12e8ab81c4adb9b999d4b425#file-blelog-txt-L641) you can see that the supervision timeout is set to 50 seconds(5000 x 10ms). 

So when you send the disconnect command in your busy environment the acknowledgement packet from the peripheral may not be received, which the controller will then resort to waiting for the timeout to expire before triggering the event and callback.  Try setting that to 200 instead and see if it resolves your issue.

btsimonh, 2020-12-10:
will change that and let you know.  At least I now have monitoring which highlights issues, and all logs stored on disk :).
?? you mean in onConnParamsUpdateRequest I should reject the proposed params?  e.g.:
```
    if(params->itvl_min < 24) { /** 1.25ms units */
      return false;
    } else if(params->itvl_max > 40) { /** 1.25ms units */
      return false;
    } else if(params->latency > 2) { /** Number of intervals allowed to skip */
      return false;
    } else if(params->supervision_timeout > 200) { /** 10ms units */
      return false;
    }
```

h2zero, 2020-12-10:
> some stupid thoughts around this.... I observed notify hitting me before the write function completed in at least two of the previous failures. Now I am only writing to remove this as a failure [...]

There are a few off the shelf devices out there that just start sending notifications as soon as a connection is established, regardless of whether the client subscribed or not.

>you mean in onConnParamsUpdateRequest I should reject the proposed params? e.g.:

    if(params->itvl_min < 24) { /** 1.25ms units */
      return false;
    } else if(params->itvl_max > 40) { /** 1.25ms units */
      return false;
    } else if(params->latency > 2) { /** Number of intervals allowed to skip */
      return false;
    } else if(params->supervision_timeout > 200) { /** 10ms units */
      return false;
    }
>and/or should I change:
pClient->setConnectionParams(12,12,0,51);

I would just use pClient->setConnectionParams(12,12,0,51); and reject the update request from the peripheral if it comes. That should force the disconnect to happen within 510ms.

btsimonh, 2020-12-10:
ok, started another run.  (also with moved m_waitingToConnect = true).

btsimonh, 2020-12-10:
a few more additions pushed in https://github.com/btsimonh/NimBLE-Arduino/tree/btsimonh-test

btsimonh, 2020-12-10:
sorry about this, but it all contributes.  Another BLE hang.  This time, it may highlight the cause:
https://gist.github.com/btsimonh/2a190e696883a6e68633a1053c0947c4

My read of the code indicates that  in this case ble_hs_hci_lock() was not released somewhere.

at line 1619, we enter disconnect() - but we never leave....
But it does tell us it's started disconnect, so must have got as far as calling ble_gap_terminate - but if you check my repo, I have logs in both exits for that, and neither show, plus never << disconnect :(.

Where can we go from there?

btsimonh, 2020-12-10:
and an unhandled exception; interesting one:
https://gist.github.com/btsimonh/da1683dd2c7ab1da10e91284d4fbf489
(line 1700 onwards.   really odd one).

btsimonh, 2020-12-10:
and another :(
https://gist.github.com/btsimonh/de15d7345c3ce99077b5f2d71e59bbba
just go to the end

h2zero, 2020-12-10:
Looks like the controller is hanging. What I think happens is it's command cache gets overwritten so it loses track of the commands. I have seen this type of behavior before and it's very difficult to debug since that is closed source. One of the more recent commits in the controller repo addresses an endless loop bug and it looks like you are hitting that bug. Another possibility is since your ble code is running on core 0 it could be interfering, try changing the core for that task to 1.

btsimonh, 2020-12-10:
I did initially try the dev version of the esp framework, but i think i'm back on the release now.
Started a new run on core 1.....

btsimonh, 2020-12-10:
no banana on core 1.
stalled in WriteValue:
 NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 4c:65:a8:da:f6:07"
D NimBLEAdvertisedDevice: "payload: 09094d4a5f48545f563105030f180a180916fffff37c8b273515"
LE advertising report. len=42 num=1 evtype=3 addrtype=1 addr=34.95.30.bc.30.3e advlen=31 rssi=-56
1e ff 06 00 01 09 20 02 
D NimBLEClient: ">> connect(00:1a:22:09:2c:db)"
D NimBLEScan: ">> stop()"
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
21:46:46 MQT: tele/tasmota_d0 8c 93 54 45 b8 28 c6 
0x0c 0x20 0x02 0x00 0x00 
21:46:46 SL8-22 MI32Command complete: cmd_pkts=5 ogf=0x8 ocf=0xc status=0 
Adv: svcdata70 82d2 2f229 e220 39 c2 db 7e 
1b bb d5 d0 14 f0 85 
D NimBLEScan: "<< stop()"
GAP procedure initiated: connect; peer_addr_type=0 peer_addr=00:1a:22:09:2c:db scan_itvl=16 scan_window=16 itvl_min=12 itvl_max=12 latency=0 supervision_timeout=51 min_ce_len=0 max_ce_len=0 own_addr_type=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000d len=25
0x0d 0x20 0x19 0x10 0x00 0x10 0x00 0x00 0x00 0xdb 0x2c 0x09 0x22 0x1a 0x00 0x00 0x0c 0x00 0x0c 0x00 0x00 0x00 0x33 0x00 0x00 0x00 0x00 0x00 
Command Status: status=0 cmd_pkts=5 ocf=0xd ogf=0x8
21:46:46 SL3-27 Scan ended
21:46:46 SL2-28 MI32: scancomplete
21:46:46 SL1-29 scancompleteCallbacks 0 0
LE connection complete. handle=0 role=0 paddrtype=0 addr=0.1a.22.9.2c.db local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=12 latency=0 spvn_tmo=51 mca=0
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
GATT procedure initiated: exchange mtu
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0x00 0x01 
txed att command: mtu req; conn=0 mtu=256
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x00 0x00 
Command Status: status=0 cmd_pkts=5 ocf=0x16 ogf=0x8
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x05 0x00 0x12 0x77 0x08 0x00 0x28 0x00 0xc8 0x00 0x00 0x00 0x88 0x13 
L2CAP - rxed signalling msg: 0x12 0x77 0x08 0x00 0x28 0x00 0xc8 0x00 0x00 0x00 0x88 0x13 
D NimBLEClient: "Got Client event "
D NimBLEClient: "Peer requesting to update connection parameters"
D NimBLEClient: "MinInterval: 40, MaxInterval: 200, Latency: 0, Timeout: 5000"
D NimBLEClient: "Rejected peer params"
host tx hci data; handle=0 length=10
ble_hs_hci_acl_tx(): 0x00 0x00 0x0a 0x00 0x06 0x00 0x05 0x00 0x13 0x77 0x02 0x00 0x01 0x00 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=0x03 0x00 0x04 0x00 0x03 0x17 0x00 
rxed att command: mtu rsp; conn=0 mtu=23
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteD21:46:46 SL1-29 onConnParamsUpdateRequest 00:1a:22:09:2c:db
escriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"
D NimBLEClient: "<< connect()"
D NimBLEClient: ">> getService: uuid: 3e135142-654f-9090-134a-a6ff5bb77046"
D NimBLEClient: ">> retrieveServices"
GATT procedure initiated: discover service by uuid; uuid=3e135142-654f-9090-134a-a6ff5bb77046
txed att command: find type value req; conn=0 start_handle=0x0001 end_handle=0xffff attr_type=10240
LE Remote Used Features. handle=0 feat=01 00 00 00 00 00 00 00 
host tx hci data; handle=0 length=27
ble_hs_hci_acl_tx(): 0x00 0x00 0x1b 0x00 0x17 0x00 0x04 0x00 0x06 0x01 0x00 0xff 0xff 0x00 0x28 0x46 0x70 0xb7 0x5b 0xff 0xa6 0x4a 0x13 0x90 0x90 0x4f 0x65 0x42 0x51 0x13 0x3e 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
21:46:46 SL2-28 onConnect 00:1a:22:09:2c:db
21:46:46 SL1-29 connected 00:1a:22:09:2c:db - will getservice
21:46:46 MQT: tele/tasmota_esp32/BLE = {"active":{"001A22092CDB":{"n":"CC-RT-M-BLE"},"4C65A8DAF607":{"n":"MJ_HT_V1"},"001A22092FB7":{"n":"CC-RT-M-BLE"},"4C65A8DAF43A":{"n":"MJ_HT_V1"},"6662A3A9B6FC":{"n":""},"43455021FD69":{"n":""},"349530BC303E":{"n":""},"15D80AB7CDB3":{"n":""},"001A22092C9A":{"n":"CC-RT-M-BLE"}}}
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x07 0x00 0x04 0x30 0x04 
rxed att command: find type value rsp; conn=0 
[... 51 lines cut ...]
21:46:47 SL1-29 got service
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=27 data=0x17 0x00 0x04 0x00 0x09 0x15 0x20 0x04 0x1a 0x21 0x04 0x2a 0xeb 0xe0 0xf4 0x90 0x6c 0x41 0xaf 0x96 0x09 0x29 0xcd 0x4d 0x43 0xe8 0xd0 
rxed att command: read type rsp; conn=0 length=21
txed att command: read type req; conn=0 start_handle=0x0421 end_handle=0x0430
host tx hci data; handle=0 length=11
ble_hs_hci_acl_tx(): 0x00 0x00 0x0b 0x00 0x07 0x00 0x04 0x00 0x08 0x21 0x04 0x30 0x04 0x03 0x28 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x08 0x21 0x04 0x0a 
rxed att command: error rsp; conn=0 req_op=8 handle=0x0421 error_code=10
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> writeValue(), length: 1"
GATT procedure initiated: write; att_handle=1041 len=1
txed att command: write req; conn=0 handle=0x0411
host tx hci data; handle=0 length=8
ble_hs_hci_acl_tx(): 0x00 0x00 0x08 0x00 0x04 0x00 0x04 0x00 0x12 0x21:46:47 SL1-29 got read/write characteristic
11 0x04 0x03 
[V][WebServer.cpp:291] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /cs search: c2=160
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.212

btsimonh, 2020-12-10:
ok, found one thing.
You can deleteClient(), and then the disconnect can come in after - my fault for adding 'force' I guess....  when it comes in after the client is gone (which seems to be immediate?), the system crashes because the is not accessible any more.  
but why does deleting the client enable the client event to come in? (immediately)?::

```
 NimBLEClient: "Got Client event "
D NimBLEClient: "Notify Recieved for handle: 1057"
D NimBLEClient: "checking service 3e135142-654f-9090-134a-a6ff5bb77046 for handle: 1057"
D NimBLEClient: "Notification characteristic not found"
D NimBLEClient: ">> disconnect()"
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3
0x06 0x04 0x03 0x00 0x00 0x13 
Command Status: status=0 cmd_pkts=5 ocf=0x6 ogf=0x1
D NimBLEClient: "ble_gap_terminate success: rc=0 "
D NimBLEClient: "<< disconnect()"
22:12:06 SL2-28 operation complete
22:12:06 SL1-29 runTaskDoneOperation: disconnecting connected client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search: m=1
......
[V][Parsing.cpp:231] _parseRequest():  Arguments: m=1
22:12:07 SL1-29 ############################################# wait disconnect 1
22:12:08 SL1-29 ############################################# wait disconnect 2
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=0, "
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

h2zero, 2020-12-10:
deleteClient() will disconnect the client before deleting the client instance. 

Please try using the master branch with the connection parameter changes in your app you did previously. In your disconnect loop do not use isConnected() as that is set before the app callback and before the event returns to the stack. I suggest using a local variable that is set in the app callback to check for connection/disconnection and loop on that.

btsimonh, 2020-12-11:
I've added more logging, including logging of host_rcv_pkt() in src\esp-hci\src\esp_nimble_hci.c

The run this morning ran for 1 hour before failing, but then failed twice more in 20 minutes.  Failures may have changed due to enviromental factors (wife on running machine -> electrical noise)
The failure is absolutely clearly that the disconnect packet does not arrive from the low level drivers:

bad disconnect:
```
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3
0x06 0x04 0x03 0x00 0x00 0x13 
D (4947685) NimBLE: >> host_rcv_pkt type:4 cmd:15
ble_hs_hci_rx_evt type:15 enqueue 0:
D (4947690) NimBLE: << host_rcv_pkt type:4 cmd:15
Command Status: status=0 cmd_pkts=5 ocf=0x6 ogf=0x1
D NimBLEClient: "ble_gap_terminate success: rc=0 SUCCESS"
D NimBLEClient: "<< disconnect()"
(nothing more from BLE for > 7s - at which time I force a restart);
```

Normal disconnect:
```
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
ble_hs_hci_cmd_send: ogf=0x01 ocf=0x0006 len=3
0x06 0x04 0x03 0x00 0x00 0x13 
D (4916034) NimBLE: >> host_rcv_pkt type:4 cmd:15
ble_hs_hci_rx_evt type:15 enqueue 0:
D (4916039) NimBLE: << host_rcv_pkt type:4 cmd:15
Command Status: status=0 cmd_pkts=5 ocf=0x6 ogf=0x1
**D (4916044) NimBLE: >> host_rcv_pkt type:4 cmd:5**
ble_hs_hci_rx_evt type:5 enqueue 1:
**D (4916055) NimBLE: << host_rcv_pkt type:4 cmd:5**
D NimBLEClient: "ble_gap_terminate success: rc=0 SUCCESS"
D NimBLEClient: "<< disconnect()"
Disconnection Complete: status=0 handle=0 reason=22
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"
I NimBLEClient: "disconnect; reason=534, Connection Terminated By Local Host"
```

I'm going to increase the amount of time I wait for the packet to 2 minutes, to see if nimble has a long timeout somewhere.
I think we can assume that because we got "host_rcv_pkt type:4 cmd:15" that the disconnect command got sent.
The fact we never see "D (4916044) NimBLE: >> host_rcv_pkt type:4 cmd:5" I believe means this never hits us at all - I can't think that the low level is stuck somehow such that host_rcv_pkt cannot be called, but I could be wrong.

h2zero, 2020-12-11:
Hi @btsimonh, sorry for your troubles. I've been looking into this as well and as you pointed out in your logs the event `host_rcv_pkt type:4 cmd:5` is missing. 

This event is: **Disconnection Complete event** 
>The HCI_Disconnection_Complete event occurs when a connection is terminated. The status parameter indicates if the disconnection was successful or not. The reason parameter indicates the reason for the disconnection if the disconnection was successful. If the disconnection was not successful, the value of the reason parameter shall be ignored by the Host. For example, this can be the case if the Host has issued the HCI_Disconnect command and there was a parameter error, or the command was not presently allowed, or a Connection_Handle that didn’t correspond to a connection was given. 

This is almost certainly an issue in the controller as it acknowledged the command but did not complete it and trigger the event.

Unfortunately the correct fix is to use a more recent IDF version with an updated controller. I don't think your PR will be merged there as the issue you are targeting only applies to the ESP32, NimBLE has their own controller for the devices they support and will not have this problem.

btsimonh, 2020-12-11:
hmmm.. would that have been in the latest framework?  i.e.
                              platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
I thought the 
D (177683) NimBLE: >> host_rcv_pkt type:4 cmd:5
was actually a packet from the BLE device itself acknowledging disconnect? - i.e. not from the controller....

btsimonh, 2020-12-11:
no, looks like they are stuck in January.
But I can add the new binary and test.

h2zero, 2020-12-11:
That event comes either when the device acknowledges the command or after the supervision timeout that you specified in the connection parameters.

btsimonh, 2020-12-11:
hmmm.. about to scream.  IDF V4.0 would have it, and arduino-esp32.git has a #idf-release/v4.0.  But Tasmota won't have that :( - so can't easily test.  And the binary alone also does not 'fit'.
I don't suppose you fancy 'shimming' the controller until we can use a fixed version! ??? :)  esp_nimble_hci.c is ours, right?  So we can intercept all send traffic, not disconnect request, and timeout there?

btsimonh, 2020-12-11:
hmm..  actually, it's quite a serious issue, and I can't really wait for the next official accepted esp32 release...  so 'fixing' it in esp_nimble_hci.c and making it depend on IDF version is probably a very good option for everyone.
Yep, my environment may be a bit crap; but aren't all BLE environments a bit crap.  every watch and phone and toaster now does something in the spectrum, and add the fact it's shared with wifi, we have to cope with failures. (my BLE fails to connect at 1m range ~10% of the time...  and that failure can be anywhere in the sequence.).
Please let me know if you want me to look at this, or if you would rather grip it.

h2zero, 2020-12-11:
Like I mentioned earlier I still think the primary cause of the trouble is using WiFi at the same time and probably triggering the controller bug. 

I have not witnessed or had reports of this issue until now. In my personal projects using mqtt and BLE I just do the BLE stuff, stop and send the WiFi data after and have no such issues. Though I do not use tasmota for this.

Unless the problem becomes widespread I would prefer not to alter the stack code as it will make maintenance difficult.

Maybe @Staars has some thoughts or experience with this in tasmota?

btsimonh, 2020-12-11:
the Tasmota code currently really only listens to advertisments...  Connection to a sensor is ONLY for battery read, and recommended once per day :(.  So @Staars will have never seen the issue, or if aware avoided it.  

OK. this is how I will play it.  I have a working version.  I'll publish that as a personal Tasmota branch for me to use.
I will make the Tasmota PR as good as it can be with the existing IDF with heavy warnings attached.  It's a big addition to Tasmota, enabling generic BLE control, and simpler specific BLE drivers.
I'll ask you for some minor, heavily tested changes.  give me a few days....

btsimonh, 2020-12-12:
just had a hang in scan->start(20):
```
D NimBLEScan: ">> start(duration=20)"
GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=0 duration=20000ms
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7
0x0b 0x20 0x07 0x01 0x70 0x00 0x50 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xb status=0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
0x0c 0x20 0x02 0x01 0x00 
```
and then nothing more for >120s (at which time I restarted the ESP)... so I think the issue goes deeper than just disconnect, and I don't think BLE is viable on ESP with IDF3.3

If I wanted to test what I'm doing in a low-network env with latest IDF, what project would you point me at?
I can't make Tasmota accept IDF4.2 easily - too many differences.

h2zero, 2020-12-12:
I just had another look at your code and I wonder if this could be caused due to the task priority of your BLE task. Just a thought but if the task priority is 0 as you have it then it's possible other tasks of higher priority are blocking your task from running for some reason. Maybe bump that up to 10 or so and check.

As for alternatives for your project, what do you want it to do? Are you just getting BLE data and sending it over MQTT? There are simpler ways to do this.

btsimonh, 2020-12-12:
"As for alternatives for your project," - was just thinking about a better test env - one which we could both use and so agree on code and results.

I just had a 5 hour run.  One ESP reset because of the scan hangup, 
More than 50 times I reset BLE because it did not disconnect.  I found that when it does not disconnect, it leaves the connection open, so decided to just reset BLE instead in this case :).

I really want this to work in Tasmota, but there is little chance of getting a late IDF into Tasmota any time soon.
SO.  Latest approach is back to square one.  Now we know the effects of the faults in the underlying systems, I'm just going to look for them, and cover them up through BLE restarts and ESP restarts.  It's the best and probably the only way to deal with it for the moment.

I will be proposing some very minor, low impact changes for NimBLE-arduino.
I need the connect() low level return code, so have added m_result and getResult() (called if connect() fails - e.g. through lack of resource - as else we have no idea the system is broken).
And I moved m_waitingToConnect = true; to before the call to the low level, as I definitely saw this set to false by the callback BEFORE being set to true after asking BLE to connect. (maybe because I changed the vTaskDelay value to 1 rather than 1/portTICK_PERIOD_MS => 0, which I still would recommend but not insist on).

To 'fix' a disconnection timeout (after 5s), I am calling 
```
          int conn_id = (*ppClient)->getConnId();
          ble_gap_conn_broken(conn_id, -1);
```
This seems to give you the disconnect callback, and things continue normally after that...
Up to you if you consider something similar within NimBLE-arduino, or doing anything with unrecoverable connect() failure.

(I have just started a run based on current Tasmota development branch including NimBLE-arduino 1.0.2 (with their & my minor mods, IDF3.3 from framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/1.0.4.2/esp32-1.0.4.2.zip, and it's looking good after 30 mins.  multiple fails, but all recovered.)

thanks again for all your attention so far, 
br,  simon.

p.s. just got a esp32-cam module with external +8db antenna.  Gives me 20dB extra on BLE compared to my 'normal' dev module.  With careful location, I may be able to cover the house with just two modules - just wish esp32 with external antenna were easier to find!

btsimonh, 2020-12-12:
pls check branch
https://github.com/btsimonh/NimBLE-Arduino/tree/1.0.2-btsimonh
for the specific changes I would really like :).  Very minor, and no impact on others....

btsimonh, 2020-12-12:
I deal with connect failure here:
https://github.com/btsimonh/Tasmota/blob/2c34353b72c3f371bb5d7d2f31c267820e3601b3/tasmota/xdrv_47_BLE_ESP32.ino#L1722
I'm not sure whether it's strictly necessary with my new approach explained above, but there are certain failures which cannot be recovered from without special attention - the two mentioned were observed.

h2zero, 2020-12-12:
>I need the connect() low level return code, so have added m_result and getResult() (called if connect() fails - e.g. through lack of resource - as else we have no idea the system is broken).

While I can see this being useful for you right now, it's indicating to me that something else is going on. Considering this:

> I moved m_waitingToConnect = true; to before the call to the low level, as I definitely saw this set to false by the callback BEFORE being set to true after asking BLE to connect. 

That is nearly impossible unless a previous disconnect call just returned in the callback at the same time as you are calling connect. Also looking at the return codes you're testing for, that seems to be exactly what is happening. 

>(maybe because I changed the vTaskDelay value to 1 rather than 1/portTICK_PERIOD_MS => 0, which I still would recommend but not insist on).

The change for vTaskDelay is an issue as arduino uses 1000hz tick rate so `1/portTICK_PERIOD_MS` is the correct parameter to use. The issue as brought up in arduino-esp32 only relates to using arduino as an IDF component where the default is 100hz tick rate.

I'm curious about the timing of the events that trigger your issues. I know I can force the failure by simply sending too many commands to the controller in a short period of time (command stack is only 1) and if done from multiple tasks it will hang. I have tested this and it is also why I modified the esp-nimble code to send commands only from tasks on core 0 to prevent such occurrences.

btsimonh, 2020-12-12:
### "That is nearly impossible unless a previous disconnect call just returned" -
no, you set the m_waitingToConnect=true AFTER calling into the low level.  
The low level calls you back, and in your callback, you set it to false..  
If it manages to do this BEFORE you hit your set to true, then the setting it to true AFTER it has been set to false will remain true forever.....
"nearly impossible" - but I observed it.  150Mbytes of logging.  Three weeks of work.  It happened twice.  The mod is harmless if it's not the case, but if it is the case, then it solves an instability.  Personally, I am pleased to have observed it :).

**vTaskDelay** - I defer to you.  so 1/portTICK_PERIOD_MS == 1.  why not use 1 as a minimum delay?  but i don't care :)

**"something else is going on"** - yes.  The low level sometimes (rarely) does not respond correctly. (150 Mbytes of logging, etc., etc.). We hope it would be fixed in the IDF.  But can we leave those who are still on 3.3 hanging with BLE which is compromised?

### Latest run: 
~23 disconnect timeouts corrected using the code I highlighted in 2 1/2 hours.  About 1 connection/disconnection attempted every 5-10 seconds.  About 1/3 or connections or more to devices on verge of range, and so failing with a variety of failures.  No reboots so far.  Looking very good considering.  Expecting reboots triggered by scan hang (nothing we can do about that?) - unless all the diags affected it. (but if it CAN happen with diags on, it CAN happen without?).

h2zero, 2020-12-12:
> no, you set the m_waitingToConnect=true AFTER calling into the low level.
The low level calls you back, and in your callback, you set it to false..

Actually, the stack returns as soon as the command is sent to the controller(non-blocking). By the time we set the flag the controller hasn't even started the receiver. That is set after for a reason (incase of failure, we aren't waiting for a connection if the command failed). I have been using this for nearly 9 months and have not had this issue or seen in reported in all this time, However the code should be changed anyway, I just wanted to point out why I think it could be indicating something.

>vTaskDelay - I defer to you. so 1/portTICK_PERIOD_MS == 1. why not use 1 as a minimum delay? but i don't care :)
The intent is to have a 1ms delay, so we don't loop too fast and stuff up the controller (it breaks easy as you can see lol). Yes we could use 1 for arduino, but this code is also used as a standalone IDF component which could have higher tick rates in config and I'd like to keep the code the same. Makes no difference here anyway :)

I'm curious if you ever see `BLE_ERR_CONN_LIMIT` or `BLE_ERR_ACL_CONN_EXISTS` in your testing. If so, do you have a log with those codes?

btsimonh, 2020-12-12:
"the stack returns as soon as the command is sent to the controller(non-blocking)" -yep, it would have to be BLE_HS_EBUSY to loop - unless the task happened to be schedule out in the time between.  (I don't know IF the task management can swap tasks if you don't yield).

"BLE_ERR_CONN_LIMIT or BLE_ERR_ACL_CONN_EXISTS" - yes.  But before I started calling ble_gap_conn_broken.  i.e. I think my timeout within the lower stack was flawed.  However; if either of these happens at connect, it should be dealt with.  In my case, I should only ever have one connection - so either of these are bad news.  I'm not specifically monitoring the logs for these now - I'll add them and let you know if they happen.

br, simon

h2zero, 2020-12-12:
>I don't know IF the task management can swap tasks if you don't yield).
It task switches on any call to `vTaskDelay()` which is why I was talking about your task priority level earlier, you delay in it and it might not get priority back for a while. 

To be honest, the code you have is quite large and I don't know what tasmota is doing under the hood so I'm speculating a bit here as to where conflicts can happen. If you are getting those return codes that suggests (from what I can see) that a call is being made more than once to `::createClient()` and possibly connecting to the same device twice.

btsimonh, 2020-12-12:
no, it was the way I was timing out the connection in the layer below you (previously), I believe.  I'm pretty sure it can never have two clients or connect twice.

btsimonh, 2020-12-12:
"21:42:23 SL2-28 Connection exists? - restarting NimBLE" - yes.  Twice in 10 minutes.
So... I'm calling ble_gap_conn_broken which lets the client continue.  But obviously does not clear the connection?

btsimonh, 2020-12-12:
I'm pretty sure I just lost a scanended callback :(.  But more checking of my code required.

h2zero, 2020-12-16:
Been thinking about this and I'm almost 100% positive you are somehow connected to a device twice. 

>ok, found one thing.
You can deleteClient(), and then the disconnect can come in after

I have been able to reproduce this behavior exactly by creating 2 connections to a device. Doing what you have stated above will close the connection because `deleteClient()` calls disconnect again.

btsimonh, 2020-12-17:
### Latest results.  
I have an esp32Cam with external antenna which i've left running.
Uptime since last flash is 2 1/2 days.
Total notify operations in that time 10000.
stats of reasons for restarting BLE: (including some from before the flash):
 "blerestart": {
  "connect failed with connection exists": 1171,
  "no adverts in 120s": 3, - I think these may be from before the last flash.
  "connect failed with connection limit reached": 5 - I think these may be from before the last flash.
 },
Stats for operation results:
"state": {
  "DONENOTIFIED": 16618,
  "FAILCONNECT": 6429,
  "FAILNOTIFY": 29,
  "FAILNOTIFYTIMEOUT": 1158,
  "FAILNONOTIFYCHAR": 323,
  "FAILNOSERVICE": 134
 },

it can currently see:
mac	name	RSSI
001A22092C9A	CC-RT-M-BLE	-77
001A22092FB7	CC-RT-M-BLE	-82
001A22092CDB	CC-RT-M-BLE	-58
4C65A8DAF43A	MJ_HT_V1	-65
001A22085B1E	CC-RT-M-BLE	-88
2A958C399F5B		-56
0750408D5F2C		-62

(the CC-RT-M-BLE are the ones it's connecting to, one at a time, with 15s between operations).

Q: would it report connection exists if the DEVICE had an existing connection?  I cannot rule out two ESPs both trying to connect at the same time...

I'm fairly pleased with the stability now, with what I have done on the outside to mitigate 'bad' things.
It would be nice if the code dealt with or gave the ability to deal with these inside.
Memory seems quite stable.

h2zero, 2020-12-17:
>Q: would it report connection exists if the DEVICE had an existing connection? I cannot rule out two ESPs both trying to connect at the same time...

The controller returns that code if it is still synced with the device you're trying to connect to. If 2 clients are connecting at the same time the device should allow both if capable or just the first one it received the request from and the other would hang until timeout.

For a test that may have interesting results what you can do is call `pClient->disconnect()` 2 or more times, maybe with a 50ms delay between each call, and see what happens. Worst case you'll get a few error messages with rc=2.

btsimonh, 2020-12-17:
"I'm almost 100% positive you are somehow connected to a device twice" - -> I'm almost 100% positive that I ditch a connection whilst active, and something remembers that :).

"call pClient->disconnect() 2 or more times" -> disconnect aborts immediately without sending anything to the controller as pClient believes itself connected :(.

h2zero, 2020-12-17:
Here is something interesting regarding this:  https://esp32.com/viewtopic.php?t=4625.

btsimonh, 2020-12-21:
yes, interesting.
I believe I did find an issue with my code where I could re-start scanning before the scan complete callback had finished.  This could be a bad thing, so will keep an eye on performance now that is resolved (now I have the scancomplete event set BLERunningScan = 2 to indicate it's over (rather than 0), and detect that elsewhere, wait 100ms, then allow it to start the scan next loop (1s away).
could it be that that if you call ->start() from outside of the complete callback, bur DURING the complete callback, that the scan class could start another scan and then forget that it did?  - I found this because I was getting multiple complete callbacks (up to 3 at one point!) when I stopped a scan.
br, simon

h2zero, 2020-12-21:
Interesting, I guess it could be possible if using the blocking scan and manually stopping/starting it from a task on another core. I'll have to look at this.

btsimonh, 2020-12-21:
non-blocking in my case.  MUCH more stable now.  I'll see if I can setup another long-running run this evening (i.e. connect one device to a PSU rather than the PC, and leave it running, logging via MQTT the number of restarts, etc.  not seen any in the few minutes that I could monitor manually.)

h2zero, 2020-12-22:
Good that you've found a way to improve this. I was doing some testing and actually stumbled into a similar fault on disconnect. In my case it eventually disconnected but it was nearly 10 minutes after. Sadly it has not happened since so I cannot trace this yet, I will continue to try reproducing.

h2zero, 2020-12-22:
It seems I may have stumbled onto something with this. I have a server device placed in a noisy area that is configured to send **indications *not* notifications** and upon calling disconnect I occasionally get this delay/hang. Just speculating here but after receiving the indication the client sends an ACK to the server on the next connection interval, if we call disconnect before that happens it may trigger this issue. This could also happen with notifications as they are ACK'd but at the link layer. 

So my theory on this is that the controller is waiting for the connection interval event to send and ACK and if we tell it to disconnect before that happens it ignores the command. I will continue testing with my setup as I can now get it to happen every 15 mins or so.

[92 more replies]

closed 2021-01-12 by h2zero