h2zero/NimBLE-Arduino #157: expose "bool response=false" from NimBLEremoteCharacteristics::writeValue in NimBLEClient::setValue(..., bool response=false)

fightforlife, 2020-12-18:
Hi there, 

would it be possible to expose the parameter  `bool response=false"` 
from `NimBLEremoteCharacteristics::writeValue (const std::string &newValue, bool response=false)` 
as an optional (fourth) parameter in 
`NimBLEClient::setValue (const NimBLEUUID &serviceUUID, const NimBLEUUID &characteristicUUID, const std::string &value, bool response=false)`

Some of my devices only work with bool response=true. (e.g. eQ3 ble thermostats)
And this would really help to declutter alot of my code.

Thanks in advance!

h2zero, 2020-12-19:
Sure, that could be added easily enough. If you want to make a PR for it please do so. Otherwise I can add it with the other updates this weekend.

h2zero, 2020-12-24:
@fightforlife this has been added in commit 135e2fa. Please let me know if this works for you.

h2zero, 2020-12-29:
Closing this as resolved.

closed 2020-12-29 by h2zero

fightforlife, 2020-12-29:
Thank you very much. I just had time to test this and it is working as expceted!