h2zero/NimBLE-Arduino #158: Cant make autnetication work.

mblasee, 2020-12-19:
Hi @h2zero,
I am trying to make secure connection between two esp32 devices. My starting points for server and client are the codes from the new user guide. My methodology was to start with an insecure connection and then try to change it to a secure one(with bonding). This one was working as expected, I managed to read the characteristic value("Hello BLE"). Then I tried to add authentication.

In the server I have changed the characteristic to be encrypted(I found it confusing that I have to add both READ and READ_ENC, I have found a discussion about this where you were thinking if you should make it automatic and I think that would be a great idea.)

NimBLECharacteristic *pCharacteristic = pService->createCharacteristic("1234", NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC);
NimBLEDevice::setSecurityAuth(true, false, true);
NimBLEDevice::setSecurityPasskey(123356);  

In the client side I have added 
pClient->secureConnection()

Strange thing is that I am still able to read the characteristic value despite the wrong passkey(123356 in server, 123456 as default in the client). Before uploading the codes I have erased both esp's so any automatically saved bondig infos would clear.

**server log:**
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "mtu update event; conn_handle=0 mtu=256"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onAuthenticationComplete: default"
D NimBLECharacteristic: "Characteristic 0x1234 Read event"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"

**client log:**
D NimBLEScan: "discovery complete; reason=0"

NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClientCallbacks: "onConnect: default"
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
D NimBLEClientCallbacks: "onAuthenticationComplete: default"
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 0xabcd"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 12"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 0x1234, handle: 12 0x0c"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 10 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 10 rc=0"
**Hello BLE**
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"

Could you please help me with this?
The code that I was using can be found here:
https://github.com/mblasee/nimble_client_auth_example/blob/main/src/main.cpp
https://github.com/mblasee/nimble_server_auth_example/blob/main/src/main.cpp

h2zero, 2020-12-19:
Your logs show that the paring is complete (and bonding) where it shows `D NimBLEClientCallbacks: "onAuthenticationComplete: default"`

However you're not getting the passkey request since MITM is off.
You'll need to enable it by changing:
```
NimBLEDevice::setSecurityAuth(true, false, true);
```
to:
```
NimBLEDevice::setSecurityAuth(true, true, true);
```

mblasee, 2020-12-19:
Thanks for the fast response!
Unfortunately this did not work, I have updated that line both in server and client and I am still able to read Hello BLE in client, despite incorrect passkey.  (I have erased flash first)

client log:
D NimBLEScan: "discovery complete; reason=0"
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClientCallbacks: "onConnect: default"
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
D NimBLEClientCallbacks: "onAuthenticationComplete: default"
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 0xabcd"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 12"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 0x1234, handle: 12 0x0c"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 10 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 10 rc=0"
Hello BLE
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"

server log:
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "mtu update event; conn_handle=0 mtu=256"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onAuthenticationComplete: default"
D NimBLECharacteristic: "Characteristic 0x1234 Read event"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"

h2zero, 2020-12-20:
My apologies, I missed something in your code.

On your server application you also need to set the IO capabilities.
```
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
```

That will set it to passkey security.

mblasee, 2020-12-20:
No worries, huge thanks for helping :)
I tried your modifications and sometimes its working sometimes its failing, not sure why. After the first reset cycle looks like it managed to connect but then for some reason it disconnected(I have tried to look up for reason codes but didnt really find anything).
**first try**
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=574, "
D NimBLEClientCallbacks: "onDisconnect: default"
failed to connect
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"

**after a few reset cycle it looks like the client managed to connect but than for some reason it couldn't read out the services and than disconnected**
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Connected
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connection parameters updated."
D NimBLEClient: "Got Client event "
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: "Got Client event "
D NimBLEClient: ">> retrieveServices"
I NimBLEClient: "disconnect; reason=520, "
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: "<< getService: not found"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices

**After another reset cycle it succeeded**
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Connected
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connection parameters updated."
D NimBLEClient: "Got Client event "
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 0xabcd"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 12"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
[... 34 lines cut ...]
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "mtu update event; conn_handle=0 mtu=256"
D NimBLEServer: ">> handleGapEvent: "     
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onAuthenticationComplete: default"
D NimBLECharacteristic: "Characteristic 0x1234 Read event"
D NimBLECharacteristicCallbacks: "onRead: default"        
D NimBLECharacteristicCallbacks: "onRead: default"        
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"

There weren't any code change between these attempts. I am confused why it works sometimes and why not.
I also don't understand why I don't see the line "Client Passkey Request" in client log when I managed to read out the characteristic value from the server. I would expect to see that line, as my understanding that's the part of code what runs when server and client exchanges passkeys. 

If you have any idea what could I do to make debug easier let me know.

h2zero, 2020-12-20:
Hi @mblasee, you can turn on debug return code messages in `nimconfig.h`, just uncomment what you want to see in [this area](https://github.com/h2zero/NimBLE-Arduino/blob/4fb60b8433a4bfb15fefe6fe128d21712f5fce56/src/nimconfig.h#L79). 

The disconnect codes you see there are:
574 = Connection Failed to be Established 
520 = Connection Timeout
546 = LMP Response Timeout / LL Response Timeout

Looks like you're connection is weak, you could try bumping up the power level to compensate.
You can do this by calling:
```
 NimBLEDevice::setPower(ESP_PWR_LVL_P9);
```
That will set them to max power.

Please forgive me, it has been a while since I last dealt with the security stuff. In my previous response I should have stated that you need to add:
```
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY);
```
to your client code. without that the security defaults to "just works"

mblasee, 2020-12-20:
I have changed my client board to another one from a different manufacturer and this seemingly solved the instability issues(lilygo ttgo t1 was the faulty board. I don't have another board from this type so I cant tell if its only one faulty board or there is a general issue with this board family. If I ever order another one I will come back and leave and update here).

These settings seems like completely turned off the passkey part of authentication or I don't know. I am able to connect and read the secured value from the server despite the passkeys are not matching.

```
    NimBLEDevice::setSecurityAuth(true, true, true);
    NimBLEDevice::setSecurityPasskey(123356);
    NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY);
```
client:
```
  uint32_t onPassKeyRequest()
  {
    Serial.println("Client Passkey Request");
    /** return the passkey to send to the server */
    return 123456;
  };
```

client log:
Starting NimBLE Client
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
D NimBLEDevice: ">> setPower: 7 (type: 11)"
D NimBLEDevice: "<< setPower"
D NimBLEDevice: "Setting bonding: 1, mitm: 1, sc: 1"
D NimBLEScan: ">> start(duration=5)"
D NimBLEScan: "<< start()"
_-- lot of nimblescan and advertised devices line here that i removed --_
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Connected
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connection parameters updated."
D NimBLEClient: "Got Client event "
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 0xabcd"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 12"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 0x1234, handle: 12 0x0c"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 10 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 10 rc=0"
Hello BLE
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"

server log:
characteristic
           uuid 0x1234
     def_handle 11
     val_handle 12
   min_key_size 0
          flags [READ|READ_ENC]
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "mtu update event; conn_handle=0 mtu=256"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onAuthenticationComplete: default"
D NimBLECharacteristic: "Characteristic 0x1234 Read event"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLECharacteristicCallbacks: "onRead: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"

h2zero, 2020-12-20:
I think I see the problem now, 
you have in the server code:
```
// client side passkey support
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY); 
```

When it should be this:
```
// server side passkey support
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY); 
```

mblasee, 2020-12-20:
Thank you very much! Passkey auth working as expected now. 
What I still don't understand how this is secure in the end? So my mistake here was that I only set up SecurityIOCap in the server side. I would expect that a request from a misconfigured client would be dropped or whatever but I was able to access the secured resources.
Isnt this making this whole passkey restriction unusable if I can just ignore it with a misconfigure client? Or is there some concept that I miss here?
Also is there an api to remove the bonded device? Right now I am erasing the flash every time. 

with wrong passkey:
D NimBLEScan: "discovery complete; reason=0"
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Connected
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connection parameters updated."
D NimBLEClient: "Got Client event "
D NimBLEClient: "Enter the passkey"
Client Passkey Request
D NimBLEClient: "ble_sm_inject_io result: 0"
D NimBLEClient: "Got Client event "
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 0xabcd"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 12"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 0x1234, handle: 12 0x0c"
I NimBLERemoteCharacteristic: "Read complete; status=261 conn_handle=0"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Enter the passkey"
Client Passkey Request
D NimBLEClient: "ble_sm_inject_io result: 0"
D NimBLEClient: "Got Client event "
E NimBLERemoteCharacteristic: "<< readValue rc=261"

D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"

with correct passkey:
NimBLE
D NimBLEClient: ">> connect(7c:9e:bd:e2:f9:c6)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=256"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Connected
D NimBLEClient: "<< connect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connection parameters updated."
D NimBLEClient: "Got Client event "
D NimBLEClient: "Enter the passkey"
Client Passkey Request
D NimBLEClient: "ble_sm_inject_io result: 0"
D NimBLEClient: "Got Client event "
D NimBLEClient: ">> getService: uuid: 0xabcd"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 0xabcd"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 12"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 0x1234, handle: 12 0x0c"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 10 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 10 rc=0"
Hello BLE
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: "Got Client event "
I NimBLEClient: "disconnect; reason=534, "
D NimBLEClientCallbacks: "onDisconnect: default"
D NimBLEClient: ">> deleteServices"
D NimBLERemoteService: ">> deleteCharacteristics"
D NimBLERemoteCharacteristic: ">> deleteDescriptors"
D NimBLERemoteCharacteristic: "<< deleteDescriptors"
D NimBLERemoteService: "<< deleteCharacteristics"
D NimBLEClient: "<< deleteServices"

I guess rc=261 is the code where we are trying to read a characteristic without permission. Right?

chegewara, 2020-12-20:
I dont know how nimble works, but with bluedroid you are using only one option:
`NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC`
Also when peer device is no capable to connect with pin/passkey then connection will be dropped after 30 seconds and with good configuration you wont be able to read/write any characteristic or descriptor which is protected. Remember that bluetooth specification allows to have some characteristics to be free to read/write without authentication and some which require authenticated on the same device. For the same reason you can retrieve services without authenticate.

> I guess rc=261 is the code where we are trying to read a characteristic without permission. Right?

thats correct.

h2zero, 2020-12-20:
There is an open PR for mynewt/nimble to add the ability to reject connection from clients that do not support passkey authentication but it has not been merged yet. As per bluetooth spec NimBLE will default to "just works" pairing when passkey is not possible. 

You can try adding the flag `NIMBLE_PROPERTY::READ_AUTHEN` to your characteristic, I have not tested it but from my understanding it should prevent reading if not authenticated.

mblasee, 2020-12-20:
I have both READ and READ_ENC becasue looks like NimBLE works this way, https://github.com/h2zero/esp-nimble-cpp/issues/11#issuecomment-643656500
I also tried to use only READ_ENC but that produced the error below:
E NimBLERemoteCharacteristic: "<< readValue rc=258"
Could you please direct me to the direction where can I find these error codes?
I think what I missed here is that just because the communication is encrypted not necessary authenticated.

But it's working perfectly now with the follwing setup.
server:
NimBLEDevice::setPower(ESP_PWR_LVL_P9);
NimBLEDevice::setSecurityAuth(true, true, true);
NimBLEDevice::setSecurityPasskey(123356);
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
NimBLECharacteristic *pCharacteristic = pService->createCharacteristic("1234", NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC | NIMBLE_PROPERTY::READ_AUTHEN);

client:
uint32_t onPassKeyRequest()
{
  Serial.println("Client Passkey Request");
  /** return the passkey to send to the server */
  return 123356;
};

NimBLEDevice::setPower(ESP_PWR_LVL_P9);
NimBLEDevice::setSecurityAuth(true, true, true);
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY);
pClient->secureConnection();

I will create an .ino file and open a PR with the working full examples. 
**Massive thanks for the help!**

h2zero, 2020-12-21:
Glad you got it working 😄 .

>Could you please direct me to the direction where can I find these error codes?

You can find all the return codes [here](https://mynewt.apache.org/latest/network/ble_hs/ble_hs_return_codes.html) (convert decimal to hex first of course) and you can enable the return code text in `nimconfig.h`.

>I think what I missed here is that just because the communication is encrypted not necessary authenticated.

I didn't think of it until you brought it up. NimBLE provides many security options you can see all the property flags in the documentation [here](https://h2zero.github.io/esp-nimble-cpp/md_docs__migration_guide.html#autotoc_md36).

h2zero, 2021-01-08:
@mblasee I'm assuming everything is working still for you, would you like to close this? 

I'm more than welcome to the PR as well if you're still thinking of it.

mblasee, 2021-01-08:
Sorry I got distracted. I just opened the PR and closing this issue. Thanks again!

closed 2021-01-08 by mblasee

xark-10, 2023-06-12:
how to implement the same in esp idf for nimble pass key authentication