h2zero/NimBLE-Arduino #160: Windows "Add Bluetooth Device" allows a second client to pair and causes ESP hangs & crashes

doudar, 2020-12-20:
After a bunch of testing (aka poking until things break) I've found that I can pair a NimBLE server to a Windows computer (right clicking on the bluetooth dock icon and then going through the screen below) even after NimBLE is already connected to another device. That would be fine, but it's causing instability, crashes and hangs.

[image]

**Mostly on 1.02, I get:**
(using https://github.com/doudar/SmartSpin2k)

Bluetooth Client Connected!

23 0 bd 0 0 1 0 aa 2 <-- CPMC sent
23 0 bd 0 0 2 0 54 5 <-- CPMC sent
23 0 bd 0 0 3 0 fe 7 <-- CPMC sent
23 0 bd 0 0 4 0 a8 a <-- CPMC sent
23 0 bd 0 0 eb 0 e 72 <-- 0x2a63 CAD: 90.00
0 47 <-- 0x2a37 HRM: 71
23 0 bd 0 0 ec 0 b8 74 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 ed 0 62 77 <-- 0x2a63 CAD: 90.00
0 46 <-- 0x2a37 HRM: 70
23 0 bd 0 0 ee 0 c 7a <-- 0x2a63 CAD: 90.00
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
Detected HR Disconnect. Trying rapid reconnect
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"

**But with master I've been commonly getting:**

LE connection complete. handle=1 role=1 paddrtype=0 addr=5c.f3.70.9e.7.f1 local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=48 latency=0 spvn_tmo=960 mca=5
Bluetooth Client Connected!
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x01 0x00
23 0 bd 0 0 0 0 0 0 <-- CPMC sent
23 0 bd 0 0 0 0 0 0 <-- CPMC sent
23 0 bd 0 0 0 0 0 0 <-- CPMC sent
23 0 bd 0 0 0 0 0 0 <-- CPMC sent
LE LTK Req. handle=1 rand=00 encdiv=0
looking up our sec; peer_addr_type=0 peer_addr=0xf1 0x07 0x9e 0x70 0xf3 0x5c  ediv=0x00 rand=0x0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x001b len=2
0x1b 0x20 0x02 0x01 0x00
Command complete: cmd_pkts=4 ogf=0x8 ocf=0x1b status=0 handle=1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x00 0x3f
rxed att command: notify req; conn=0 handle=0x0028
0 3f <-- 0x2a37 HRM: 63
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x00 0x3e
rxed att command: notify req; conn=0 handle=0x0028
0 3e <-- 0x2a37 HRM: 62
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, HCI request timed out; controller unresponsive."
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
Detected HR Disconnect. Trying rapid reconnect
assertion "0" failed: file ".pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c", line 1020, function: ble_gap_master_failed
abort() was called at PC 0x40103e93 on core 0

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffd1e40 0x4008ec81:0x3ffd1e60 0x40103e93:0x3ffd1e80 0x400e4d73:0x3ffd1eb0 0x400e5d1f:0x3ffd1f20 0x400e98ac:0x3ffd1f40 0x400e991b:0x3ffd1f60 0x400f2636:0x3ffd1f80 0x400df053:0x3ffd1fa0 0x40093bf9:0x3ffd1fc0

Rebooting...

Decoding stack results
0x4008ea04: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008ec81: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x40103e93: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400e4d73: ble_gap_master_failed at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 1020
0x400e5d1f: ble_gap_reset_state at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 1285
0x400e98ac: ble_hs_reset at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 390
0x400e991b: ble_hs_event_reset at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 538
0x400f2636: nimble_port_run at .pio/libdeps/esp32doit/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 121
0x400df053: NimBLEDevice::host_task(void*) at .pio\libdeps\esp32doit\NimBLE-Arduino\src\NimBLEDevice.cpp line 478
0x40093bf9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

------------------------------------------------------------------------------------------------------------------

Trying to run a scan in this condition causes:

Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 61:74:53:d9:dd:cc, manufacturer data: 4c0010051318b70702, txPower: 12
BLE Advertised Device found: Name: Polar OH1 58D02721, Address: a0:9e:1a:58:d0:27, manufacturer data: 6b007208f4f79d02000000007a0105230041, serviceUUID: 0x180d
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
BLE Advertised Device found: Name: , Address: 24:62:ab:fc:31:66
BLE Advertised Device found: Name: , Address: 35:85:e4:4a:80:95, manufacturer data: 060001092002da1bc9eaf408d634cb6c8a8a1ff25d0d1cec71bf087867
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
BLE Advertised Device found: Name: , Address: 16:2d:74:bb:0f:5e, manufacturer data: 060001092002a624f593de36205c1ecbed21007024fb6ce1b22cd08fbe
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
BLE Advertised Device found: Name: SHIELD, Address: 59:6e:ec:a8:40:e2, manufacturer data: ffff32363838623530363039346661303136, serviceUUID: 0x3456, txPower: -15
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
BLE Advertised Device found: Name: , Address: 16:7b:2d:42:58:17, manufacturer data: 4c0009060320c0a80186
BLE Advertised Device found: Name: , Address: 50:32:37:9e:bc:57, manufacturer data: 4c0010050110df8620, txPower: 12
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
E NimBLEAdvertisedDevice: "Gap Event Parse ERROR."
Initiating Server Connection

h2zero, 2020-12-20:
How many connections are active when this happens? 

The host is getting de-synced from the controller causing a reset. I need to fix up some of the condition handling for that still, it's hard to test it however since it's pretty rare to see.

h2zero, 2020-12-20:
Since you seems to have a good test setup for this, I wonder if you could try the new branch I just pushed?

https://github.com/h2zero/NimBLE-Arduino/tree/Host-reset-handling

doudar, 2020-12-20:
Thanks! I'll test it out a bit later. 

As to your first question, I discovered the issue with 2 connected 2 servers and one connected client (adding the additional windows client caused the issue), but I also tested it with one connected server and the same connected clients.

doudar, 2020-12-20:
Cloning it into my project right now.

doudar, 2020-12-20:
It doesn't seem to be causing the same issue (or any that I'm detecting). Continuing.

doudar, 2020-12-21:
Too soon. 

Looks like it settled into this routine with just one server attached, an iPhone Client and Windows "paired"

23 0 bd 0 0 25 b 92 b0 <-- CPMC sent
23 0 bd 0 0 8f 11 f6 c6 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 90 11 a0 c9 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 91 11 4a cc <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 92 11 f4 ce <-- 0x2a63 CAD: 90.00
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
Reconnected client
Found PM on reconnect
We are now connected to the BLE Server.
23 0 bd 0 0 96 11 9c d9 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 97 11 46 dc <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 98 11 f0 de <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 99 11 9a e1 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 9a 11 44 e4 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 9b 11 ee e6 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 9c 11 98 e9 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 9d 11 42 ec <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 9e 11 ec ee <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 9f 11 96 f1 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a0 11 40 f4 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a1 11 ea f6 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a2 11 94 f9 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a3 11 3e fc <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a4 11 e8 fe <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a5 11 92 1 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a6 11 3c 4 <-- 0x2a63 CAD: 90.00
Bluetooth Client Connected!

23 0 bd 0 0 26 b 3c b3 <-- CPMC sent

23 0 bd 0 0 27 b e6 b5 <-- CPMC sent

23 0 bd 0 0 28 b 90 b8 <-- CPMC sent

23 0 bd 0 0 29 b 3a bb <-- CPMC sent
23 0 bd 0 0 a7 11 e6 6 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a8 11 90 9 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 a9 11 3a c <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 aa 11 e4 e <-- 0x2a63 CAD: 90.00
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
Reconnected client
Found PM on reconnect
[... 161 lines cut ...]
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
Reconnected client
Found PM on reconnect
We are now connected to the BLE Server.
23 0 bd 0 0 c 12 f8 13 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 d 12 a2 16 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 e 12 4c 19 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 f 12 f6 1b <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 10 12 a0 1e <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 11 12 4a 21 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 12 12 f4 23 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 13 12 9e 26 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 14 12 48 29 <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 15 12 f2 2b <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 16 12 9c 2e <-- 0x2a63 CAD: 90.00
23 0 bd 0 0 17 12 46 31 <-- 0x2a63 CAD: 90.00
Bluetooth Client Connected!

doudar, 2020-12-21:
subsequently disconnecting the iPhone client caused this:

CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
9 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Failed to connect to device; " "addr=24:62:ab:fc:31:66, rc=19; "
Reconnect failed
8 left.
CRIT NimBLEDevice: "Resetting state; reason=19, "
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
7 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Failed to connect to device; " "addr=24:62:ab:fc:31:66, rc=19; "
Reconnect failed
6 left.
CRIT NimBLEDevice: "Resetting state; reason=19, "
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
5 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Failed to connect to device; " "addr=24:62:ab:fc:31:66, rc=19; "
Reconnect failed
4 left.
CRIT NimBLEDevice: "Resetting state; reason=19, "
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
3 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Failed to connect to device; " "addr=24:62:ab:fc:31:66, rc=19; "
Reconnect failed
2 left.
CRIT NimBLEDevice: "Resetting state; reason=19, "
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
1 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -58
E NimBLEClient: "Error: Failed to connect to device; " "addr=24:62:ab:fc:31:66, rc=19; "
Reconnect failed
0 left.

doudar, 2020-12-21:
I sort of wish I could get a backtrace, but this error just won't cause a kernel panic. Sorry.

h2zero, 2020-12-21:
That's perfect! I wanted to prevent the panic first because the library should handle host resets gracefully, so hopefully that bug is now fixed.

As to why it's happening for you I'm unsure at this point. I have to try a setup with a server and clients simultaneously to see if I can reproduce (which I can't do at the moment). If you could provide more logs at debug level I might spot something.

doudar, 2020-12-21:
Well the workaround is "just don't use windows OS bluetooth pairing". There's really no reason to unless you're setting up a HID device.  

I just stumbled onto it causing issues. 

The biggest issue I could see is that an outsider could possibly break the ESP32 bluetooth by simply trying to pair to your ESP32 with their windows OS since windows will occasionally display the NimBLE server as an available device to pair, even when its already paired to something else. And once it's paired in Windows, it will forever corrupt that ESP32 until it's either out of range or "disconnect" is selected in windows.

h2zero, 2020-12-21:
Yes it's a very good catch, one that I cannot test as my windows machine doesn't support BLE. This is certainly something that needs to be addressed though, I don't know what windows does that may cause this.

As for the issue of not being able to reconnect I would need to see a log at debug level so I can see if the host gets synced or not. I have more adjustments ready to push that will solve the issue if it does re-sync.

doudar, 2020-12-21:
on it boss...give me a bit. Working on a few different projects. I'm guessing you know exactly how that goes :)

h2zero, 2020-12-21:
Lol, you bet I do. No hurry, I don't think theres millions of devices failing because of this (or is there?) 😄 .

Thanks for the help!

doudar, 2020-12-21:
Crap...you're going to be mad at me.....

This was with bluetooth paired in windows and happened the instant Zwift (windows) also tried to connect. 

assertion "0" failed: file ".pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c", line 1020, function: ble_gap_master_failed
abort() was called at PC 0x40103edb on core 0

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffd1e30 0x4008ec81:0x3ffd1e50 0x40103edb:0x3ffd1e70 0x400e4dbb:0x3ffd1ea0 0x400e5d67:0x3ffd1f10 0x400e98f4:0x3ffd1f30 0x400e9963:0x3ffd1f50 0x400f267e:0x3ffd1f70 0x400df07f:0x3ffd1f90 0x40093bf9:0x3ffd1fb0

Rebooting...

Decoding stack results
0x4008ea04: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008ec81: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x40103edb: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400e4dbb: ble_gap_master_failed at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 1020
0x400e5d67: ble_gap_reset_state at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 1285
0x400e98f4: ble_hs_reset at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 390
0x400e9963: ble_hs_event_reset at .pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 538
0x400f267e: nimble_port_run at .pio/libdeps/esp32doit/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 121
0x400df07f: NimBLEDevice::host_task(void*) at .pio\libdeps\esp32doit\NimBLE-Arduino\src\NimBLEDevice.cpp line 480
0x40093bf9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Before and after log coming in the next message.

doudar, 2020-12-21:
6 0 <-- 0x2a37 HRM: 0
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x69 0x03 0xba 0x15 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 69 3 ba 15 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x6a 0x03 0x64 0x18 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 6a 3 64 18 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x06 0x00 
rxed att command: notify req; conn=1 handle=0x0028
6 0 <-- 0x2a37 HRM: 0
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x6b 0x03 0x0e 0x1b 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 6b 3 e 1b <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x6c 0x03 0xb8 0x1d 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 6c 3 b8 1d <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x6d 0x03 0x62 0x20 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 6d 3 62 20 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x6e 0x03 0x0c 0x23 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 6e 3 c 23 <-- 0x2a63 CAD: 90.00
LE connection complete. handle=2 role=1 paddrtype=0 addr=5c.f3.70.9e.7.f1 local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=48 latency=0 spvn_tmo=960 mca=5
Bluetooth Client Connected!
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x02 0x00 

23 0 0 0 0 4c 1 78 74 <-- CPMC sent 

23 0 0 0 0 4d 1 22 77 <-- CPMC sent 

23 0 0 0 0 4e 1 cc 79 <-- CPMC sent 

23 0 0 0 0 4f 1 76 7c <-- CPMC sent 
ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x06 0x00 
rxed att command: notify req; conn=1 handle=0x0028
6 0 <-- 0x2a37 HRM: 0
ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x06 0x00 
rxed att command: notify req; conn=1 handle=0x0028
6 0 <-- 0x2a37 HRM: 0
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x6f 0x03 0xb6 0x25 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 6f 3 b6 25 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x70 0x03 0x60 0x28 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 70 3 60 28 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x06 0x00 
rxed att command: notify req; conn=1 handle=0x0028
6 0 <-- 0x2a37 HRM: 0
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x71 0x03 0x0a 0x2b 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 71 3 a 2b <-- 0x2a63 CAD: 90.00

23 0 0 0 0 50 1 20 7f <-b-
le_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x72 0x03 0xb4 0x2d 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 72 3 b4 2d <-- 0x2a63 CAD: 90.00
LE LTK Req. handle=2 rand=00 encdiv=0
looking up our sec; peer_addr_type=0 peer_addr=0xf1 0x07 0x9e 0x70 0xf3 0x5c  ediv=0x00 rand=0x0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x001b len=2
0x1b 0x20 0x02 0x02 0x00 
Command complete: cmd_pkts=4 ogf=0x8 ocf=0x1b status=0 handle=2
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, HCI request timed out; controller unresponsive."
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
Detected HR Disconnect. Trying rapid reconnect
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
[... 94 lines cut ...]
ac fb a8 e5 14 29 4a 28 
28 d2 9f 7a 46 4e 51 1b 
2e 41 60 e9 b8 b8 72 
BLE Advertised Device found: Name: , Address: 1b:b1:42:bb:71:98, manufacturer data: 060001092002acfba8e514294a2828d29f7a464e511b2e4160e9b8b872
LE advertising report. len=31 num=1 evtype=2 addrtype=0 addr=a0.9e.1a.58.d0.27 advlen=20 rssi=-69
02 01 04 10 ff 6b 00 72 
08 f4 f7 9d 02 00 00 00 
00 3b 00 00 
LE advertising report. len=31 num=1 evtype=4 addrtype=0 addr=a0.9e.1a.58.d0.27 advlen=20 rssi=-69
13 09 50 6f 6c 61 72 20 
4f 48 31 20 35 38 44 30 
32 37 32 31 
BLE Advertised Device found: Name: Polar OH1 58D02721, Address: a0:9e:1a:58:d0:27, manufacturer data: 6b007208f4f79d02000000003b0000
LE advertising report. len=28 num=1 evtype=0 addrtype=0 addr=50.32.37.9e.bc.57 advlen=17 rssi=-76
02 01 1a 02 0a 0c 0a ff 
4c 00 10 05 01 10 df 86 
20 
LE advertising report. len=33 num=1 evtype=0 addrtype=0 addr=24.62.ab.fc.31.66 advlen=22 rssi=-64
02 01 06 07 03 18 18 26 
18 0d 18 0a 09 53 6d 61 
72 74 54 65 73 74 
LE advertising report. len=14 num=1 evtype=4 addrtype=0 addr=24.62.ab.fc.31.66 advlen=3 rssi=-64
02 0a 03 
BLE Advertised Device found: Name: SmartTest, Address: 24:62:ab:fc:31:66, serviceUUID: 0x1818, txPower: 3
LE advertising report. len=28 num=1 evtype=0 addrtype=1 addr=44.f8.2b.57.d9.63 advlen=17 rssi=-89

doudar, 2020-12-21:
I''ll try the exact same thing and see if it's repeatable

doudar, 2020-12-21:
yep. 100% repeatable:

LE connection complete. handle=2 role=1 paddrtype=0 addr=5c.f3.70.9e.7.f1 local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=48 latency=0 spvn_tmo=960 mca=5
Bluetooth Client Connected!
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x02 0x00 

23 0 0 0 0 f 0 f6 27 <-- CPMC sent 

23 0 0 0 0 10 0 a0 2a <-- CPMC sent 

23 0 0 0 0 11 0 4a 2d <-- CPMC sent 

23 0 0 0 0 12 0 f4 2f <-- CPMC sent 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x16 0x06 0x9c 0x36 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 16 6 9c 36 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x00 0x4f 
rxed att command: notify req; conn=1 handle=0x0028
0 4f <-- 0x2a37 HRM: 79
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x17 0x06 0x46 0x39 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 17 6 46 39 <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x18 0x06 0xf0 0x3b 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 18 6 f0 3b <-- 0x2a63 CAD: 90.00
ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x1b 0x28 0x00 0x00 0x4f 
rxed att command: notify req; conn=1 handle=0x0028
0 4f <-- 0x2a37 HRM: 79
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x00 0x00 0x00 0x19 0x06 0x9a 0x3e 
rxed att command: notify req; conn=0 handle=0x0010
23 0 0 0 0 19 6 9a 3e <-- 0x2a63 CAD: 90.00
LE LTK Req. handle=2 rand=00 encdiv=0
looking up our sec; peer_addr_type=0 peer_addr=0xf1 0x07 0x9e 0x70 0xf3 0x5c  ediv=0x00 rand=0x0 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x001b len=2
0x1b 0x20 0x02 0x02 0x00 
Command complete: cmd_pkts=4 ogf=0x8 ocf=0x1b status=0 handle=2
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, HCI request timed out; controller unresponsive."
Bluetooth Client Disconnected!
E NimBLEAdvertising: "Host reset, wait for sync."
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
Detected HR Disconnect. Trying rapid reconnect
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
assertion "0" failed: file ".pio\libdeps\esp32doit\NimBLE-Arduino\src\nimble\host\src\ble_gap.c", line 1020, function: ble_gap_master_failed
abort() was called at PC 0x40103edb on core 0

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffd1e40 0x4008ec81:0x3ffd1e60 0x40103edb:0x3ffd1e80 0x400e4dbb:0x3ffd1eb0 0x400e5d67:0x3ffd1f20 0x400e98f4:0x3ffd1f40 0x400e9963:0x3ffd1f60 0x400f267e:0x3ffd1f80 0x400df07f:0x3ffd1fa0 0x40093bf9:0x3ffd1fc0

Rebooting...
xΓÉÇΓÉÇ∩┐╜xΓÉÇΓÉÇΓÉÇΓÉÇxxxx∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇx∩┐╜ΓÉÇ∩┐╜∩┐╜ΓÉÇxΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇ∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇx∩┐╜xxx∩┐╜ΓÉÇΓÉÇxxxxxΓÉÇΓÉÇxxΓÉÇxΓÉÇxxΓÉÇΓÉÇxxΓÉÇxΓÉÇxΓÉÇΓÉÇx∩┐╜x∩┐╜ΓÉÇ∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇxΓÉÇxxΓÉÇxxΓÉÇΓÉÇxΓÉÇxΓÉÇxΓÉÇxxxΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇxΓÉÇxΓÉÇΓÉÇxxΓÉÇΓÉÇxΓÉÇxΓÉÇΓÉÇxxΓÉÇxxΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇx∩┐╜x∩┐╜xΓÉÇxΓÉÇ∩┐╜xΓÉÇxΓÉÇxΓÉÇ∩┐╜ΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇ∩┐╜ΓÉÇxΓÉÇxxΓÉÇxxΓÉÇxΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxxxxx∩┐╜ΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇ∩┐╜xxxxΓÉÇΓÉÇxΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇxxΓÉÇΓÉÇxΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇxxΓÉÇΓÉÇxΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇxΓÉÇΓÉÇΓÉÇ∩┐╜xΓÉÇΓÉÇxΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇ∩┐╜ΓÉÇΓÉÇxxΓÉÇΓÉÇxΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇx∩┐╜∩┐╜ΓÉÇxxΓÉÇΓÉÇxΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇx∩┐╜x∩┐╜ΓÉÇΓÉÇxxx∩┐╜ΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇxΓÉÇΓÉÇxΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇ∩┐╜x∩┐╜xΓÉÇxxΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇ∩┐╜x∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜x∩┐╜xΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇ∩┐╜xxx∩┐╜x∩┐╜ΓÉÇΓÉÇ∩┐╜ΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜x∩┐╜xΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇxΓÉÇ∩┐╜xxx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇ∩┐╜ΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜x∩┐╜xΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇ∩┐╜xxx∩┐╜x∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇ∩┐╜ΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜x∩┐╜xΓÉÇΓÉÇΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇ∩┐╜xxx∩┐╜x∩┐╜ΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇ∩┐╜∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇxΓÉÇΓÉÇΓÉÇxΓÉÇx∩┐╜ΓÉÇ∩┐╜xΓÉÇxΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇΓÉÇ∩┐╜ΓÉÇΓÉÇxΓÉÇx∩┐╜ΓÉÇΓÉÇΓÉÇCompiled Dec 20 202017:13:24
Mounting Filesystem
Reading File: /config.txt
Config File Loaded: /config.txt
Contents of file: /config.txt
{"firmwareUpdateURL":"https://raw.githubusercontent.com/doudar/OTAUpdates/main/","incline":0,"simulatedWatts":0,"simulatedHr":60,"simulatedCad":90,"deviceName":"SmartSpin2K","shiftStep":400,"stepperPower":1000,"inclineMultiplier":2,"simulatePower":false,"simulateHr":true,"ERGMode":false,"autoUpdate":true,"ssid":"SmartSpin2K","password":"password","connectedPowerMeter":"SmartTest","connectedHeartMonitor":"Polar OH1 58D02721"} 
Writing File: /config.txt
equest
 read:current=
equest
Setting up cpu Tasks
Starting Arduino BLE Client application...
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0003 len=0
0x03 0x0c 0x00

doudar, 2020-12-21:
I may not test anymore tonight...Beer :30       =D

h2zero, 2020-12-21:
That's awesome, thanks! It's definitely beer:30 🥂.  I think I have enough there to track things down, looks like windows is trying to initiate security immediately and for some reason that bugs out the controller. Also looks like the controller gets stuck in a loop and dies in your previous logs.

I'll see what I can do.

h2zero, 2020-12-21:
BTW the assert gets called because you had `CONFIG_BT_NIMBLE_DEBUG` enabled. 

If you leave that enabled and comment out [this line](https://github.com/h2zero/NimBLE-Arduino/blob/569eb8a188c78fe780f4c2a24cf9247532cf55ea/src/nimble/host/src/ble_gap.c#L1020) there might be more info to find the failure. 

Go have a beer or 6 though, there's enough info here for now. Much appreciated!

doudar, 2020-12-21:
I'm not seeing the hard crash with that line commented out. It does still seem to have some problems for a bit with the device paired & Zwift connected but looks like it recovers. I haven't run it long enough to determine if it settles into the pattern like it did above (that will be after the honey-do list :). I'll attach the full log so you can sift through. Search for "host reset". Somewhere above the first instance of "Target Incline" is where Zwift would have additionally connected after being previously paired in Windows OS. 

[SmartSpin_logfile.txt](https://github.com/h2zero/NimBLE-Arduino/files/5725192/SmartSpin_logfile.txt)

h2zero, 2020-12-21:
Thanks for that, I'm just thinking out loud here but I believe there could be a matter of too many "connections". If you have a server with a client connected and also connected to devices on your 2 client instances, then begin advertising and another client connects to your server that could cause this (assuming max connections = 3). 
Maybe try bumping the max connections to 5 or 6?

The log was very useful, I can see things are resetting correctly. Also looks like windows tries to encrypt the connection immediately which is where the host gets reset. I'm not sure why that would happen but it could just be a symptom of the connection limits exceeded.

doudar, 2020-12-21:
Trying it now. I've been meaning to ask - what's the difference between MAX_CONNECTIONS and MAX_BONDS?

h2zero, 2020-12-21:
`MAX_CONNECTIONS` is just the number of device connections (also advertising counts as a connection).
`MAX_BONDS` is just the number of peers you can bond i.e the number of devices bonding info stored in flash.

doudar, 2020-12-21:
Increasing MAX_CONNECTIONS to 4 didn't help. In fact this time it didn't pass any characteristic information to Zwift (just the service). I'll try MAX_BONDS next, then should I try increasing CONFIG_BT_NIMBLE_TASK_STACK_SIZE?

Here's the full log again. 

[SmartSpin_logfile.txt](https://github.com/h2zero/NimBLE-Arduino/files/5725565/SmartSpin_logfile.txt)

doudar, 2020-12-21:
MAX_CONNECTIONS = 4 & MAX_BONDS = 4

Complete controller hang but no panic or crash. (I didn't try accessing it via http but otherwise it appears none of the BLE calls returned. 

Log for your pleasure. (you have weird pleasures :D )
[SmartSpin_logfile.txt](https://github.com/h2zero/NimBLE-Arduino/files/5725595/SmartSpin_logfile.txt)

doudar, 2020-12-21:
MAX_CONNECTIONS = 4 & MAX_BONDS = 4 & CONFIG_BT_NIMBLE_TASK_STACK_SIZE 6096

Same as above. I tested the HTTP this time, ESP32 is responsive but BLE calls don't return.

[SmartSpin_logfile.txt](https://github.com/h2zero/NimBLE-Arduino/files/5725612/SmartSpin_logfile.txt)

doudar, 2020-12-21:
> `MAX_CONNECTIONS` is just the number of device connections (also advertising counts as a connection).
> `MAX_BONDS` is just the number of peers you can bond i.e the number of devices bonding info stored in flash.

If advertising counts as a connection, is that per service or just per device? I'm advertising 3 services.  Should I try increasing MAX_CONNECTIONS further?

doudar, 2020-12-21:
MAX_CONNECTIONS = 5 & MAX_BONDS = 3 & CONFIG_BT_NIMBLE_TASK_STACK_SIZE 4096

Controller doesn't lock up but same as connections = 3

[SmartSpin_logfile.txt](https://github.com/h2zero/NimBLE-Arduino/files/5725649/SmartSpin_logfile.txt)

doudar, 2020-12-21:
MAX_CONNECTIONS = **9** & MAX_BONDS = 3 & CONFIG_BT_NIMBLE_TASK_STACK_SIZE 4096

Same result.
[SmartSpin_logfile.txt](https://github.com/h2zero/NimBLE-Arduino/files/5725675/SmartSpin_logfile.txt)

LE connection complete. handle=2 role=1 paddrtype=0 addr=5c.f3.70.9e.7.f1 local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=48 latency=0 spvn_tmo=960 mca=5
Bluetooth Client Connected!
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x02 0x00

23 0 98 0 0 1 0 aa 2 <-- CPMC sent

23 0 98 0 0 2 0 54 5 <-- CPMC sent

23 0 98 0 0 3 0 fe 7 <-- CPMC sent

23 0 98 0 0 4 0 a8 a <-- CPMC sent
LE LTK Req. handle=2 rand=00 encdiv=0
looking up our sec; peer_addr_type=0 peer_addr=0xf1 0x07 0x9e 0x70 0xf3 0x5c  ediv=0x00 rand=0x0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x001b len=2
0x1b 0x20 0x02 0x02 0x00
Command complete: cmd_pkts=4 ogf=0x8 ocf=0x1b status=0 handle=2
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x04 0x00 0x1b 0x10 0x00 0x23 0x00 0x98 0x00 0x00 0x26 0x14 0x3c 0xad

doudar, 2020-12-21:
If you make any changes to that branch, let me know and I'll have to pull the library to my machine again.

h2zero, 2020-12-21:
I've spent some time digging through the core files and came up empty on this. It must be related to the security code but without being able to reproduce locally it's quite difficult to trace.

From your logs I can see it loads a bonded peer device, I wonder if erasing the flash and re flashing has any effect?

At least the host reset handling seems better in that branch, I will clean it up and merge it.

doudar, 2020-12-22:
It's probably best to sit on this one for a while.

If the one of the BLE architects can't figure it out, it may be for the best :) 

If you *really* wanted me to look at this closely I have a ESP32 JTAG debugger. I probably won't have time for the in the next day though. Maybe sometime later this week if you can give me an idea on what to look for/monitor and where to put break points. .

h2zero, 2020-12-22:
Well, there are many factors against us at work here. 

Since arduino-esp32 is still using IDF 3.2/3.3, and the issue could be caused by the closed source controller, it might already be fixed in IDF latest versions. So raising the issue there would seem moot at this time.

Also, as we are using an espressif adapted fork of NimBLE intended for use with newer IDF versions. So we are also limited in resources for resolving these types of issues if they are controller related. I have had a couple PR's merged upstream but it was all host related and unless we can prove the cause is from the host side there isn't much we can do.

JTAG would be great, if you want to play around I could point out a few spots. Breakpoints can be their own issues though.

doudar, 2020-12-22:
I really don't care if this gets fixed, so the priority is all yours. 

I guess maybe we could setup a video chat (i could be your hands) and have you remote in to a guest session on my computer **if** it is something you'd like to investigate.

h2zero, 2020-12-22:
That's and interesting thought but probably not necessary. The main concern was recovery for these errors, which appears to work so at least things don't fail completely. 

It's just my OCD that wants to figure it out lol, I hate when things that should work don't and I can't understand why (closed source GRRR). Maybe something else comes up in the future that solves the issue, or the IDF version is updated in arduino (coming soon I think) and resolves it. I will keep this issue in mind though, we should leave it open in case others have a similar experience.

h2zero, 2022-04-05:
Closing as resolved. Please reopen if not.

closed 2022-04-05 by h2zero