h2zero/NimBLE-Arduino #161: Scan duration inaccurate

kerimt-dek, 2020-12-23:
Hi,
I use this simple sketch based on BLE_scan example  to test BLE range for couple of my devices

`/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <NimBLEDevice.h>
#include <NimBLEAdvertisedDevice.h>

int scanTime = 5; //In seconds
uint32_t freeMem;
BLEScan* pBLEScan;

char* deviceMacAddress = "B4:52:A9:28:8A:CB";

char* sensorMacAddress =   "db:fe:5d:bf:c4:24";  // SN007

BLEAddress BTTAddress(deviceMacAddress);
BLEAddress WLTSAddress(sensorMacAddress);

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice* advertisedDevice) {
      if (advertisedDevice->getAddress() == BTTAddress) {
        Serial.printf("BTT RSSI: %d \n", advertisedDevice->getRSSI());
      }

      if (advertisedDevice->getAddress() == WLTSAddress) {
        Serial.printf("WLTS RSSI: %d \n", advertisedDevice->getRSSI());
      }

    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEDevice::setPower(ESP_PWR_LVL_P9);

  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {

  // put your main code here, to run repeatedly:
  Serial.println("--------------- start scan -------------------------");
  BLEScanResults foundDevices =  pBLEScan->start(scanTime, false);
  Serial.println("--------------- scan done ------------------------\n");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  //delay(2000);
}

`

Most of the time actual duration is 5 seconds, but intermittently it takes 15 seconds, as you can see from this log

`17:01:33.677 -> Scanning...
17:01:33.935 -> --------------- start scan -------------------------
17:01:38.926 -> --------------- scan done ------------------------
17:01:38.926 -> 
17:01:38.926 -> --------------- start scan -------------------------
17:01:43.935 -> --------------- scan done ------------------------
17:01:43.935 -> 
17:01:43.935 -> --------------- start scan -------------------------
17:01:58.930 -> --------------- scan done ------------------------
17:01:58.930 -> 
17:01:58.930 -> --------------- start scan -------------------------
17:02:03.941 -> --------------- scan done ------------------------
17:02:03.941 -> 
17:02:03.941 -> --------------- start scan -------------------------
17:02:18.936 -> --------------- scan done ------------------------`

h2zero, 2020-12-24:
Well that doesn't seem right, any chance you could provide a more detailed log, with debug level set?

kerimt-dek, 2020-12-24:
I've tried with different debug levels, but interestingly I wasn't able to reproduce. As soon as I switch to debug level "none" it starts happening and it's quite often. 
Have you tried at your end?

7:11:34.172 -> --------------- start scan -------------------------
17:11:39.200 -> --------------- scan done ------------------------
17:11:39.200 -> --------------- start scan -------------------------
17:11:54.198 -> --------------- scan done ------------------------
17:11:54.198 -> --------------- start scan -------------------------
17:11:59.241 -> --------------- scan done ------------------------
17:11:59.241 -> --------------- start scan -------------------------
17:12:14.185 -> --------------- scan done ------------------------
17:12:14.185 -> --------------- start scan -------------------------
17:12:19.202 -> --------------- scan done ------------------------
17:12:19.202 -> --------------- start scan -------------------------
17:12:34.178 -> --------------- scan done ------------------------
17:12:34.178 -> --------------- start scan -------------------------
17:12:39.183 -> --------------- scan done ------------------------
17:12:39.183 -> --------------- start scan -------------------------
17:12:54.203 -> --------------- scan done ------------------------
17:12:54.203 -> --------------- start scan -------------------------
17:13:09.206 -> --------------- scan done ------------------------
17:13:09.206 -> --------------- start scan -------------------------
17:13:24.186 -> --------------- scan done ------------------------
17:13:24.186 -> --------------- start scan -------------------------

kerimt-dek, 2020-12-24:
This is what happens when device is found, looks like scan start is delayed by 10 seconds and this is typical. Whenever actual scan duration is 15 seconds device is found around 10 seconds after scan start
17:18:24.371 -> --------------- start scan -------------------------
17:18:35.512 -> BTT RSSI: -93 
17:18:39.364 -> --------------- scan done ------------------------
17:18:39.364 -> --------------- start scan -------------------------
17:18:44.354 -> --------------- scan done ------------------------
17:18:44.354 -> --------------- start scan -------------------------

h2zero, 2020-12-24:
hi @kerimt-dek I have tested this, using your code, so far have been unable to reproduce the issue. I added a time test that will stop the loop if the time between scan start and stop is > 6 seconds. I will let it run for a a few hours and see if the issue presents itself.

kerimt-dek, 2020-12-24:
I was suspecting it has to do with my setup, but I still don't know why. 
My arduino version is 1.8.13 on macOS big sur.

h2zero, 2020-12-24:
I have been running this for over 6 hours with no issues. I think you are correct in that there is something going on with your setup, I don't know what that would be though.

kerimt-dek, 2020-12-25:
Thanks, I will let you know if I ever find out what was the issue

closed 2020-12-25 by kerimt-dek