h2zero/NimBLE-Arduino #162: Compilation "warnings"

tonton81, 2020-12-27:
After a week on BLEClient testing constant reconnects by forcing disconnects every 7 seconds to simulate reconnects over time, BLEClient would always stop working after several reconnects, leaving the loop() locked.

As simple as NimBLE was to "adapt" over, I just changed the header, and my code still worked as is.
`//#include "BLEDevice.h"
#include "NimBLEDevice.h"
`
Now running the reconnect tests, surprisingly so far it is not stopping... Still gonna run that overtime to see if it behaves any different over more reconnects.

Anyways, during compilation with NimBle, I get these errors:
`C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\BLE_DATA.ino: In function 'void auto_connect_ble()':
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\BLE_DATA.ino:41:15: warning: unused variable 'r' [-Wunused-variable]
       uint8_t r = random(1, 255);
               ^
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\BLE_DATA.ino:42:15: warning: unused variable 'g' [-Wunused-variable]
       uint8_t g = random(1, 255);
               ^
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\BLE_DATA.ino:43:15: warning: unused variable 'b' [-Wunused-variable]
       uint8_t b = random(1, 255);
               ^
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino: In function 'void XBT_power(int, bool)':
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino:3:137: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
     if ( XBT_client[controller]->isConnected() ) XBT_characteristic[controller]->writeValue(controller_switch, sizeof(controller_switch));
                                                                                                                                         ^
In file included from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteService.h:25:0,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEClient.h:27,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:31,
                 from C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\ESP32_BTLE_MORIMOTO-2.ino:13:
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:115:52: note: candidate 1: bool NimBLERemoteCharacteristic::writeValue(const uint8_t*, size_t, bool)
     bool                                           writeValue(const uint8_t* data,
                                                    ^
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:126:10: note: candidate 2: bool NimBLERemoteCharacteristic::writeValue(const T&, bool) [with T = unsigned char [2]]
     bool writeValue(const T &s, bool response = false) {
          ^
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino: In function 'void XBT_setColor(int, uint8_t, uint8_t, uint8_t, int, int, int)':
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino:43:147: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
     if ( XBT_client[controller]->isConnected() ) XBT_characteristic[controller]->writeValue(XBT_colors[controller], sizeof(XBT_colors[controller]));
                                                                                                                                                   ^
In file included from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteService.h:25:0,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEClient.h:27,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:31,
                 from C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\ESP32_BTLE_MORIMOTO-2.ino:13:
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:115:52: note: candidate 1: bool NimBLERemoteCharacteristic::writeValue(const uint8_t*, size_t, bool)
     bool                                           writeValue(const uint8_t* data,
                                                    ^
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:126:10: note: candidate 2: bool NimBLERemoteCharacteristic::writeValue(const T&, bool) [with T = unsigned char [20]]
     bool writeValue(const T &s, bool response = false) {
          ^
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino: In function 'void sync_XBT_controllers(int)':
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino:87:102: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
     XBT_characteristic[controller]->writeValue(XBT_colors[controller], sizeof(XBT_colors[controller]));
                                                                                                      ^
In file included from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteService.h:25:0,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEClient.h:27,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:31,
                 from C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\ESP32_BTLE_MORIMOTO-2.ino:13:
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:115:52: note: candidate 1: bool NimBLERemoteCharacteristic::writeValue(const uint8_t*, size_t, bool)
     bool                                           writeValue(const uint8_t* data,
                                                    ^
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:126:10: note: candidate 2: bool NimBLERemoteCharacteristic::writeValue(const T&, bool) [with T = unsigned char [20]]
     bool writeValue(const T &s, bool response = false) {
          ^
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino: In function 'void smooth_color_switching(int, int, int, int, int, int)':
C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\XBT_Controls.ino:135:151: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
         if ( XBT_client[controller]->isConnected() ) XBT_characteristic[controller]->writeValue(XBT_colors[controller], sizeof(XBT_colors[controller]));
                                                                                                                                                       ^
In file included from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteService.h:25:0,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEClient.h:27,
                 from C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLEDevice.h:31,
                 from C:\Users\Tony\Documents\Arduino\ESP32_BTLE_MORIMOTO-2\ESP32_BTLE_MORIMOTO-2.ino:13:
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:115:52: note: candidate 1: bool NimBLERemoteCharacteristic::writeValue(const uint8_t*, size_t, bool)
     bool                                           writeValue(const uint8_t* data,
                                                    ^
C:\Users\Tony\Documents\Arduino\libraries\NimBLE-Arduino\src/NimBLERemoteCharacteristic.h:126:10: note: candidate 2: bool NimBLERemoteCharacteristic::writeValue(const T&, bool) [with T = unsigned char [20]]
     bool writeValue(const T &s, bool response = false) {
          ^
`

Other than that, it does compile, and upload fine.

h2zero, 2020-12-27:
Looks like the compiler is complaining about the data types in your arrays. You might need to manually cast these to `uint8_t*` to clear the warnings or just remove the `sizeof(XBT_colors[controller]` and the template will sort it out for you.

>Interesting, after a longer run with NimBLE, disconnect() stops working after awhile, I noticed that the clients don't cut out if i cut their power

This is something that has come up recently and I have done a lot of testing. What happens is the response to disconnect from the peripheral device for one reason or another does not get received so the controller waits for the supervision timeout (+ a few ms) while waiting for the response. When that time expires it will send the disconnect event at that time. As you have seen, if you set the connection parameters with a lower supervision timeout this happens faster.

tonton81, 2020-12-27:
thanks, its working really well in terms of reconnects and BLE client disconnects, however I cant get getService to work with either 128 or 16 bit UUID, so ot never finalized the connection to write anything, been at it for awhile last night and put BLEClient back to make sure I didnt kill my BLE devices lol and they connect and write fine there

closed 2020-12-28 by tonton81