h2zero/NimBLE-Arduino #163: h2zero works too hard closing issues and needs a beer :)

doudar, 2020-12-29:
See title^^

Seriously, are you even human? :)

h2zero, 2020-12-29:
I can drink beer and close issues at the same time 🥂 😆.

Just going through things to remind myself what needs to be done for the upcoming release, good time to cleanup too.

doudar, 2020-12-29:
Nicely done![image]

I'm printing new parts I designed today and drinking beer and watching you close issues. Enjoy your night!

h2zero, 2020-12-29:
Haha, awesome. Can't say I've seen that beer but I'd buy it if I did! 

Thanks for the reminder, I need to print a few things as well.

doudar, 2020-12-29:
Me too. Had to. 

If you ever need an excuse to print parts, head over to my smart spin. Lots of stuff to print, lots of code that needs more contributors.

h2zero, 2020-12-29:
Actually I need more excuses not to 😄 I will check out the repo for sure!

doudar, 2020-12-29:
Haha! Exactly!

closed 2020-12-29 by doudar

chegewara, 2020-12-29:
This is the way BLE library has been "outsourced" from Kolbans repo ;)