h2zero/NimBLE-Arduino #164: NimBLEScan : NimBLEAdvertisedDevice refactor?

h2zero, 2020-12-29:
Hi all, currently the advertised device reporting from NimBLEScan continuously parses and stores the data, also if scan is continuing indefinitely, each found device will consume resources.

I propose 3 things to make this more efficient:
- 1 If the payload from the advertised device hasn't changed since last seen, only the RSSI should be updated.
- 2 The payload data should be stored and not parsed immediately, instead it would be parsed only when the application asks for specific data from it. Doing this will allow the advertising packet + scan data to be stored more efficently (62 bytes max per device) and allow the `getPayload()` function to provide the complete data.
- 3 Adding a limit to the number of advertised devices stored by adding a parameter to `NimBLEScan::start()`. I'm thinking it could have a default value of 0xFF == unlimited and 0x00 == store none (callback only), any other value would be the limiter. No change to app code needed unless the functionality is used.

Thoughts? Ideas? Opinions?

justoke, 2020-12-29:
Hi. Firstly, thank you for developing this version. It literally saved my project as the original lib is just too big for a OTA use case on ESP32. My use case would benefit from these changes - I have a controller based on the ESP32 which has BLE devices linked by means of a list of linked devices - this is a logical link and saved in a configuration file on the SPIFFs file system. I don't want to pair the devices but rather scan for devices - some devices like leak sensors only broadcast events like alarm and heartbeat, so only appear every 10 minutes at most or in alarm conditions. So typically I just check if the device address matches one in my list and depending on the type of device I get the payload and decode as per the specific manufacturer data definition. If the device is not of interest I just exit at that point. 

I see a number of advantages to option 1 - this will be helpful as I currently check the decoded value for change to decide if the controller should do something, nothing if the data hasn't changed, so this would save this extra step. Option 2 just makes sense, again only if the device is in my list do I care about the payload and actually want to parse and decode it. Option 3 is ideal as I know how many linked devices I have.

Sample code
```
void TaskManagerService::onResult(BLEAdvertisedDevice* advertisedDevice) {
  // Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());

  bool linkedDevice = false;
  LinkedDeviceType typeDevice = LinkedDeviceType::UNKNOWN;

  BLEAdvertisedDevice* myTempSensor;
  BLEAdvertisedDevice* myLeakDetector;

  // Get the linked device, check if this is the advertised device
  _statsService->read([&](ConfigStatsState& state) {
    linkedDevice = state.deviceCheck(String(advertisedDevice->getAddress().toString().c_str()));
    // Serial.printf("linkedDevice found: %s\r", linkedDevice ? "Yes" : "No");

    if (linkedDevice) {
      typeDevice = state.deviceType(String(advertisedDevice->getAddress().toString().c_str()));
    }
  });

  if (!linkedDevice)
    return;

  if (linkedDevice && typeDevice == LinkedDeviceType::TEMP) {
    myTempSensor = advertisedDevice;

    uint8_t* buf1 = myTempSensor->getPayload();
.........
```

chegewara, 2020-12-29:
ad 3) seems reasonable,
ad 1) it may be problematic in case when peer device is using private address (most devices, excluding sensors), because after some time it will lead to the same situation, but not bad idea,

h2zero, 2020-12-29:
>ad 1) it may be problematic in case when peer device is using private address (most devices, excluding sensors), because after some time it will lead to the same situation, but not bad idea,

Yes, random addresses would still be picked up as new devices when the address changes. Not much that can be done about that other than bonding with it once if possible and resolving it in the scan result.

I will put these ideas to test in a new branch scan-advertisedDevice-rework for anyone that would like to test, not much has been done yet though.

doudar, 2020-12-29:
In my project I don't care about 98% of the devices I scan for. It would be pretty cool if I could just add the UUID's to the NimBLE scan in which I wish to be returned, maybe like we do to setup advertising. This might be a way to free up resources.

chegewara, 2020-12-29:
> In my project I don't care about 98% of the devices I get in a scan. It would be pretty cool if I could just add the UUID's to the NimBLE scan in which I wish to be returned, maybe like we do to setup advertising. This might be a way to free up resources.

But you know you can do it very quick and easy in your code?
If advertised device does not match your UUID then delete that device.

doudar, 2020-12-29:
> But you know you can do it very quick and easy in your code?
> If advertised device does not match your UUID then delete that device.

Yes, that's exactly what I do.

 I'm just postulating that If it's consuming resources before it even gets to that point, maybe it would be better implemented upstream?

justoke, 2020-12-29:
> Yes, that's exactly what I do.
>
> I'm just postulating that If it's consuming resources before it even gets to that point, maybe it would be better implemented upstream?

Please could you share the details on this - this is exactly what I need to do too. I didn't realize this could be done.

doudar, 2020-12-29:
> Please could you share the details on this - this is exactly what I need to do too. I didn't realize this could be done.

When you initiate a scan, create a callback that checks the characteristics against your criteria and sets a connect flag if it finds a device matching your criteria. Such as:

`pClient->setClientCallbacks(new MyClientCallback(), true);`

That callback's "onresult" gets initiated for every device NimBLE finds, so then inside the callback do something like:

void MyAdvertisedDeviceCallback::onResult(BLEAdvertisedDevice *advertisedDevice)
{
    debugDirector("BLE Advertised Device found: " + String(advertisedDevice->toString().c_str()));
    const char *c_PM = userConfig.getconnectedPowerMeter();
    const char *c_HR = userConfig.getconnectedHeartMonitor();
    if ((advertisedDevice->haveServiceUUID()) && (advertisedDevice->isAdvertisingService(CYCLINGPOWERSERVICE_UUID)))
    {
        if ((advertisedDevice->getName() == c_PM) || (advertisedDevice->getAddress().toString().c_str() == c_PM) || (String(c_PM) == ("any")))
        {
            spinBLEClient.myPowerMeter = advertisedDevice;
            spinBLEClient.doConnectPM = true;
            return;
        }
    }
    if ((advertisedDevice->haveServiceUUID()) && (advertisedDevice->isAdvertisingService(HEARTSERVICE_UUID)))
    {
        if ((advertisedDevice->getName() == c_HR) || (advertisedDevice->getAddress().toString().c_str() == c_HR) || (String(c_HR) == ("any")))
        {
            spinBLEClient.myHeartMonitor = advertisedDevice;
            spinBLEClient.doConnectHR = true;
            return;
        }
    }
}

h2zero, 2021-01-04:
Pushed an update to the scan-advertisedDevice-rework if anyone would like to test it. Full payload available from getPayload and dynamic parsing implemented, not fully tested yet.

h2zero, 2021-01-27:
I created a new branch since it was easier than messing around squashing commit's. Please test out my PR #181 aka branch refactor-advertised-device for the latest iteration. I would appreciate any bug reports / feedback.

lbussy, 2021-01-31:
Has this been merged?  [refactor-scan](https://github.com/h2zero/NimBLE-Arduino.git#refactor-scan) seems to have gone away.

h2zero, 2021-01-31:
Wow you're quick lol, yes it was just merged now. Please post back here if you encounter any issues.

lbussy, 2021-01-31:
I *just* happened to be doing my "hourly" and recompiling. :)

h2zero, 2021-01-31:
Beware, more to come :) but It'll be in a branch 😄

h2zero, 2021-02-01:
>Beware, more to come, but It'll be in a branch 😄

This was a lie, pushed an update to master just now. 

I think most of the bases in this proposal are covered now, please report any issues.

h2zero, 2021-02-11:
Closing as implementation is complete.

closed 2021-02-11 by h2zero