h2zero/NimBLE-Arduino #166: Arduino Framework 2.1.0 compile issue

doudar, 2021-01-07:
I don't see this anywhere yet. I just went to do a fresh build with no changes to my code and received this after updating to the latest Arduino framework. 

Compiling .pio\build\esp32doit\lib8d1\NimBLE-Arduino\NimBLEDevice.cpp.o
In file included from .pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.cpp:8:0:
.pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.h:72:31: error: 'RingbufferType_t' has not been declared
     Ringbuffer(size_t length, RingbufferType_t type = RINGBUF_TYPE_NOSPLIT);
                               ^
.pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.cpp:268:39: error: 'RingbufferType_t' has not been declared
 Ringbuffer::Ringbuffer(size_t length, RingbufferType_t type) {
                                       ^
.pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.cpp: In constructor 'Ringbuffer::Ringbuffer(size_t, int)':
.pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.cpp:272:48: error: invalid conversion from 'int' to 'ringbuf_type_t' [-fpermissive]
     m_handle = ::xRingbufferCreate(length, type);
                                                ^
In file included from .pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.h:14:0,
                 from .pio\libdeps\esp32doit\NimBLE-Arduino\src\FreeRTOS.cpp:8:
C:/Users/Anthony/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_ringbuf/freertos/ringbuf.h:69:17: note:   initializing argument 2 of 'void* xRingbufferCreate(size_t, ringbuf_type_t)'
 RingbufHandle_t xRingbufferCreate(size_t xBufferSize, ringbuf_type_t xBufferType);
                 ^
*** [.pio\build\esp32doit\lib8d1\NimBLE-Arduino\FreeRTOS.cpp.o] Error 1
======================================================================================= [FAILED] Took 48.35 seconds =======================================================================================

doudar, 2021-01-07:
Nevermind... a fresh pull of NimBLE fixed it. That was weird.

closed 2021-01-07 by doudar

h2zero, 2021-01-07:
Yes I recently fixed that in master branch. I need to get a release out soon before that becomes a problem for others as well.

doudar, 2021-01-07:
It seems PlatformIO is pushing the 2.1.0 update (as default) as of this
afternoon so the reports may roll in soon.

h2zero, 2021-01-07:
Damn, thanks for the heads up! At least most platformio users know how to change library dependency location. I'm mostly concerned with the arduino IDE users in this respect.