h2zero/NimBLE-Arduino #167: Panic reset when scanning and several concurrent http requests occur

pletch, 2021-01-07:
Working with others on a project which leverages NimBLE as a client only receiving advertisements (ADV_NONCONN_IND) from up to 8 devices.  We have recently converted the project over to use ESPAsyncWebserver and have started experiencing sporadic PANIC resets.  A reset can be fairly reliably triggered by hard reloading one of the pages with several concurrent jQuery requests repeatedly.  We suspect this relates to competition for radio time between WiFi and BLE but are unsure about this aspect. Have done some troubleshooting such as trying to adjust the Scan Interval and Scan Window but have not been able to reliably eliminate the issue and am looking for ideas.  The backtrace is always the same and appears to indicate that the PANIC is being triggered in NimBLE related code.

Any guidance on whether this likely points to a problem with our implementation or whether we have uncovered a library issue is much appreciated!

**Logging output:**
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
0x0c 0x20 0x02 0x00 0x00
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xc status=0 
GAP procedure initiated: discovery; own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=0 duration=3000ms
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7
0x0b 0x20 0x07 0x00 0x54 0x00 0x3b 0x00 0x00 0x00 
Command complete: cmd_pkts=5 ogf=0x8 ocf=0xb status=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
0x0c 0x20 0x02 0x01 0x00

Backtrace: 0x40091fa4:0x3ffbdc20 0x400921d5:0x3ffbdc40 0x40085a9d:0x3ffbdc60 0x40085bc9:0x3ffbdc90 0x40180377:0x3ffbdcb0 0x4017ad15:0x3ffbdf70 0x40177de9:0x3ffbdfc0 0x400ee0e2:0x3ffbdff0 0x401c75ea:0x3ffbe010 0x401bba2a:0x3ffbe030 0x4001791d:0x3ffbe050 0x401bc029:0x3ffbe070 0x401bc106:0x3ffbe0b0 0x400188f5:0x3ffbe0e0 0x400175ee:0x3ffbe120 0x40017688:0x3ffbe140 0x4008cb36:0x3ffbe170 0x4008cd0b:0x3ffbe190 0x401bb9d7:0x3ffbe1b0 0x4008cd28:0x3ffbe1d0 0x40085025:0x3ffbe1f0 0x4008583e:0x3ffb9d00 0x4008cadd:0x3ffb9d20 0x401c7789:0x3ffb9d40 0x401aa088:0x3ffb9d60 0x400edd8d:0x3ffb9d80 0x40082b53:0x3ffb9da0 0x4008e33d:0x3ffb9dc0
  #0  0x40091fa4:0x3ffbdc20 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x400921d5:0x3ffbdc40 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x40085a9d:0x3ffbdc60 in lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
  #3  0x40085bc9:0x3ffbdc90 in _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
  #4  0x40180377:0x3ffbdcb0 in _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
  #5  0x4017ad15:0x3ffbdf70 in fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
  #6  0x40177de9:0x3ffbdfc0 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
  #7  0x400ee0e2:0x3ffbdff0 in host_rcv_pkt at .pio\libdeps\lcd_ssd1306\NimBLE-Arduino@src-e248a3b4ca528fb2d5b4751077939ebd\src\esp-hci\src/esp_nimble_hci.c:365
  #8  0x401c75ea:0x3ffbe010 in vhci_send at ??:?
  #9  0x401bba2a:0x3ffbe030 in r_eif_send at ??:?
  #10 0x4001791d:0x3ffbe050 in ?? ??:0
  #11 0x401bc029:0x3ffbe070 in hci_tx_start at hci_tl.c:?
  #12 0x401bc106:0x3ffbe0b0 in r_hci_tl_send at ??:?
  #13 0x400188f5:0x3ffbe0e0 in ?? ??:0
  #14 0x400175ee:0x3ffbe120 in ?? ??:0
  #15 0x40017688:0x3ffbe140 in ?? ??:0
  #16 0x4008cb36:0x3ffbe170 in vhci_recv_end at vhci.c:?
  #17 0x4008cd0b:0x3ffbe190 in r_vhci_isr at ??:?
  #18 0x401bb9d7:0x3ffbe1b0 in r_eif_isr at ??:?
  #19 0x4008cd28:0x3ffbe1d0 in r_eif_isr_wrapper at intc.c:?
  #20 0x40085025:0x3ffbe1f0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #21 0x4008583e:0x3ffb9d00 in cause_sw_intr_to_core_wrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c:1322
  #22 0x4008cadd:0x3ffb9d20 in vhci_set_interrupt at vhci.c:?
  #23 0x401c7789:0x3ffb9d40 in API_vhci_host_send_packet at ??:?
  #24 0x401aa088:0x3ffb9d60 in esp_vhci_host_send_packet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c:1322
  #25 0x400edd8d:0x3ffb9d80 in ble_hci_trans_hs_cmd_tx_on_core_0 at .pio\libdeps\lcd_ssd1306\NimBLE-Arduino@src-e248a3b4ca528fb2d5b4751077939ebd\src\esp-hci\src/esp_nimble_hci.c:171
  #26 0x40082b53:0x3ffb9da0 in ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62
  #27 0x4008e33d:0x3ffb9dc0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

doudar, 2021-01-07:
I can only add that I fought the exact same issue for a couple weeks with espasyncwebserver and the old Bluetooth library. I ended up dropping back to the regular webserver and using client side JavaScript to smooth things over.

h2zero, 2021-01-07:
Well I cannot say for sure what is happening here but I found this interesting, in 7 [this assert](https://github.com/h2zero/NimBLE-Arduino/blob/bc52eac5618c45bcfc00b1004bc25b3b1709f0ae/src/esp-hci/src/esp_nimble_hci.c#L365) is triggered. This is a message from the controller that it has encountered a hardware error. 

Not much to work with here, except perhaps a power issue. If not that, then perhaps a bug in the controller, guessing by the previous area of the backtrace (who knows here, it's black magic). 

One other thing it could be that could be tested however, is the commands being sent from core 0. This is a modification made to the esp-nimble files to address delay issues when sending commands from core 1. If you revert [this commit](https://github.com/h2zero/NimBLE-Arduino/commit/c35e5fde8bf94638e8680b318c7ded62934f6d71#diff-daeddae4d22f2593460986d69beb34f09a7cb2fb2cb8171b0510e70ded33c7b2) and test I'd be **very curious** of the results.

justoke, 2021-01-07:
Hi. I have run into the same issue running an http server using ESPAsyncWebserver and scanning BLE - my understanding is that this is hardware problem in that the WiFi and BLE share the same hardware radio. Meaning that you have a harder software problem to balance scanning BLE for devices and servicing Wifi packets as they share the data pipeline. I've not found a solution yet and may have to make a separate controller for BLE scanning as the ESP32 crashes often or the http server takes as long to respond as the shortest scanning cycle. It seems while BLE scan is in progress you can't service http requests. 

I'll share my findings if I manage to come up with a workable approach.

justoke, 2021-01-07:
> Backtrace: 0x40091fa4:0x3ffbdc20 0x400921d5:0x3ffbdc40 0x40085a9d:0x3ffbdc60 0x40085bc9:0x3ffbdc90 0x40180377:0x3ffbdcb0 0x4017ad15:0x3ffbdf70 0x40177de9:0x3ffbdfc0 0x400ee0e2:0x3ffbdff0 [...]

I've put that through an Exception Decoder. It looks like its related to the WiFi data, DNS and so on - I get the same sort of issues. If I find a workable approach I'll share it.
[image]

doudar, 2021-01-07:
Nimble works perfectly fine with me with all of these libraries, even during scan. It's just the espasync . Which sucks because it is a nice webserver. You can however spin off the regular webserver below on its own task and it works reasonably well. Good luck!

#include <WebServer.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>
#include <SPIFFS.h>
#include <ESPmDNS.h>
#include <WiFiClientSecure.h>
#include <Update.h>
#include <DNSServer.h>

justoke, 2021-01-07:
Someone did mention power supply maybe needing a capacitor to deal with extra power needed during scan. Has anyone tried the setActiveScan with false or observed anything related to this setting and stability with Wifi and BLE. I'll try this myself anyway and see if this helps.

[image]

lbussy, 2021-01-07:
I'm one of the unnamed co-conspirators.  

> Not much to work with here, except perhaps a power issue.

I can help rule-out the power unless it's a design flaw in the dev board itself (Lolin D32 Pro.)  There's three of us working on this with similar results, it seems unlikely all of us have flakey power.  I'd also expect to see a reset caused by brownout, not a panic.  Unless you see something I do not?

> Has anyone tried the setActiveScan with false or observed anything related to this setting and stability with Wifi and BLE.

`setActiveScan` is set to false.  We are just using the iBeacon in this project.

@justoke which exception decoder are you using?  I've had horrible results trying to consistently use those, but that would be a nice thing to figure out.  I am using the esp32_exception_decoder built in to PIO and I get this which does not show the same information:

```
Backtrace: 0x40091978:0x3ffbe190 0x40091ba9:0x3ffbe1b0 0x40121f58:0x3ffbe1d0 0x400847ed:0x3ffbe1f0 0x401c4ecf:0x3ffb5cf0 0x401c65c5:0x3ffb5d60 0x401c3adb:0x3ffb5db0 0x401c106e:0x3ffb5e00 0x40019d11:0x3ffb5e40 0x40055b4d:0x3ffb5e60 0x401bb1e7:0x3ffb5e80 0x401bb731:0x3ffb5ea0 0x4008dd45:0x3ffb5ed0
  #0  0x40091978:0x3ffbe190 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x40091ba9:0x3ffbe1b0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x40121f58:0x3ffbe1d0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x400847ed:0x3ffbe1f0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x401c4ecf:0x3ffb5cf0 in r_llm_le_adv_report_ind at ??:?
  #5  0x401c65c5:0x3ffb5d60 in r_llm_pdu_defer at ??:?
  #6  0x401c3adb:0x3ffb5db0 in r_lld_pdu_check at ??:?
  #7  0x401c106e:0x3ffb5e00 in r_lld_evt_deffered_elt_handler at ??:?
  #8  0x40019d11:0x3ffb5e40 in ?? ??:0
  #9  0x40055b4d:0x3ffb5e60 in ?? ??:0
  #10 0x401bb1e7:0x3ffb5e80 in r_rw_schedule at ??:?
  #11 0x401bb731:0x3ffb5ea0 in btdm_controller_task at ??:?
  #12 0x4008dd45:0x3ffb5ed0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

> I ended up dropping back to the regular webserver

Rolling back to use the synchronous web server is an undesirable outcome.  The performance difference between the two is quite pronounced for this application.  Unless I inline all of the code (undesirable from a re-use and maintenance standpoint), a single connection would be quite bad.

> One other thing it could be that could be tested however, is the commands being sent from core 0. This is a modification made to the esp-nimble files to address delay issues when sending commands from core 1. If you revert this commit and test I'd be very curious of the results.

I noticed the core - the web server (async specifically) is pinned to core 1 per [this note](https://github.com/me-no-dev/ESPAsyncWebServer/issues/731#issuecomment-628163515).  Removing that makes things quite a bit worse,  I did attempt to pin NimBLE to core 1 as well with no improvement.

Reverting that commit has proved challenging for me, and I'm sure it's something I am forgetting but I don't see it:

```
git log --oneline  
bc52eac (grafted, HEAD -> master, origin/master, origin/HEAD) Fix attributes not found with 16/32bit UUIDs.
```

I appreciate the folks here willing to help out.  If it's something we need to bring upstream we can obviously do that.  It's just quite a bit more challenging to work there.

justoke, 2021-01-07:
I think this might resolve your issue on ESP32. Add this define you your code - in main.cpp if you are using Platform IO
#define CONFIG_ASYNC_TCP_RUNNING_CORE 1

This will pin the AsyncWebserver which runs as a FreeRTOS task on CORE 1. I tried this and although the web server can still get blocked by the BLE(apparent because the web server response times are longer when the scan is in progress), it has not crashed since I tried this and often it responds in sub 100ms times which is as good as I could hope for. I implemented some other things specific to my use case - like blocking my loop() task logic while updating data from the BLE onResult() event.

lbussy, 2021-01-07:
> I think this might resolve your issue on ESP32.

Yes, sir - already doing that.  My comment about it was not horribly clear in retrospect.  I've edited my post above.

justoke, 2021-01-07:
@lbussy I used this decode the exceptions https://github.com/me-no-dev/EspExceptionDecoder. Although I mainly use PIO, this is the only thing I use Arduino IDE  for.

justoke, 2021-01-07:
> > I think this might resolve your issue on ESP32.
> 
> Yes, sir - already doing that. My comment about it was not horribly clear in retrospect. I've edited my post above.

I had replied before I saw your comment. I'm happy to try help you. I was definitely in a world of pain before I made the core assignment. Of course my use case will differ from yours, but maybe there is an approach to what I have done that might help you.

lbussy, 2021-01-07:
Thanks - I really hate going back to the Arduino IDE after PIO.

So here's something interesting.  I have a fork of ESPAsynchWebServer I was using elsewhere and I pulled that in instead.  The behavior is much improved (but not perfect).  Here's the PR I merged in my fork:

https://github.com/me-no-dev/ESPAsyncWebServer/pull/836/files

As you can see the difference is basically cleaning up connections.  It's not making perfect sense yet as I've not had my coffee - but you are 5 hours ahead so maybe it strikes a chord with you.

lbussy, 2021-01-07:
> I was definitely in a world of pain before I made the core assignment. Of course my use case will differ from yours, but maybe there is an approach to what I have done that might help you.

I agree - and we're using that:

```
    -D CONFIG_ASYNC_TCP_RUNNING_CORE=1  ; per: https://github.com/me-no-dev/ESPAsyncWebServer/issues/731#issuecomment-628163515
    -D CONFIG_ASYNC_TCP_USE_WDT=1
```

So aside from the [PR up there](https://github.com/me-no-dev/ESPAsyncWebServer/pull/836/files) having a net positive impact, if I remove the `CONFIG_ASYNC_TCP_USE_WDT=1` define I get no crashes, but eventually the webserver stops responding.

We're poking in the right place for sure.

h2zero, 2021-01-07:
> I can help rule-out the power unless it's a design flaw in the dev board itself (Lolin D32 Pro.) There's three of us working on this with similar results, it seems unlikely all of us have flakey [...]

If the regulator is weak, which it may be on this board, the voltage might dip low enough to cause an error but not low enough to trigger the brownout detection (2.43v default in arduino). So if your regulator dropped to, say, 2.9v, that might be enough to trigger this. Just speculating here though, you would have to measure with a scope or meter to detect min levels.

Hopefully your PR there will lead to a software resolution though.

doudar, 2021-01-07:
> Thanks - I really hate going back to the Arduino IDE after PIO.
>
>
>
> So here's something interesting.  I have a fork of ESPAsynchWebServer I was using elsewhere and I pulled that in instead.  The [...]

FYI, you can still work in PIO and just use the trace debugger 
In Arduino IDE separately. When you select it from the menu, it asks for your firmware.elf file (grab it from your .pio folder) and paste the debug message into the window that pops up.

justoke, 2021-01-07:
> I agree - and we're using that:
>
> ```
>     -D CONFIG_ASYNC_TCP_RUNNING_CORE=1  ; per: https://github.com/me-no-dev/ESPAsyncWebServer/issues/731#issuecomment-628163515
>     -D CONFIG_ASYNC_TCP_USE_WDT=1 [...]

So I've had 5 hrs of up time on my ESP32 and not a single panic crash. The basic app use case is as follows:
Scan BLE every 2 seconds and read sensor values from detected beacons/sensors
Manage thermostat task when sensor readings change
When sensor  values changes a flag is set and this prevents further BLE scanning until the thermostat task has responded and then the state machine repeats the cycle. 
The loop() function simply services the web server - again while scan results in sensor changes, the web server is not serviced.

To test the web api response, I fired up Postman and fired requests at 500 millisecond intervals for a few minutes. What is apparent is during the BLE scan response time goes up to 4s, otherwise typical response is sub 200 millisecond with 80 to 90 ms being fairly average.

lbussy, 2021-01-07:
Stopping the BLE scans while servicing the jquery (which are JSON requests) is a possibility.  That might help.  Thank you for your ideas - there might be something in there we can use!

I wonder if the AsyncWebServer can be paused during BLE scans - sort of the opposite.

lbussy, 2021-01-07:
> Hopefully your PR there will lead to a software resolution though.

He has 40 open PRs going back to 2017.  I'm not going to hold my breath.

justoke, 2021-01-07:
> Stopping the BLE scans while servicing the jquery (which are JSON requests) is a possibility. That might help. Thank you for your ideas - there might be something in there we can use!
> 
> I wonder if the AsyncWebServer can be paused during BLE scans - sort of the opposite.

I do this - so when BLE onResult() event finds a matching device that I am after, it reads the sensor data and sets HasChanges to true. In that state I don't process web requests until the changes have been processed by the relevant observers. In practice this doesn't take long and the stability gained is worth the slightly slower request response.

[image]

justoke, 2021-01-07:
> > Hopefully your PR there will lead to a software resolution though.
> 
> He has 40 open PRs going back to 2017. I'm not going to hold my breath.

I'm using the latest version and haven't had to rely on any PRs which is preferred in my case as I don't want to have to keep track of such nuances in my projects

lbussy, 2021-01-07:
Yeah, that would be fine if I was synchronous, I'm not.  I hope to *not* have to go back to the other webserver.  If I can somehow "pause" the async, that might help.

doudar, 2021-01-07:
Are you able to completely shut down the async webserver then restart it?
Last I tried, it led to a kernel panic but that may have been my error.

pletch, 2021-01-07:
I reverted the mentioned commit and it did not appear to make much difference as both myself and my fellow conspirators continue to experience the PANIC.   Thank you for providing the suggestion.  We'll continue to brainstorm ideas.

> One other thing it could be that could be tested however, is the commands being sent from core 0. This is a modification made to the esp-nimble files to address delay issues when sending commands [...]

lbussy, 2021-01-07:
> Are you able to completely shut down the async webserver then restart it?

I do this in a different project (`ESPAsyncWebServer::stop()` if I remember correctly) and it works fine.  It seems heavy-handed here, it takes a sec to start and stop, so I'm more looking for a "hold up a sec" method. :)

h2zero, 2021-01-07:
Thanks for letting me know. 

Could I ask what your scan interval and window values are? I know you've tried adjusting them I'm just curious.if I get time tonight I'll load up my async server BLE project and see if I can reproduce.

lbussy, 2021-01-07:
Here's our init:

```
void tiltScanner::init()
{
    NimBLEDevice::init("");
    pBLEScan = NimBLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(callbacks);

    pBLEScan->setActiveScan(false);
    pBLEScan->setInterval(53);
    pBLEScan->setWindow(37);
}

h2zero, 2021-01-08:
Those values look pretty tight in reality if you are sharing the antenna with a webserver. 

You've got 20ms open for the webserver to do its job between scan windows. If your latency to the server is more than that you'll have issues. Try increasing the interval by 200, you'll get less responses on BLE but it may solve the abort.

You can work your way up or down from there with those values to find the sweet spot.

lbussy, 2021-01-08:
I had randomly chosen some values with (as expected) no improvement.  Can you recommend what might work?

For reference, we're working with up to eight devices that are BLEacon-only.

h2zero, 2021-01-08:
Only values I can suggest would be setting the interval much higher and keeping the window as is. Something extreme, say 5000 interval.

This is not an easy issue to solve for sure and the feedback we have is not ideal. We don't know where or why the problem occurs but we can try.

If the extreme interval value does not improve the results we need to look elsewhere. Let me know what happens if you try that.

lbussy, 2021-01-08:
Well, it's consistent anyway :) :

```
Backtrace: 0x40091978:0x3ffbe190 0x40091ba9:0x3ffbe1b0 0x40122074:0x3ffbe1d0 0x400847ed:0x3ffbe1f0 0x401d6617:0x3ffbd8e0 0x40123626:0x3ffbd900 0x4008f539:0x3ffbd920 0x4008dd45:0x3ffbd940
  #0  0x40091978:0x3ffbe190 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x40091ba9:0x3ffbe1b0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x40122074:0x3ffbe1d0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x400847ed:0x3ffbe1f0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x401d6617:0x3ffbd8e0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x40123626:0x3ffbd900 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008f539:0x3ffbd920 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #7  0x4008dd45:0x3ffbd940 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

h2zero, 2021-01-08:
That looks to be a watchdog timeout. I'd like to see the actual error message, not just the backtrace if possible.

pletch, 2021-01-08:
Our project is [Tiltbridge](https://github.com/thorrak/tiltbridge) and the relevant branch is devel_ardjson.  So very much appreciate you dedicating some time to look at this!

h2zero, 2021-01-08:
Ahh, a repo I can appreciate for sure, BLE beer brewing, LOVE IT! 😄 

Looks like very nice work, I'll see if I can come up with anything.

doudar, 2021-01-08:
That's what I was thinking....

Can I give up on this BLE fitness stuff yet and just drink beer?
😁🍻

*----------------------*
*Anthony Doud*

pletch, 2021-01-08:
If you do get it compiled and loaded, we have generally found it to be easiest to trigger the PANIC reset by repeated hard-reloading of the 'about' tab on the web interface.  This page has several jquery json requests being simultaneously requested.

h2zero, 2021-01-08:
>Can I give up on this BLE fitness stuff yet and just drink beer?
😁🍻

Join the club!

>If you do get it compiled and loaded, we have generally found it to be easiest to trigger the PANIC reset by repeated hard-reloading of the 'about' tab on the web interface. 

Going to give it a shot now.

lbussy, 2021-01-08:
> I'd like to see the actual error message, not just the backtrace if possible.

That's all of it.  I am using the `esp32_exception_decoder` monitor filter so maybe it eats it?

justoke, 2021-01-08:
I've had a very stable ESP32 and reliable web server and BLE scans for around 16 hrs now. To recap I added this 
#define CONFIG_ASYNC_TCP_RUNNING_CORE 1

and used the values from the original scanner example for the BLE:

[image]

lbussy, 2021-01-08:
We had not used the active scan previously.  Using it, I can still get it to crash quite reliably (but maybe it's my imagination but the scans seem quicker?)  My best fail case is reloading this "About" page, which has four JQuery calls - still crashes.

h2zero, 2021-01-08:
Sorry I could not get this loaded last night, seems I may need to use a WROVER module and didn't have any on hand, I'll bring one home from work tonight and give it a try then.

justoke, 2021-01-08:
> We had not used the active scan previously. Using it, I can still get it to crash quite reliably (but maybe it's my imagination but the scans seem quicker?) My best fail case is reloading this "About" page, which has four JQuery calls - still crashes.

This may seem obvious, probably you have already tried this. But can you disable the BLE scan and cause the ESP32 to crash just by hammering the web server concurrently from the About page?

lbussy, 2021-01-08:
> This may seem obvious, probably you have already tried this. But can you disable the BLE scan and cause the ESP32 to crash just by hammering the web server concurrently from the About page?

Nothing wrong with making sure I'm not overlooking the obvious.  While I hate when that happens, it happens. :)

I did disable BLE, and the crashing stopped.  That doesn't mean it's *in* the scan, it just means it's involved.

h2zero, 2021-01-09:
I got a chance to get this loaded and I cannot access the webserver, browser complains of too many redirects... so I am unfortunately unable to test this locally.

h2zero, 2021-01-09:
Nevermind, works now. I realized I had not got the data folder uploaded into SPIFFS. Just got the error to occur on my board as well.

h2zero, 2021-01-09:
Well after some testing and removing the assert in nimble I found the errors after were many watchdog timeout issues. After looking at the relevant code in the backtrace I cannot find anything wifi or ble related. My only thought then was there is a controller code problem (closed source :( ). So I updated my IDF and looked at the update logs:

```
Submodule components/bt/controller/lib 0fd5fffa0..a799b1910 (rewind):
  < bugfix: clear the timer after remote reject when respond the parameters update
  < fix exl200, add btdm_disable_adv_delay, fix RPA addr update error(c03ddb37)
  < Add ACL-U flow on & off when slave accept packet type request (4ae2b95)
  < Free EM ACL RX buffer in disconnection procudure (a61cd5d)
  < Add protection for interval_min when it comes to zero (6bf0910)
  < Remove dummy assert when receive passkey requst (f060711)
  < Fix two bugs:(cfa40b27)
  < fix hci timeout, add timer to monitor scan evt(9fd95e0)
  < Fix controller dead when clkn overflow (1c5361f)
  < fix assert(10, 9), when the AFPM happens(019fd74b)
  < fixed tx_prog insert error(7760a76)
  < Fix waking up fail while sleeping(330943b)
  < make sleep avaliable in hli(for future use)(845e59c)
  < btdm fix cache disable crash(b73c48e)
  < Some bugfix about ble dynamic prio(3901b15)
  < Insert the llcp packet to the top of the tx_preg linked list(153fac6c)
  < refactor coexist event in order to adapter to new chips(07adeadf)
  < Confirmed bugfix from Baidu Project (4bb1a62)
```
Seems to be a few fixes in there that may be related to this issue, I will load this using the IDF master branch and check again.

pletch, 2021-01-10:
It is great to hear that you got it running.  

After further troubleshooting, I think we may have a couple of issues at play which is why the backtraces are slightly different between @lbussy and myself.  

1.  The "normal" configuration for this project is supposed to be a Lolin D32 Pro and Lolin TFT (d32_pro_tft firmware build which is what @lbussy is using). This build will not work if trying to run the ESP32 without a screen attached.  When using a TFT display, we are seeing watchdog trigger errors and this appears to relate to interaction with the TFT during several simultaneous requests to the Async Webserver.  

2. If running the OLED (lcd_ssd1306 firmware) configuration without a screen (which is what I was initially using), the watchdog timers from the main loop are no longer an issue but we were still receiving sporadic RESETs corresponding to the backtrace I posted in the original message.  Leaving the scan Window at 37 and widening the Interval to 197 seemed to eliminate these resets. However, it would be great to be able to tighten the Interval back to improve probability of receiving advertisements if possible so please do let us know if the IDF fixes appear to improve things on the BLE/Wifi side.

lbussy, 2021-01-10:
> I updated my IDF and looked at the update logs:

Some of those look real promising.  Also makes me wonder where all the baling wire and bubble gun is coming from. :)

h2zero, 2021-01-10:
Ok, so my results are sadly non-existent. I had hoped to find out if the controller was the cause of this but after a couple hours futzing around and modifying all the libraries to get them to compile with IDF4.2 I ran into a roadblock where the webserver would say it start and I can see it connected on my router but there is no response in the browser. 

So, I cannot say for sure that is the cause, if I had the time to go about updating a few libraries (already did asyncWifiManager) I might have an answer. I don't think asyncWebServer plays nice with IDF4.2 yet, and I got hung up on the `data_sender.init();` call in your repo.

If you want to invest the time to try this, I can help and give you the changes I made.

@pletch I did use the `lcd_ssd1306` config and confirmed the exceptions both of you experienced with the changes I made in NimBLE, both lead to the same place. I wish I could confirm the fix :(

lbussy, 2021-01-10:
I really appreciate you trying it @h2zero.  At least determining the exceptions with the 1306 and TFT led to the same place was more information than we had.  It means that whether we make the jQuery calls "kinder" or not, we still have the potential to hit them in other areas.

Yes, I'd love to see the work you did to get that far.  You never know.

h2zero, 2021-01-10:
Found an interesting post here https://www.esp32.com/viewtopic.php?f=19&t=16545 that describes a similar problem. Seems possible the issue is triggered by ESPAsyncWebServer taking too long to do something with the radio. 

Here is the modified wifiManager.
[AsyncWiFiManager.zip](https://github.com/h2zero/NimBLE-Arduino/files/5792819/AsyncWiFiManager.zip)
 
And the platformio.ini
[platformio.ini.txt](https://github.com/h2zero/NimBLE-Arduino/files/5792822/platformio.ini.txt)

That will use the idf4.2 branch for the ardruino-esp32 core. You will need to run the get.py or get.exe after it clones the repo and copy the contents from the platformio/packages/framework-arduinoespressif32/tools/xtensa-esp32-elf folder to platformio/packages/toolchain-xtensa32. 
This is what I did to get it compiling, I'm not a platformio user so I don't know it well, there is probably a better way.

lbussy, 2021-01-10:
Thank you very much for that.  It gives me a great starting place to possibly keep prodding this.

h2zero, 2021-01-10:
@lbussy I had so good results in testing against https://github.com/espressif/arduino-esp32/tree/idf-release/v3.3 while investigating other bugs. That branch has the controller updates and seems more reliable, not sure how well it will work for you but it should be easy enough to test. You will need to use NimBLE with the commit I mentioned above reverted as that now causes stack overflow exceptions on that branch.

h2zero, 2021-01-12:
Just to update this, the branch mentioned above is now merged into the arduino-esp32 master.

Updating your sources to use that should provide the fix for this issue. It has the latest controller bug fixes, at least for IDF 3.3 anyway.

justoke, 2021-01-12:
> Just to update this, the branch mentioned above is now merged into the arduino-esp32 master.
> 
> Updating your sources to use that should provide the fix for this issue. It has the latest controller bug fixes, at least for IDF 3.3 anyway.

Would you mind just outlining how one would use a specific branch as opposed to whatever the latest is in Platform IO? I can't see how the release versions are related to the idf and arduino parts. Thank you.
[image]

h2zero, 2021-01-12:
You can just change it in the platformio.ini:
`platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2452c1fb539246e47a715b74a3ad25b8a7ebbec7`

lbussy, 2021-01-12:
I am WAY behind on stuff but I just wanted to answer and thank you for following up.  

I decided to move my domain registrars, hosting, etc., and let's just say I provided myself the opportunity to screw up a number of things. :)

As soon as I have a chance I'll take a look and see if things are better.

h2zero, 2021-01-12:
I'm putting some finishing touches on a couple patches, after that, hopefully I'll get a chance to test this out as well.

justoke, 2021-01-14:
> You can just change it in the platformio.ini:
> `platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2452c1fb539246e47a715b74a3ad25b8a7ebbec7`

I have managed to reference the framework as you suggested. I get this compile error, so is there a specific version of NimBLE that I should reference in my lib_deps:
[image]

Thank you.

justoke, 2021-01-14:
Not to worry, I grabbed master as I see a fix for this was pushed recently

h2zero, 2021-01-15:
I'm trying to push a nimble core update later tonight that works better with the latest arduino master. There are a few core file patches as well as undoing some workarounds for older arduino versions. 

My guess is that it will fix the issues reported here and I will test it once done.

[30 more replies]

closed 2021-01-25 by pletch