h2zero/NimBLE-Arduino #168: Ignore ble device temporarily during scan (time limit)

m-menon, 2021-01-07:
Something I am trying to figure out the best way to achieve - During the BLE scan sometimes I need to ignore a device for a short duration (2-3-4 seconds perhaps). I am storing the time that the device was first found and then subsequently ignoring if the ble device is found again (during the same scan, cos the scan duration is like 8-10 seconds). But the scan won't report the same device (during the same scan) unless I set the find duplicates to  true. Is there a way to achieve this without having to set the find duplicates=true cos that will mess up some other parts of my program. Any tips will be of great while I try to figure out the different ways. Thanks.

h2zero, 2021-01-07:
Hi @m-menon, yes, in fact the callback is the best place to do that as no further scan results will come until you return from the callback so it is unlikely to conflict.

As to ignoring for a duration, this could be done with a task, perhaps triggered after the first result in the callback. 
In the task you could check `NimBLEDevice::isIgnored()` and if false call `::addIgnored()` delay for x time and then `::removeIgnored()`.

m-menon, 2021-01-07:
> Hi @m-menon, yes, in fact the callback is the best place to do that as no further scan results will come until you return from the callback so it is unlikely to conflict.
>
> As to ignoring for a [...]

Fantastic! Thanks for that solution. I will check that out.

closed 2021-01-07 by m-menon