h2zero/NimBLE-Arduino #172: Could you give me some advice to use xbox controller: HID client + L2CAP + Privacy

asukiaaa, 2021-01-09:
I try to get HID notification from xbox controller (for series x, type 1914) but not succeeded.
I can read HID info map and register notification (I checked CCCD become 0100 from 0000) but any notification is not published and a led on the controller is blinking (that means not finish communication setting).

[A driver for linux (xpadneo)](https://github.com/atar-axis/xpadneo) requires [privacy settings](https://github.com/atar-axis/xpadneo/issues/166#issuecomment-747010866)(setting `Privacy=device` to `etc/bluetooth/main.conf`) and seems using [L2CAP](https://github.com/atar-axis/xpadneo/blob/d55e6d42ecb53f3ebe91e7a43574c35e79146dfd/misc/kernel_patches/0001-fix_bluetooth_reconnect.patch#L24) so I think they are needed to communicate with the controller.

Questions

- Does this library support L2CAP?

I found [code about L2CAP on this project](https://github.com/h2zero/NimBLE-Arduino/blob/master/src/nimble/host/mesh/src/src/ble_l2cap_priv.h) but I'm not sure it is activated on client mode.

- Is there any option to use privacy setting on client mode?

I found privacy setting on this project.
However if I call `ble_hs_pvcy_rpa_config(NIMBLE_HOST_ENABLE_RPA);` (or `NIMBLE_HOST_ENABLE_NRPA`) after `NimBLEDevice::init("");` with setting `BLE_HOST_BASED_PRIVACY=1` as a build flag, esp32 cannot read information from characteristics after connection with the controller.

If you have some idea to start notification of the controller, I'm happy if you share that.

Code that I tried: https://github.com/asukiaaa/esp32-client-for-xbox-controller-with-nim-ble/blob/master/src/main.cpp

Thank you.

h2zero, 2021-01-09:
Hi @asukiaaa there should not be a need to use L2CAP directly as NimBLE should take care of that when needed.

If you need to use RPA [here is the details](https://github.com/h2zero/NimBLE-Arduino/blob/537629f471eea1017b28a2482c64a0e7cfbeed55/src/host/ble_hs_pvcy.h#L37), as the API has not be written for this yet, it will require you to make a few modifications to `NimBLEDevice`.

asukiaaa, 2021-01-09:
Thank you very much for the information.
I succeeded in receiving notifications from the controller!

I could set values with using the following code except for `m_own_addr_type`.
```cpp
#include <host/ble_hs_pvcy.h>

void setup () {
    NimBLEDevice::init("");
    NimBLEDevice::setSecurityInitKey(BLE_SM_PAIR_KEY_DIST_ID);
    NimBLEDevice::setSecurityRespKey(BLE_SM_PAIR_KEY_DIST_ID);
    ble_hs_pvcy_rpa_config(NIMBLE_HOST_ENABLE_NRPA);
    // ..
}
```

`m_own_addr_type` is initialized in the constructor but is seems private.
https://github.com/h2zero/NimBLE-Arduino/blob/537629f471eea1017b28a2482c64a0e7cfbeed55/src/NimBLEScan.cpp#L33
Can I create PR to add a function like `setOwnAddrType` to NimBLEScan?

h2zero, 2021-01-09:
I'm thinking that a proper api to support this would be to have NimBLEDevice as the place to set the address type. Then have all the classes where that value is needed make a call to there. Also needs the other stuff to make it work added as well.

asukiaaa, 2021-01-09:
I see, how about this?
https://github.com/h2zero/NimBLE-Arduino/pull/174

asukiaaa, 2021-01-13:
https://github.com/h2zero/NimBLE-Arduino/pull/174 is merged so we can use RPA by calling `NimBLEDevice::setOwnAddrType(BLE_OWN_ADDR_RANDOM)` like this from next release.
```
void setup () {
    NimBLEDevice::init("");
    NimBLEDevice::setOwnAddrType(BLE_OWN_ADDR_RANDOM);
    // ..
}
```

closed 2021-01-13 by asukiaaa