h2zero/NimBLE-Arduino #173: 'Ringbuffertype_t' has not been declared

Proto-ventures, 2021-01-09:
Hi, I'm having an issue compiling this library on one installation of Arduino, but not others. Two windows machines have no issues compiling for the ESP32, though a Mac (Big Sur) fails with the following error:

```
Arduino: 1.8.13 (Mac OS X), Board: “Node32s, Default, 80MHz, 921600, None”
In file included from /Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp:8:0:
/Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.h:72:31: error: ‘RingbufferType_t’ has not been declared
     Ringbuffer(size_t length, RingbufferType_t type = RINGBUF_TYPE_NOSPLIT);
                               ^
/Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp:268:39: error: ‘RingbufferType_t’ has not been declared
 Ringbuffer::Ringbuffer(size_t length, RingbufferType_t type) {
                                       ^
/Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp: In constructor ‘Ringbuffer::Ringbuffer(size_t, int)’:
/Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp:272:48: error: invalid conversion from ‘int’ to ‘ringbuf_type_t’ [-fpermissive]
     m_handle = ::xRingbufferCreate(length, type);
                                                ^
In file included from /Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.h:14:0,
                 from /Users/antonis/Documents/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp:8:
/Users/antonis/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/tools/sdk/include/esp_ringbuf/freertos/ringbuf.h:69:17: note:   initializing argument 2 of ‘void* xRingbufferCreate(size_t, ringbuf_type_t)’
 RingbufHandle_t xRingbufferCreate(size_t xBufferSize, ringbuf_type_t xBufferType);
                 ^
Multiple libraries were found for “WiFi.h”
 Used: /Users/antonis/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for “SD.h”
 Used: /Users/antonis/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc4/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
Error compiling for board Node32s.
This report would have more information with
“Show verbose output during compilation”
option enabled in File -> Preferences.
```

Any idea what's happening here?

h2zero, 2021-01-09:
Please update to the master branch of this repo, a commit was recently pushed to address this.

h2zero, 2021-02-01:
Closing as resolved with 1.1.0 release.

closed 2021-02-01 by h2zero