h2zero/NimBLE-Arduino #175: Return code from NimBLEScan::start is always 0

pletch, 2021-01-13:
Since the addition of commit [413e64640946195bf5f68c384279a615aefe8600](https://github.com/h2zero/NimBLE-Arduino/commit/413e64640946195bf5f68c384279a615aefe8600), NimBLEScan::start always gives a return code of 0.  

Using this function to initiate a scan and always exit through 'the scan failed to start' branch since this commit added.

`bool tiltScanner::scan()
{
    if (!pBLEScan->isScanning())  // Check if scan already in progress
    //Try to start a new scan
    {
        pBLEScan->clearResults(); 
        if (pBLEScan->start(BLE_SCAN_TIME, nullptr,true))  //This no longer ever returns true...possibly a bug?? 
        {
            return true;  //Scan successfully started.
        }
        else
        {
            return false;  //Scan failed to start.
        }
    }
    return false;
}`

h2zero, 2021-01-13:
Any error messages, debug log? Nothing in this respect should have changed for you in that commit.

h2zero, 2021-01-13:
I see the problem, sorry about that [this line](https://github.com/h2zero/NimBLE-Arduino/blob/737de67bc68cdd86a37f3efaea3c00aed9b17997/src/NimBLEScan.cpp#L307) should have been `&&` not `||` .

Fix incoming now.

pletch, 2021-01-13:
Great!  Thank you for the quick fix.

h2zero, 2021-01-13:
Fixed :), thanks for the report!

closed 2021-01-13 by pletch