h2zero/NimBLE-Arduino #176: was this a know crash?

btsimonh, 2021-01-13:
Possibly related to >40 devices present?
@h2zero, was this one of the things you found?
Appears on a device in holland but not for me - I don't have 40 devices - most I get is ~15.

Or, it could be a specific device.
Crash has been seen marked core 0 and another the same marked core 1.
Will be trying you latest push and latest arduino push tonight, but just wanted to confirm if you had seen this before?

```
13:06:18.884 WIF: Checking connection...
abort() was called at PC 0x401c6be7 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40090524:0x3ffe4f50 0x400907a1:0x3ffe4f70 

0x401c6be7:0x3ffe4f90 .text._ZNSt12length_errorC2EPKc
0x401c6c2e:0x3ffe4fb0 
0x401c61a9:0x3ffe4fd0 
0x401c5ef4:0x3ffe4ff0 
0x401c63f1:0x3ffe5010 
0x4011c36f:0x3ffe5030 _S7_St20forward_iterator_tag$isra$33
0x4011c3b9:0x3ffe5060 .text._ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC2EPKcjRKS3_$isra$32
0x4011ca91:0x3ffe5080 .text._ZN22NimBLEAdvertisedDevice18parseAdvertisementEPhh
0x4011e9a7:0x3ffe5160 .text._ZN10NimBLEScan14handleGapEventEP13ble_gap_eventPv
0x4012155a:0x3ffe51a0 .text.ble_gap_rx_adv_report
0x4012657d:0x3ffe5210 
0x40126317:0x3ffe5250 
0x4012687c:0x3ffe5270 
0x40124c31:0x3ffe5290 
0x4012c306:0x3ffe52b0 
0x4011d517:0x3ffe52d0  .text._ZN12NimBLEDevice9host_taskEPv
0x400917a2:0x3ffe52f0

Rebooting...
ets Jun  8 2016 00:22:57
```

h2zero, 2021-01-13:
Yes, this is the issue on the new arduino core. I'll try for a fix today.

Ysbrand, 2021-01-13:
I've got 2 ESP32 unit running and with the regular reboots (uptime between 10-20 minutes on average) I have seen already 3 times that they are rebooting at more or less the same time, So I tend to agree that it has something to do with the parsing of incoming advertisements.

h2zero, 2021-01-13:
Thanks for the confirmation @Ysbrand.

If you're willing, could you try this branch https://github.com/h2zero/NimBLE-Arduino/tree/scan-advertisedDevice-rework
and let me know if the issue persists?

btsimonh, 2021-01-13:
"more or less the same time" - yes, points to possibly a bad advert packet (not to spec, odd in some way).
Will be interesting to see if we could get a dump of it, and maybe the device MAC so we can identify the device.

btsimonh, 2021-01-13:
@h2zero - i've given @Ysbrand a binary with it printing mac & payload for EVERY advert direct from NimBLE - so hopefully if it is a specific device, we may be able to identify it, and if it is a bad payload, you'll have something to qualify the new decode with....

btsimonh, 2021-01-13:
should I merge scan-advertisedDevice-rework into master to make a combined branch before re-testing everything in Tasmota?

h2zero, 2021-01-13:
If you'd like to test it, please do so. Just know that there are possible bugs still.

btsimonh, 2021-01-13:
:)

h2zero, 2021-01-13:
Thanks for the reminder, I just merged master into branch. Should be good to test now

btsimonh, 2021-01-13:
on old code, just changed some debug.  SLIGHTLY different, but very related:
Backtrace: 
0x40090524:0x3ffe38c0 
0x400907a1:0x3ffe38e0 
0x401c6cbf:0x3ffe3900 
0x401c6d06:0x3ffe3920 
0x401c5ec3:0x3ffe3940 
0x401c5fda:0x3ffe3960 
0x4011e7ae:0x3ffe3980 .text._ZNSt6vectorIP22NimBLEAdvertisedDeviceSaIS1_EE19_M_emplace_back_auxIJRKS1
0x4011e9a6:0x3ffe39a0 .text._ZN10NimBLEScan14handleGapEventEP13ble_gap_eventPv
0x40121632:0x3ffe3a00 
0x40126655:0x3ffe3a70 
0x401263ef:0x3ffe3ab0 
0x40126954:0x3ffe3ad0 
0x40124d09:0x3ffe3af0 
0x4012c3de:0x3ffe3b10 
0x4011d527:0x3ffe3b30 
0x400917a2:0x3ffe3b50

so, still within handleGapEvent, but failed at the push_back, rather than the parse.  Are we running out heap maybe?

h2zero, 2021-01-13:
It appears you could be, maybe too fragmented to allocate a bigger block?

Ysbrand, 2021-01-13:
As a bonus I've got also 2 occurrences of a Panic. I do not have the map file, so we will need to wait for Simon to fill in the blanks (And confirm that it's in the BLE code). @btsimonh I've send you the entire log.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4010fe30  PS      : 0x00060130  A0      : 0x8010fef8  A1      : 0x3ffd20f0
A2      : 0x00000000  A3      : 0x3ffd213f  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x5cffffe1  A11     : 0x3fff358c  A12     : 0x00000050  A13     : 0x00000001
A14     : 0x5cffffe1  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace:
0x4010fe30:0x3ffd20f0
0x4010fef5:0x3ffd2110
0x401cf5e6:0x3ffd2130
0x40139c1a:0x3ffd2160
0x40139e51:0x3ffd2180
0x40119aaa:0x3ffd21a0
0x401181bd:0x3ffd2390
0x400dcbf7:0x3ffd23e0
0x4010f648:0x3ffd2400
0x400e91ea:0x3ffd2420
0x4010f93a:0x3ffd2440
0x4010f9eb:0x3ffd2460
0x4013da14:0x3ffd2480
0x400917a2:0x3ffd24a0

h2zero, 2021-01-13:
Looks like a nullptr error, don't know where from without backtrace decoded.

btsimonh, 2021-01-14:
"It appears you could be, maybe too fragmented to allocate a bigger block?"
hmm .. hate new and delete in MCUs :(  i'll decode further up the trace.

btsimonh, 2021-01-14:
2nd type of crash not (directly) BLE related:
Backtrace: 
0x4010fe30:0x3ffd20f0 .text._ZN18WiFiClientRxBuffer4readEPhj
0x4010fef5:0x3ffd2110 
0x401cf5e6:0x3ffd2130 
0x40139c1a:0x3ffd2160 
s

btsimonh, 2021-01-14:
ok, for the record.
@Ysbrand has some device nearby which is making random addresses and changing them very quickly - like 400 times in 10 minutes. (not his 2 android phones)
I was scanning requesting a continuation every time.
So, it crashes at 400+ devices logged in the scan.
@h2zero, thanks for your help, will try the new branch(s) soon, but the issue here is not necessarily the code :).
Also, looks like the random address used does not follow the guidelines, and so cannot easily be excluded :(.
(unless there is a more accurate resource than https://www.novelbits.io/bluetooth-address-privacy-ble/).

I will change to start a new scan every time, and report if we see anything else.  (we know that we can see ~40 in the 20s scan, so this should be manageable).

closed 2021-01-14 by btsimonh

Ysbrand, 2021-01-14:
For the record: it turned out to be a Bluetooth speaker spraying random advertisements, each time with a different mac address.

I will report this back to the vendor as its probably a bug.

tonton81, 2021-01-14:
i know what you are talking about, after the scan callback i wipe the list using clearDevices() and the list clears instead of constantly appending, this prevents it from growing and growing until it eats all the heap. never had issues after that

h2zero, 2021-01-14:
>I will report this back to the vendor as its probably a bug.

Probably not a bug, it's using privacy so it can change it's address often.

>i know what you are talking about, after the scan callback i wipe the list using clearDevices()

I'm working on implementing a user defined limit, or 0, for this issue. Watch the scan-advertisedDevice-rework branch.

btsimonh, 2021-01-14:
it's a pity the random addresses did not seem to match the above link/spec.  Else we could exclude 'random' addresses....  I think in our code, we'll imlement a dual-timeout - things we know we want to hear will last longer in our (external to NimBLE store) than things which we have no clue what they are.  We COULD detect things like Android contact trace adverts - but this just proves there are an infinite number of possibilities...

h2zero, 2021-01-14:
What address type value did this device have?

btsimonh, 2021-01-14:
not familiar with that.  some call we can make on NimBLEAddress? - we should be able to check.

btsimonh, 2021-01-14:
so... getType will return 0 (public static) or 1 (random).  If random, then the low two bits of the address (which byte?) indicate the type of random, in theory?

h2zero, 2021-01-14:
This has the info you need:
https://github.com/h2zero/NimBLE-Arduino/blob/3a4c2635ed4ebd9368bd3cad3491c2f7f8d18a3e/src/nimble/ble.h#L253

NimBLE should report one of those 4 types.