h2zero/NimBLE-Arduino #180: NimBLE and iTag

Edzelf, 2021-01-20:
I have tried to connect to an iTag (see my github page).  That works correctly.  After that I tried to subscribe to the button notification.  The iTag is not fully compliant and does not have descriptor 0x2902.
NimBLERemoteCharacteristic::setNotify() returns "false" because of this missing descriptor.  Setnotify wants to write to this descriptor.  But the iTag notification will work without this write operation.  So I changed setNotify in the following way:
```
bool NimBLERemoteCharacteristic::setNotify(uint16_t val, notify_callback notifyCallback, bool response) {
    bool res = true ;

    NIMBLE_LOGD(LOG_TAG, ">> setNotify(): %s, %02x", toString().c_str(), val);
    // Set callback, even if 2902 descriptor is missing
    m_notifyCallback = notifyCallback;
    // Get 2902 descriptor if it exists
    NimBLERemoteDescriptor* desc = getDescriptor(NimBLEUUID((uint16_t)0x2902));
    if(desc) {
        res = desc->writeValue((uint8_t *)&val, 2, response);
    }else{
        NIMBLE_LOGE(LOG_TAG, "<< setNotify(): Could not get 2902 descriptor");
    }
    NIMBLE_LOGD(LOG_TAG, "<< setNotify()");
    return res;
} // setNotify

Would it be a good idea to have this modification included in you library?
```

h2zero, 2021-01-20:
Hi @Edzelf this issue has been resolved already in commit 8310b7b56058ccc976d294fc990166d95b14744c .

Edzelf, 2021-01-21:
Thank you for your support.

I still have a strange problem, I am using PlatformIO for ESP32.  In platformio.ini:
```
lib_deps =
	h2zero/NimBLE-Arduino@^1.0.2
Then I get NimBLERemoteCharacteristic.cpp date 14 September.

```
A newer version is not available in PlatformIO.  I tried:
```
lib_deps =
	h2zero/NimBLE-Arduino@^1.1.0
```
But that gives me:
```
Library Manager: Installing h2zero/NimBLE-Arduino @ ^1.1.0
KeyError: 'framework-stm32cube':
  File "C:\users\hp\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\hp\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\hp\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\hp\.platformio\platforms\espressif32\builder\main.py", line 224:
    target_elf = env.BuildProgram()

```

h2zero, 2021-01-21:
Platformio has not updated with the latest release yet. You can change your dependencies to point at the repo like this:

lib_deps = https://github.com/h2zero/NimBLE-Arduino.git#1.1.0

Edzelf, 2021-01-21:
That did the trick.
Thank you!

h2zero, 2021-01-27:
@Edzelf can we close this?

closed 2021-01-27 by Edzelf