h2zero/NimBLE-Arduino #183: Max data packet length and callback timing on NimBLE vs. Arduino standard BLE library

ClaesClaes, 2021-01-29:
Hi,
I'm porting some file transfer code to NimBLE that previously used the Arduino standard BLE library but are running into some issues. I wonder if I can get some help sorting this out as I really like (and need) the exceptionally low footprint of NimBLE. 

Firstly, pls note that the file transfer works fine with the standard Arduino BLE library. However the write callback, when using NimBLE, is not called (at all it seems) if the data package sent from the client is longer than 512 bytes.

Secondly, when limiting data package length to 512 or lower on the client side, NimBLE seems to drop data packages when I perform something slightly time consuming inside the callback (like writing to a SD card, even if I buffer before writing to the card). In fact I always randomly loose between 2 to 40 data. This works without any issue with the Arduino standard BLE library. 

**Here's some details:**
1. My ESP32 is running an BLE server with a characteristic (WRITE_NR) used for receiving files from an iOS app on an iPhone X. 

2. My iOS app report a possible MTU of 514 bytes after connection to the ESP32. I use this length to chunk up the transfer file into chunks of 514 bytes. These are transmitted to the ESP32 with "write without response" (WRITE_NR). The iOS app checks if the the ESP32 is ready before sending another chunk. It works reasonably well, for my application, as I achieve 16-18kB/s. As mentioned above a chunk size of 514 only works with Arduino standard BLE library. For NimBLE I need to lower it to 512 or less for the callback to register the data package.

3. On the ESP32 side I use a 16 kB buffer in the callback which, when full is offloaded to an SD card and thereafter continues until all packages are received.

4. Heap memory seems fine but the stack HWM runs much lower on NimBLE than Arduino BLE.

I would really appreciate any help I could get :-)

**And some code which I use for testing:**

```
/*
  #include <BLEDevice.h>
  #include <BLEServer.h>
  #include <BLEUtils.h>
  #include <BLE2902.h>
*/
#include "NimBLEDevice.h"

#include <SPI.h>
#include "FS.h"
#include "FFat.h"
#include "SD.h"

File uploadFile;

BLEServer* pServer = NULL;
BLECharacteristic * pTransferCharacteristic;

bool      deviceConnected = false;
bool      oldDeviceConnected = false;
bool      sdCardOk      = false;

#define SERVICE_UUID                  "4FAFC201-1FB5-459E-8FCC-C5C9C331914B"
#define CHARACTERISTIC_TRANSFER_UUID  "62ec0272-3ec5-11eb-b378-0242ac130004"

bool downloadFlag = false;
int bufferIndex = 0;
int bufferCount = 0;
char dataBuffer[16448];

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
      Serial.println(F("*** Client connected"));
      Serial.print(F("Max heap size: "));
      Serial.print(ESP.getFreeHeap());
      Serial.print(F(" bytes, largest continous size: "));
      Serial.print(ESP.getMinFreeHeap());
      Serial.print(F(" bytes, "));
      Serial.printf("Stack HWM: %i \n", uxTaskGetStackHighWaterMark(NULL));
    }

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
      Serial.println(F("*** Client disconnected"));
    }
};

class transferCallback: public BLECharacteristicCallbacks
{
    void onWrite(BLECharacteristic *pCharacteristic)
    {
      std::string rxData = pCharacteristic->getValue();
      totBytesReceived += rxData.length();
      if (rxData.length() < 512)
      {
        Serial.print(F("Finished"));
      }
    }
};

class transferCallbackSD: public BLECharacteristicCallbacks
{
    void onWrite(BLECharacteristic *pCharacteristic)
    {
      std::string rxData = pCharacteristic->getValue();
      for (int i = 0; i < rxData.length(); i++)
      {
        dataBuffer[bufferIndex++] = rxData[i];
      }
      bufferCount++;
      if (!downloadFlag)
      {
        if (SD.exists("/temp.dat"))
        {
          Serial.println("File already exist");
          if (!SD.remove("/temp.dat"))
          {
            Serial.println("Failed to remove file");
          } else {
            Serial.println("File removed");
          }
        }
      }
      downloadFlag = true;
      if (bufferCount > 31 || rxData.length() < 512)
      {
        uploadFile = SD.open("/temp.dat", FILE_APPEND);
        if (uploadFile.write((uint8_t *) dataBuffer, bufferIndex))
        {
          Serial.println(bufferIndex);
          /*
            Serial.print(F("Max heap size: "));
            Serial.print(ESP.getFreeHeap());
            Serial.print(F(" bytes, largest continous size: "));
            Serial.print(ESP.getMinFreeHeap());
            Serial.print(F(" bytes, "));
            Serial.printf("Stack HWM: %i \n", uxTaskGetStackHighWaterMark(NULL));
          */
        } else {
          Serial.println(F("Write buffer to SD failed"));
        }
        uploadFile.flush();
        bufferIndex = 0;
        bufferCount = 0;
      }
      if (rxData.length() < 512)
      {
        Serial.print("Finished: ");
        Serial.println(uploadFile.size());
        downloadFlag = false;
        uploadFile.close();
      }
    }
};

void setup() {
  Serial.begin(115200);
  // SD card setup 
  SPIClass SDSPI(VSPI);
  SDSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  SD.begin();
  if (!SD.begin(SD_CS, SDSPI))
  {
    Serial.println(F("SDCard Mount Failed"));
    sdCardOk = false;
  } else {
    Serial.println(F("SDCard Mount PASS"));
    sdCardOk = true;
  }

  // 1. Create the BLE Device
  BLEDevice::init("BlipZero");
  BLEDevice::setMTU(517);

  // 2. Create the BLE server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // 3. Create BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // 4. Create BLE Characteristics inside the service(s)
  pTransferCharacteristic = pService->createCharacteristic(CHARACTERISTIC_TRANSFER_UUID,
                            NIMBLE_PROPERTY::WRITE_NR);
  //BLECharacteristic::PROPERTY_WRITE_NR);
  pTransferCharacteristic->setCallbacks(new transferCallbackSD());

  // 5. Start the service(s)
  pService->start();

  // 6. Start advertising
  pServer->getAdvertising()->addServiceUUID(pService->getUUID());
  pServer->getAdvertising()->start();
  BLEDevice::startAdvertising();
  Serial.println(F("Waiting a client connection to notify..."));
}

void loop() {
  if (!deviceConnected && oldDeviceConnected) {
    delay(100);
    pServer->startAdvertising();
    Serial.println(F("start advertising"));
    oldDeviceConnected = deviceConnected;
  }
  if (deviceConnected && !oldDeviceConnected) {
    oldDeviceConnected = deviceConnected;
  }
}  
```

h2zero, 2021-01-29:
Hi @ClaesClaes, I'm not sure what bluedroid may be doing differently but as per the bluetooth spec: 
>The maximum length of an attribute value shall be 512 octets.

So that is the max length of the characteristic value. 
The MTU is a different thing since a 3 byte header is prepended to the data (5 bytes for long write). That being said, if your phone has a max MTU of 514 then your max value data length would be 511 bytes, this should not be an issue for NimBLE as it will only reject the data if longer than 512. 

My guess then is, if you are sending 514 (MTU) bytes from the phone, it might be trying to send via long write (since it needs room for the header) but your characteristic does not have `NIMBLE::PROPERTY_WRITE` which would be required for that to work. In that case the suggestion would be to reduce the chunk size sent from the phone to 511 to avoid this or, if possible, set the phone MTU to 515 and then you will be able to send 512bytes. 

>Heap memory seems fine but the stack HWM runs much lower on NimBLE than Arduino BLE.

NimBLE task stack only has 4096 bytes, not sure about bluedroid. If you're running into issues overflowing you can adjust the task stack size in [nimconfig.h](https://github.com/h2zero/NimBLE-Arduino/blob/4170bc9b48ddba5945cb115f3e0ad915b9164026/src/nimconfig.h#L111).

ClaesClaes, 2021-01-30:
Thanks @h2zero, I really appreciate you quick reply. Thanks for the hint. I revisited the BLE specs (3.4.5.3) and indeed ATT MTU-3 it is. So for "write without response" I'm limited to 511 bytes payload with an MTU of 514. Thanks for pointing that out. Still I've no idea how Arduino BLE library does it, but it's solid every time I send out +100kB files using a 514 (or less) byte WRITE_NR transfer. Regardless it's not a big deal with the extra 3 bytes so I'm happy if I can at least get 511 bytes. 

However, while 511 bytes did the trick (mostly) I still occasionally see dropped data using NimBLE (511 bytes payload, WRITE_NR), while Arduino BLE library never drops any data. How can I troubleshoot this? My guess would be that the peripheral respons to a WRITE_NR is sent out to the client before the callback has finished and therefore are not ready to receive a new package. Could this be the case?

The phones MTU is as far as I know not adjustable. It depends on the inner works of iOS CoreBluetooth. So I'm just a taker there. For writes with respons it gives me 512 and 514 for writes without response.

When using "WRITE" with response, NimBLE performs flawless, however the transfer rate drops as expected to half at around 7kB/s which is a tad too low for me.

Again grateful for any help I could get to get WRITE_NR to work stably.

h2zero, 2021-01-30:
You're welocme @ClaesClaes .

>My guess would be that the peripheral respons to a WRITE_NR is sent out to the client before the callback has finished and therefore are not ready to receive a new package. Could this be the case?

This is certainly possible, the write request is always responded to on the link layer which will tell your phone app that it's complete but that does no mean the peripheral is done processing the data. Since you are writing to SPIFFS in your callback that could be the cause, as that is a time consuming process. I would think the best approach would be a message queue and a separate task that does the SPIFFS write while the callback just copies the data into a buffer and sends a queue message.

ClaesClaes, 2021-02-01:
Thanks @h2zero, 
I think I 've been thrown off the track by the fact that the same code produce different results depending upon which Arduino Library I use. Hence I thought there could be an issue on how callbacks are implemented. 

However the real issue seems to be how I have interpreted the iOS CoreBluetooth framework function that supposedly should give a "go a head" signal when the peripheral is ready to receive another "write without response" data package.

Apple's documentation does not reveal any details but I now suspect that the function just monitors the internal iOS peripheral managers transmit buffer and calculates, based on the connection interval time, when the server "should" be ready. But that does not take into account the time the callback takes to complete. 

I've read on a [Nordic Dev forum](https://devzone.nordicsemi.com/f/nordic-q-a/30113/packet-loss-sending-data-from-ios-to-nrf52-dk) that their own solution on how to speed up OTA DFU traffic using write without respons involves a function that sends a (n) number of packages then waits for package recept notification sent by the server after (n) packages been received/stored at the end of the callback. Sounds like a promising way forward.

Thanks for your time and patience :-)

h2zero, 2021-02-02:
>I think I 've been thrown off the track by the fact that the same code produce different results depending upon which Arduino Library I use. Hence I thought there could be an issue on how callbacks are implemented.

This is possible due to internal BLE core handling of these packets. It's quite likely that NimBLE just responds to them sooner.

I suspect that IOS responds with a ready signal when the link layer packet confirmation and or packet sent confirmation message is received. As the nature of the write without response is "does not matter if it was received correctly" then you very well may send data before it the previous data was processed. 

> I've read on a Nordic Dev forum that their own solution on how to speed up OTA DFU traffic using write without respons involves a function that sends a (n) number of packages then waits for package [...]

That seems like a reasonable way to do things, let me know what you come up with.

ClaesClaes, 2021-02-03:
Finally, stability established. All packages arrives safely and with twice the speed (average 32kB/s) using a loop approach with WRITE_NR packages and leveraging iOS CoreBluetooth framework functions. 

The stability comes from sending, in my case the sweet spot is 16 x 511 byte, WRITE_NR packages from my phone in a loop. Inside the loop I use the iOS CoreBluetooth functions to make sure the peripheral is ready. The iOS app then waits for the BLE server to send a specific ACK on a separate Characteristic which then triggers my iOS app to loop another batch of WRITE_NR packages until all packages are transfered. 

The speed increase was mostly achieved by lowering the max connection intervall to 15 ms (lowest allowed by current iOS CoreBluetooth framework for iPhones). I've totally missed to set this in the BLE server as I assumed it was set automatically to the lowest at the time of the connection.

h2zero, 2021-02-03:
Nice, I'm glad it's working out! Thanks for reporting your solution.

> I've totally missed to set this in the BLE server as I assumed it was set automatically to the lowest at the time of the connection.

The client controls the connection, no need to set/change anything on the server side.

ClaesClaes, 2021-02-03:
> The client controls the connection, no need to set/change anything on the server side.

Sorry I'm a bit confused I thought that the server (peripheral), not the client, set the boundaries for the connection intervall length?  I've set the max connection interval to 15 ms (12 x 1.25 ms) on the server after connection. This gives me 32kB/s using this:
`(pServer->updateConnParams(desc->conn_handle, 9, 12, 0, 50);`
 If I omit this setting and use the library defaults, speed drops to 13-16 kB/s. What's the default min/max connection intervall?

ClaesClaes, 2021-02-03:
...or are there no defaults and the client (the phone in my case) proposes the min/max intervalls during the connection interval, which the server accepts? If so I assume that iOS starts with a higher (probably 30 ms) connection interval and only goes down to the lowest (15 ms) after a request from the server (peripheral).

h2zero, 2021-02-03:
The intervals should be specified specifically by the application and if not I would assume iOS uses some default values. I have no idea what those would be, however, whatever they are the peripheral has no choice but to accept them or disconnect and NimBLE will accept any value provided if it is a legitimate value.

ClaesClaes, 2021-02-03:
Ok, that might be it. -> iOS proposes something and in this case I tell it (via the server side setting) that I'd rather prefer a much shorter (15 ms) max interval. Anyhow all is well and good so a super thanks for your help in guiding me to a solution, it was super helpful of you.

h2zero, 2021-02-03:
My pleasure, quite an interesting topic as I have not done any iOS BLE applications. This will be a good reference for the others as well. Let me know if you have any other issues/questions.

ClaesClaes, 2021-02-03:
Will do, thanks again.

closed 2021-02-03 by ClaesClaes