h2zero/NimBLE-Arduino #184: Log Level Info

lbussy, 2021-01-29:
I feel silly asking this because I did creep the source code, but I cannot seem to find a way to filter these:

```
LE advertising report. len=42 num=1 evtype=0 addrtype=0 addr=fc.8f.90.a6.1b.ec advlen=31 rssi=-69
02 01 00 1b ff 75 00 42
04 01 01 ae fc 8f 90 a6 
1b ec fe 8f 90 a6 1b eb
01 00 00 00 00 00 00
```

I went looking for all the log tags and thought I collected them all, but apparently I am missing something:

```
esp_log_level_set("FreeRTOS", ESP_LOG_WARN);
esp_log_level_set("NimBLEAddress", ESP_LOG_WARN);
esp_log_level_set("NimBLEAdvertisedDevice", ESP_LOG_WARN);
esp_log_level_set("NimBLEAdvertising", ESP_LOG_WARN);
esp_log_level_set("NimBLEAdvertisingReport", ESP_LOG_WARN);
esp_log_level_set("NimBLEBeacon", ESP_LOG_WARN);
esp_log_level_set("NimBLECharacteristic", ESP_LOG_WARN);
esp_log_level_set("NimBLEClient", ESP_LOG_WARN);
esp_log_level_set("NimBLEDescriptor", ESP_LOG_WARN);
esp_log_level_set("NimBLEDevice", ESP_LOG_WARN);
esp_log_level_set("NimBLEEddystoneTLM", ESP_LOG_WARN);
esp_log_level_set("NimBLEEddystoneURL", ESP_LOG_WARN);
esp_log_level_set("NimBLERemoteCharacteristic", ESP_LOG_WARN);
esp_log_level_set("NimBLERemoteDescriptor", ESP_LOG_WARN);
esp_log_level_set("NimBLERemoteService", ESP_LOG_WARN);
esp_log_level_set("NimBLEScan", ESP_LOG_WARN);
esp_log_level_set("NimBLEServer", ESP_LOG_WARN);
esp_log_level_set("NimBLEService", ESP_LOG_WARN);
esp_log_level_set("NimBLEUtils", ESP_LOG_WARN);
esp_log_level_set("NimBLEUUID", ESP_LOG_WARN);
```

Did I miss something defining these in the docs?

h2zero, 2021-01-29:
Not sure how you have it turned on but [this line in nimconfig](https://github.com/h2zero/NimBLE-Arduino/blob/4170bc9b48ddba5945cb115f3e0ad915b9164026/src/nimconfig.h#L87) is what you need to disable.

lbussy, 2021-01-29:
I didn't think I did any explicit changes, but you never know. :)

It's already commented ... 

```
/*  Uncomment to see debug log messages from the NimBLE host
 *  Uses approx. 32kB of flash memory.
 */
// #define CONFIG_BT_NIMBLE_DEBUG
```

I searched for every define and commented it out.  I also included:

```
#undef CONFIG_NIMBLE_DEBUG
#undef CONFIG_BT_NIMBLE_DEBUG
```

... in my `main.cpp`.  Still happens.  You ever feel like you were jinxed?

h2zero, 2021-01-29:
Is it defined in platformio.ini or something? Are you using arduino as an IDF component and have it enabled in menuconfig?

lbussy, 2021-01-29:
Nope and nope.  I mean, I have done something silly of course.  Just not sure where.  It's in [async_work](https://github.com/thorrak/tiltbridge/tree/async_work) right now.

h2zero, 2021-01-29:
That's really odd lol, the only other thing that would enable these messages is having `MYNEWT_VAL_BLE_HS_DEBUG(1)` defined somewhere. I searched your code for that but could not find it either.

You could try editing this: https://github.com/h2zero/NimBLE-Arduino/blob/08be9140c313af980331d7d8967563ee8c254489/src/esp_nimble_cfg.h#L435 

So only MYNEWT_VAL_BLE_HS_DEBUG(0) can be defined, let me know if that works. I don't have the ability to test this right now but I can look at it later if you don't find the issue first.

lbussy, 2021-01-29:
Yeah saw those.  Tried both commenting the line setting it to 1 and set it to 0 in platformio.ini.  I'm just jinxed.  The whole world hates me.

h2zero, 2021-01-29:
Was looking at the modlog code, I wonder if you may not have ARDUINO_ARCH_ESP32 defined?

lbussy, 2021-01-29:
I believe that's something PIO adds (and when I add it, it complains that I am redefining it.)  Nevertheless, I tried an `#undef` and `#define` with no change.

h2zero, 2021-01-29:
Lol well the whole world really does hate you then 😄.

 I will see what I can find when I get home tonight, was this always happening or is it the result of a recent change?

lbussy, 2021-01-29:
I turned up logging overall to help me troubleshoot some secure client issues (still plaguing me) and it's hard to see the forest for the trees with the advertising messages.  I could turn off the scanner, but then that disables a whole block of code that I'm trying to troubleshoot.

corneliusmunz, 2021-01-29:
@lbussy Have you changed it in the correct location? I also have an issue using PlatformIO. I changed it in the library file but nothing happend. You have to change the library file inside your project directory `{projectdirectory}/.pio/libdeps/{boardname}/NimBLE-Arduino/src/nimconfig.h`

lbussy, 2021-01-29:
I have, I only use project libs so I don't get confused (anymore).

I'm wondering if it's not time to flush my entire PIO ecosystem and start over again.  I've found sometimes it starts to act flakey and this may be one of those times.

h2zero, 2021-01-29:
A complete clean by deleting the projectdir/.pio folder might work actually.

lbussy, 2021-01-29:
Yeah, that's what we call "the hourly."  :P

With three boards configured it's easy for me to lose track of what I am doing so I end up doing that often.

h2zero, 2021-01-30:
Haha "the hourly", sounds about right. 
I just got home, going to check this out and let you know if I can find the issue.

h2zero, 2021-01-30:
While working on this I was able to remove those messages but I also got this:
```
D NimBLEScan: "<< start()"
I NimBLEScan: "New advertiser: 5b:b5:b6:01:26:dc"
I (538863) wifi:state: run -> init (0)
I (538865) wifi:pm stop, total sleep time: 488110947 us / 537232278 us

I (538866) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:2
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
I (538941) wifi:flush txq
I (538941) wifi:stop sw txq
I (538942) wifi:lmac stop hw txq
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```
Did not get a backtrace sadly but I thought I'd post it here for reference.

Also, so far, commenting out `LOG_LOCAL_LEVEL` in platformio.ini seems to remove the log messages, I'm still unsure why.

lbussy, 2021-01-30:
Oh my god thank you!  It was driving me nuts!  That said, clearly, I had a misconception about the proper way to enable logging.  I thought you turned up a common log level and turned it down on individual models.  Lesson learned there.

The wifi disconnect is troubling, hopefully, it was a fluke.  These things are never going to be used to save any lives, that's for usre.

h2zero, 2021-01-30:
>That said, clearly, I had a misconception about the proper way to enable logging. I thought you turned up a common log level and turned it down on individual models. Lesson learned there.

I don't have a lot of experience with this either, was quite puzzled that it turned on the NimBLE host logs as well, I could understand the others but that seemed odd.

The wifi disconnect wasn't much of an issue, was more concerned about the crash and reboot. It happened a couple times but I did not get a trace on it sadly.

> These things are never going to be used to save any lives, that's for usre.

Scary thought lol

lbussy, 2021-01-30:
The crash was known - that's one of the reasons I had debug turned way up.  We're past that now, I think, fingers crossed ... :)

closed 2021-01-30 by lbussy