h2zero/NimBLE-Arduino #185: Question about pin codes & pairing

btsimonh, 2021-01-30:
Hi h2zero,

looking at the sample
https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Secure_Client/NimBLE_Secure_Client.ino

it looks like to do pairing, I just have to implement onPassKeyRequest().

documentation on 'pClient->secureConnection();' says it will be called automatically?   or should i add it for all connect calls regardless of if they may need a code or not?

So..  in my existing code, I would need to:

1/ record the MAC of the device I am connecting to.

2/ if onPassKeyRequest() is called, return the pin associated with the MAC address.

3/ everything else is unmodified?

Is this correct?

best regards,

Simon

p.s. thanks for a great library...  Tasmota is working quite well with it now (in by default in current version); but awaiting ESP updates to solve the remaining hangups :).

h2zero, 2021-01-30:
Hi @btsimonh glad to hear things are going well!

Yes that sums up all you would need to do on the client, other than perhaps turning on bonding if desired.

btsimonh, 2021-01-30:
what does bonding mean/do?
how?
(we don't have 'easy' persistent storage here..  but people would be happy to set some pairing keys at startup via a startup script)

h2zero, 2021-01-30:
It stores the results from pairing so when re-connecting it wont be necessary to go through the pairing process again.

h2zero, 2021-01-30:
I should mention also, for passkey to work you must have this after NimBLEDevice::init()
```
NimBLEDevice::setSecurityAuth(false, true, true);
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_KEYBOARD_ONLY);
```
To enable bonding you just need to change first parameter in `setSecurityAuth()` to true.
NimBLE takes care of the persitant storage and loading on reboot.

h2zero, 2022-04-05:
Closing as resolved.

closed 2022-04-05 by h2zero