h2zero/NimBLE-Arduino #186: Update to migration guide

j4m3s, 2021-02-01:
I've come to migrate and noticed that the change to the way characteristics are created has changed more than the migration guide suggests.  Using Bluedroid I subclassed `BLECharacteristic` and called `service->addCharacteristic()`.  Could I suggest including a note in the migration guide to say that the constructor is now private and the only way to create one is now `createCharacteristic`?

h2zero, 2021-02-01:
Hi @j4m3s, thank you for reporting this. That is certainly something that should be done, sorry for the lack of documentation. 

I will be revisiting that code soon as well as it is overdue for some updates, any feedback regarding this would be much appreciated.

j4m3s, 2021-02-01:
I'd be happy to submit a PR for the update to the guide if it would help? 

When you say you're going to revisit the code do you mean you might be reintroducing the public constructor for the Characteristics? I only ask because I'm about to change my own code to use the createCharacteristic method but might hold off if you'll be making changes shortly.

h2zero, 2021-02-02:
That would be great if you'd like to PR the guide update.

I'm not sure if that part of the code will be changed yet. I need to do a full review of the server code as there are a couple nitpicks I have and some things I'd like to possibly add.

Spent too much time on the client side of things lately so I need to refresh my mind with the server code.

PR #189 (2021-02-02): Update migration guide to include Characteristic constructor now being private
Closes #186
closed 2021-02-02 by h2zero