h2zero/NimBLE-Arduino #187: SSL Connection Error?

doudar, 2021-02-02:
@h2zero , I've searched the other issues and don't see an exact match but I'm pretty sure it's been reported before.

I'm trying to implement UniversalTelegramBot for a little debugging feedback. It works if I use WiFiClientSecure::setInsecure(); (disable SSL)but if I omit doing that (and use SSL), it returns a continuous stream of "[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1".  It seems to work fine previous to starting the NimBLE Client and Server. ESP.getFreeHeap() called during the routine reports > 130000

I know that's not a ton of information to go off of - I'm hoping you've seen the issue before. 

Any suggestions?

h2zero, 2021-02-02:
This is somewhat of a known issue, though I have yet to find a reason for it. 

For reference: https://github.com/1technophile/OpenMQTTGateway/issues/794

doudar, 2021-02-02:
Thanks for the ref. I'll just leave it unsecured for now as nothing sensitive is being transmitted.

closed 2021-02-02 by doudar

h2zero, 2021-02-04:
Tried messing with this today, interestingly I had no issue scanning BLE and maintaining an ssl connection. 

Is there any code I should add to reproduce? I'd like to see if I can find a solution to this.

wakwak-koba, 2021-02-05:
The ESP32's WiFi Bluetooth has a shared antenna, so simultaneous use should have been a problem.
https://www.esp32.com/viewtopic.php?t=6707
https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/wifi_bt_ble_coex

It seems that consideration is needed to make them coexist, but I am not very familiar with them...

doudar, 2021-02-05:
@h2zero, if you take my telegram branch of SmartSpin2K, go to the bottom of
http_server.cpp and comment out the unsecured line. Crash every time. Btw,
I’m not sure it will compile without your own telegram bot info. It’s on my
todo list before I merge with main.

h2zero, 2021-02-05:
>SSL has a long payload, so the impact may be particularly noticeable ?

@wakwak-koba WiFi and BLE can coexist just fine normally and is handled by internal espressif libraries, however you are quite right here. There is a lot of timing dependent operations going on.

@doudar are you referencing the firmware update handler?  I don't see any reference to `setInsecure` there.

doudar, 2021-02-05:
Hmm...helps if you push changes doesn't it 😅

Line 608 in http_server_basic.cpp in the telegram branch.

1technophile, 2021-04-24:
>This is somewhat of a known issue, though I have yet to find a reason for it.
For reference: 1technophile/OpenMQTTGateway#794

This is now resolved in OMG with the introduction of a queue for message publishing

h2zero, 2021-04-24:
That looks like an ideal way to handle the issue @1technophile, thanks.