h2zero/NimBLE-Arduino #188: silly question - when can I start a scan?

btsimonh, 2021-02-02:
Hi H2zero,

sorry to bug you again,  this is probably obvious to 'one skilled in the art'.

A problem I have is wanting to hear adverts at the same time as having connected, and doing an 'operation' (read/write, await notify).

At the moment, the scan is automatically stopped when a connection is attempted.
I don't restart scanning until I have disconnected.

Can I restart scanning whilst still connected?  If so, in what phase?

br,

Simon.

(use case - connecting to devices to poll them for info, but still wanting to hear adverts for presence monitoring).

h2zero, 2021-02-02:
You sure can, that's what the NimBLE_Client example does. You can start scanning anytime after the connect call returns.

btsimonh, 2021-02-03:
thankyou :), seems to be working.
At a lower level, could the scan theoretically be started earlier, or not need to stop?
(for distant devices, the connect can take a *long* time to fail)....

h2zero, 2021-02-03:
The connection process uses the scan to find the device so the discovery scan needs to stop while attempting connection.

chegewara, 2021-02-03:
Connection process is using a lot ble radio, so it is not possible to continue scan at the same time.

btsimonh, 2021-02-07:
thanks both, it's fairly clear now :)

closed 2021-02-07 by btsimonh