h2zero/NimBLE-Arduino #191: advert data

btsimonh, 2021-02-07:
Hi,

In the below image:

[image]

Any reason why NimBLE would not report a payload for the device starting E52E?

using the code:
```
  BLEAdvertisedDevice *advertisedDevice = ad->advertisedDevice;

  uint8_t* payload = advertisedDevice->getPayload();
  size_t payloadlen = advertisedDevice->getPayloadLength();
  if (payloadlen  && (maxlen > 30)){ // will truncate if not enough space
    strcpy(p, ",\"p\":\"");
    p += 6;
    maxlen -= 6;
    dump(p, maxlen-10, payload, payloadlen);
    int len = strlen(p);
    p += len;
    maxlen -= len;
    *(p++) = '\"'; maxlen--;
  }

  int svcdataCount = advertisedDevice->getServiceDataCount();
  if (svcdataCount){

```

both payloadlen and svcdataCount appear to be zero?
10:45:07.079 MQT: tele/tasmota_BLE3_98BFAC/BLE = {"details":{"mac":"E52E1C92779F","RSSI":-51}}

br,

Simon

h2zero, 2021-02-07:
I can't think of a reason for this. If you call the toString method on the advertisedDevice does it produce anything?

btsimonh, 2021-02-18:
closing because the guy asking was not me.....

closed 2021-02-18 by btsimonh