h2zero/NimBLE-Arduino #192: Data privacy

DrSnuggles, 2021-02-10:
Hi,

and first of all i have to thank you for this nice library. Without it my PIN secured device had no client to talk to.

I found that the path to the included lib is found in binary files.
This also includes the name of my user profile. Not a big deal for me but i have to inform other programmers in our company about that because of GDPR :(
Same sketch compiled on same machine but with different account have different binaries.
I use ArduinoIDE and compile for ESP32. All other libs i use do not leave such a trace.

Reproduce: Compile client example and open .bin file in editor. Look for C:\Users\

h2zero, 2021-02-10:
Hi @DrSnuggles, glad it's working for your needs. 

Very interesting that the file path is placed in the bin file, I can confirm the same result here. Unfortunately I do not know why this occurs or what can be done about it. I will look into it for sure.

h2zero, 2021-02-10:
Did a quick look at this and I think it occurs because of various assert() calls for stack tracing. I looked at my IDF compiled bin files and the same occurs there (many more actually, from non-ble core files).

DrSnuggles, 2021-02-11:
Thanks for looking into.

Did i just had luck with my other includes?
I couldn't believe that and tried setting core debug level in ArduinoIDE to verbose.
Thats it! Now most/all libs write the path into binary. SPIFFS just to mention an example.
Setting it back to none and only NimBLE was left.

h2zero, 2021-02-11:
That's curious indeed, I'll do some experiments and see what can be done about this.

h2zero, 2021-02-12:
Spent a little time to see what can be done about this and the conclusion is that all of the assert calls need to be changed or removed, including the ones in the NimBLE core files. I'll look more into this to see if it can be done reasonably.