h2zero/NimBLE-Arduino #194: Memory leak when connecting/disconnecting sensors

louismulder, 2021-02-20:
Hi,

I have changed the NimBLE_Client a bit to get the data of a LYWSD03MMC and a Flower care sensor.
The client will be created on connection and deleted after the disconnection.
What I see is that the Free memory Heap is decreasing after each cycle.

See the debug output:
Starting BLE Client
New sensor created:c4:7c:8d:6c:64:d7
Scan Ended
Free Memory=266068
New client created:c4:7c:8d:6c:64:d7
Connected: c4:7c:8d:6c:64:d7
address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=0,quality=44,temperature=20.4,illuminance=7717,moisture=0
address=c4:7c:8d:6c:64:d7, model=Flower care, bat=90 %
Disconnected: c4:7c:8d:6c:64:d7
End Free Memory=265700
New sensor created:Scan Ended
Free Memory=265376
New client created:a4:c1:38:19:1f:f3
a4:c1:38:19:1f:f3
Connected: a4:c1:38:19:1f:f3
address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=0,quality=70,temperature=18.2,humidity=52.0
address=a4:c1:38:19:1f:f3, model=LYWSD03MMC, bat=99 %
Disconnected: a4:c1:38:19:1f:f3
End Free Memory=265288
Scan Ended
Free Memory=264672
New client created:c4:7c:8d:6c:64:d7
Connected: c4:7c:8d:6c:64:d7
address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=90,quality=42,temperature=20.4,illuminance=7798,moisture=0
Disconnected: c4:7c:8d:6c:64:d7
End Free Memory=265212
Scan Ended
Free Memory=264744
New client created:a4:c1:38:19:1f:f3
!! Failed to connect 1, deleted client: a4:c1:38:19:1f:f3
End Free Memory=264640
Free Memory=264684
New client created:a4:c1:38:19:1f:f3
Connected: a4:c1:38:19:1f:f3
address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=99,quality=90,temperature=18.2,humidity=52.0
Disconnected: a4:c1:38:19:1f:f3
End Free Memory=265088
Scan Ended
Free Memory=264508
New client created:c4:7c:8d:6c:64:d7
Connected: c4:7c:8d:6c:64:d7
address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=90,quality=26,temperature=20.5,illuminance=8069,moisture=0
Disconnected: c4:7c:8d:6c:64:d7
End Free Memory=265028

See my code:

/** NimBLE_Server Demo:
 *
 *  Demonstrates many of the available features of the NimBLE client library.
 *
 *  Created: on March 24 2020
 *      Author: H2zero
 *
*/

#include <NimBLEDevice.h>

#define POLL_TIME 60000         // 1 minute
#define RETRY_AFTER_ERROR 10000 // 10 seconds
#define REFRESH_BATTERY_CYCLE   20

typedef void(*ParseSensorData) (uint8_t* pData, size_t length, int sensorID);
[... 528 lines cut ...]
     *  but will use more energy from both devices
     */
    pScan->setActiveScan(true);
    /** Start scanning for advertisers for the scan time specified (in seconds) 0 = forever
     *  Optional callback for when scanning stops.
     */
    pScan->start(scanTime, scanEndedCB);
}

void loop()
{
    /** Loop here until we find a device we want to connect to */
    for (int sensorID = 0; sensorID < sensors.size(); sensorID++)
    {
        // Read the data of a sensor each POLL_TIME
        if (sensors[sensorID].timer == 0 || millis() > sensors[sensorID].timer + POLL_TIME)
        {
            sensors[sensorID].timer = millis();
            NimBLEDevice::getScan()->stop();
            processSensorData(sensorID);
        }
    }
}

regards Louis

h2zero, 2021-02-20:
Hi @louismulder, in an application like this where you are allocating and deallocating a lot, which is what happens when the clients are created and deleted, the idle task needs time to clean up. I see your loop does not call `delay()`, if you added a call to delay for a few millis there it would give the idle task the time it needs. 

If that does not solve it I'd be interested to know if removing the `deleteClient()` calls has any effect? Something may not get deallocated in that operation.

louismulder, 2021-02-21:
Hi @h2zero , 
I have changed the code, to reuse the Client (removed NimBLEDevice::deleteClient(pClient);) and changed the loop.
See code:

static uint32_t highestMemory = 0;
static uint32_t lowestMemory = 500000UL;
void determineMemory(String text, uint32_t memory)
{
    if (memory > highestMemory)
    {
        highestMemory = memory;
        Serial.print(text+ "Highest Free Memory=");
        Serial.print(highestMemory);
        Serial.println();
    }
    if (memory < lowestMemory)
    {
        lowestMemory = memory;
        Serial.print(text+ "Lowest Free Memory=");
        Serial.print(lowestMemory);
        Serial.println();
    }
}

static int loopCnt = 0;
void loop()
{
    /** Loop here until we find a device we want to connect to */
    for (int sensorID = 0; sensorID < sensors.size(); sensorID++)
    {
        // Read the data of a sensor each POLL_TIME
        if (sensors[sensorID].timer == 0 || millis() > sensors[sensorID].timer + POLL_TIME)
        {
            sensors[sensorID].timer = millis();
            NimBLEDevice::getScan()->stop();
            Serial.print("Loop cont= ");
            Serial.println(loopCnt++);
            determineMemory("Start Memory: ",ESP.getFreeHeap());
            processSensorData(sensorID);
            delay(100);
            determineMemory("End Memory: ",ESP.getFreeHeap());
//            Serial.println(ESP.getFreeHeap());
        }
    }
}

The result:
Still leaking memory

See logging:
11:19:47:737 -> Starting BLE Client
11:19:50:270 -> New sensor created:Scan Ended
Loop cont= 0
Start Memory: Highest Free Memory=265876
c4:7c:8d:6c:64:d7
Start Memory: Lowest Free Memory=265876
New client created:c4:7c:8d:6c:64:d7
11:19:51:258 -> Connected: c4:7c:8d:6c:64:d7
11:19:54:425 -> address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=0,quality=26,temperature=19.1,illuminance=7674,moisture=0
11:19:55:958 -> address=c4:7c:8d:6c:64:d7, model=Flower care, bat=95 %
11:19:56:066 -> End Memory: Lowest Free Memory=265040
Disconnected: c4:7c:8d:6c:64:d7
11:20:12:800 -> New sensor created:a4:c1:38:19:1f:f3
Scan Ended
Loop cont= 1
Start Memory: Lowest Free Memory=264196
Reuse client:a4:c1:38:19:1f:f3
11:20:14:447 -> Connected: a4:c1:38:19:1f:f3
11:20:24:398 -> address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=0,quality=68,temperature=17.2,humidity=53.0
11:20:26:147 -> address=a4:c1:38:19:1f:f3, model=LYWSD03MMC, bat=99 %
11:20:26:259 -> End Memory: Lowest Free Memory=264052
11:20:26:366 -> Disconnected: a4:c1:38:19:1f:f3
11:20:50:227 -> Scan Ended
Loop cont= 2
Start Memory: Lowest Free Memory=263948
Reuse client:c4:7c:8d:6c:64:d7
11:20:52:301 -> Connected: c4:7c:8d:6c:64:d7
11:20:55:477 -> address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=95,quality=36,temperature=19.2,illuminance=7594,moisture=0
11:20:56:150 -> Disconnected: c4:7c:8d:6c:64:d7

..............

12:45:24:259 -> Connected: c4:7c:8d:6c:64:d7
Reconnected client:c4:7c:8d:6c:64:d7
12:45:25:259 -> address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=100,quality=16,temperature=20.5,illuminance=6734,moisture=1
12:45:25:368 -> Disconnected: c4:7c:8d:6c:64:d7
12:46:12:181 -> Scan Ended
Loop cont= 178
Start Memory: Lowest Free Memory=262260
12:46:13:399 -> Connected: a4:c1:38:19:1f:f3
Reconnected client:a4:c1:38:19:1f:f3
12:46:18:006 -> address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=99,quality=78,temperature=18.6,humidity=52.0
12:46:18:462 -> End Memory: Lowest Free Memory=262228
Disconnected: a4:c1:38:19:1f:f3
12:46:23:836 -> Scan Ended

h2zero, 2021-02-21:
@louismulder thank you for the update. 

With the changes you've made we can rule out anything being leaked in the delete client code. That leaves me suspecting the scan code. Are you using version 1.2.0? If so, in your scan setup could you try setting maxResults to 0? `pScan->setMaxResults(0);` 

That will still give you the callback but will not store the scanned device data anymore.

louismulder, 2021-02-21:
Hi @h2zero, I have added the setMaxResults(); in the setup() just before the start.
This is the logging:

03:59:25:870 -> Starting BLE Client
03:59:30:792 -> New sensor created:Scan Ended
Loop cont= 0
Start Memory: Highest Free Memory=265992
Start Memory: Lowest Free Memory=265992
a4:c1:38:19:1f:f3
New client created:a4:c1:38:19:1f:f3
03:59:34:292 -> Connected: a4:c1:38:19:1f:f3
03:59:43:151 -> address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=0,quality=78,temperature=20.0,humidity=51.0
03:59:44:902 -> address=a4:c1:38:19:1f:f3, model=LYWSD03MMC, bat=99 %
03:59:45:011 -> End Memory: Lowest Free Memory=265056
03:59:45:122 -> Disconnected: a4:c1:38:19:1f:f3
03:59:47:969 -> New sensor created:c4:7c:8d:6c:64:d7
Scan Ended
Loop cont= 1
Start Memory: Lowest Free Memory=265004
Reuse client:c4:7c:8d:6c:64:d7
03:59:49:064 -> Connected: c4:7c:8d:6c:64:d7
03:59:52:469 -> address=c4:7c:8d:6c:64:d7,model=Flower care,type=1,bat=0,quality=24,temperature=21.9,illuminance=4822,moisture=1
03:59:54:230 -> address=c4:7c:8d:6c:64:d7, model=Flower care, bat=100 %
03:59:54:342 -> End Memory: Lowest Free Memory=264688
03:59:54:449 -> Disconnected: c4:7c:8d:6c:64:d7
04:00:30:809 -> Scan Ended

..............

05:19:13:467 -> Loop cont= 167
05:19:15:144 -> Connected: a4:c1:38:19:1f:f3
Reconnected client:a4:c1:38:19:1f:f3
05:19:17:998 -> address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=99,quality=62,temperature=21.3,humidity=49.0
05:19:18:142 -> Disconnected: a4:c1:38:19:1f:f3
05:19:18:266 -> End Memory: Lowest Free Memory=263644
05:19:27:049 -> Scan Ended
Loop cont= 168
05:19:29:032 -> Reconnect failed: c4:7c:8d:6c:64:d7
05:20:13:563 -> Loop cont= 169
05:20:13:907 -> Connected: a4:c1:38:19:1f:f3
Reconnected client:a4:c1:38:19:1f:f3
05:20:18:406 -> address=a4:c1:38:19:1f:f3,model=LYWSD03MMC,type=0,bat=99,quality=62,temperature=21.3,humidity=49.0
05:20:19:080 -> Disconnected: a4:c1:38:19:1f:f3
05:20:27:061 -> Scan Ended

h2zero, 2021-02-21:
I'm not sure if this is a proper test for a memory leak as your code is only tracking the minimum and maximum free heap. 

Heap will be allocated and deallocated constantly in the BLE stack. The important thing is the current free heap, it should not change much over a long period of time. a few bytes is normal due to vectors never reducing their size and other such structures being allocated etc.

I will try to reproduce this today.

louismulder, 2021-02-22:
Hi @h2zero, I was forgotten to thank you for your great work and support.

Here is the new code which could be used by other users too and a small tutorial to add new sensor types.

`
/** NimBLE_Server Demo:
 *
 *  Demonstrates many of the available features of the NimBLE client library.
 *
 *  Created: on March 24 2020
 *      Author: H2zero
 *
*/

#include <NimBLEDevice.h>

#define POLL_TIME 60000         // 1 minute
#define RETRY_AFTER_ERROR 10000 // 10 seconds
#define REFRESH_BATTERY_CYCLE   20

typedef void(*ParseSensorData) (uint8_t* pData, size_t length, int sensorID);
typedef void(*ReadSensorData) (NimBLEClient* pClient, int sensorID);

NimBLERemoteService* getService(NimBLEClient* pClient, NimBLEUUID serviceUUID);
bool writeData(NimBLERemoteService* pService, NimBLEUUID uuid, uint8_t* data, int length);
void readData(NimBLERemoteService* pService, NimBLEUUID uuid, int sensorID, boolean battery);
void scanEndedCB(NimBLEScanResults results);

// Scan variables
static NimBLEScan* pScan;
static uint32_t scanTime = 0; /** 0 = scan forever */

static boolean notified = false;

static boolean doNotReuseClient = false;

// The connected sensors
struct sensor_t {
    std::string address;
    uint8_t sensorType;
    uint8_t quality;
    uint8_t battery;
    int batteryRefresh;
    long timer;
    boolean waitForNotify;
};
static std::vector<sensor_t> sensors{};

/***************************** Sensors ***********************************/
/**
* To add a new Sensor do the following:
* 1. Add the name to search for (in the AdvertisedDevice) in the array sensorNames[]
* 2. Create a read_Type(NimBLEClient* pClient, int sensorID) method
* 3. Add the read_Type to the array readSensorData[]
* 4. Create a parse_Type(NimBLEClient* pClient, int sensorID) method
* 5. Add the parse_Type to the array parseSensorData[]
*/

// Array of device names found in AdvertisedDevices
char* sensorNames[] = {
    "LYWSD03MMC",
    "Flower care"
};

/******************************* methods *********************************************/
/***************************** LYWSD03MMC ********************************************/
void read_LYWSD03MMC(NimBLEClient* pClient, int sensorID) {
    // Connect Data service
    NimBLERemoteService* pService = nullptr;
    pService = getService(pClient, NimBLEUUID("ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"));
[... 467 lines cut ...]
        Serial.print(lowestMemory);
        Serial.println();
    }
}

static int loopCnt = 0;
void loop() {
    /** Loop here until we find a device we want to connect to */
    for (int sensorID = 0; sensorID < sensors.size(); sensorID++) {
        // Read the data of a sensor each POLL_TIME
        if (sensors[sensorID].timer == 0 || millis() > sensors[sensorID].timer + POLL_TIME) {
            sensors[sensorID].timer = millis();
            NimBLEDevice::getScan()->stop();
            Serial.print("Loop count= ");
            Serial.println(loopCnt++);
            determineMemory("Start Memory: ",ESP.getFreeHeap());
            processSensorData(sensorID);
            delay(100);
            determineMemory("End Memory: ",ESP.getFreeHeap());
            Serial.print("Free memory: ");
            Serial.println(ESP.getFreeHeap());
        }
    }
}
`

h2zero, 2021-02-24:
Thanks for that, sorry I haven't been able to test this as I've been quite busy as of late. 

If you could post the code inside code blocks it would be much easier to read (use 3 backticks at the beginning and the end for code with indentation).

louismulder, 2021-02-24:
`
/** NimBLE_Server Demo:
 *
 *  Demonstrates many of the available features of the NimBLE client library.
 *
 *  Created: on March 24 2020
 *      Author: H2zero
 *
*/

#include <NimBLEDevice.h>

#define POLL_TIME 60000         // 1 minute
#define RETRY_AFTER_ERROR 10000 // 10 seconds
#define REFRESH_BATTERY_CYCLE   20

typedef void(*ParseSensorData) (uint8_t* pData, size_t length, int sensorID);
typedef void(*ReadSensorData) (NimBLEClient* pClient, int sensorID);

NimBLERemoteService* getService(NimBLEClient* pClient, NimBLEUUID serviceUUID);
bool writeData(NimBLERemoteService* pService, NimBLEUUID uuid, uint8_t* data, int length);
void readData(NimBLERemoteService* pService, NimBLEUUID uuid, int sensorID, boolean battery);
void scanEndedCB(NimBLEScanResults results);

// Scan variables
static NimBLEScan* pScan;
static uint32_t scanTime = 0; /** 0 = scan forever */

static boolean notified = false;

static boolean doNotReuseClient = false;

// The connected sensors
struct sensor_t {
    std::string address;
    uint8_t sensorType;
    uint8_t quality;
    uint8_t battery;
    int batteryRefresh;
    long timer;
    boolean waitForNotify;
};
static std::vector<sensor_t> sensors{};

/***************************** Sensors ***********************************/
/**
* To add a new Sensor do the following:
* 1. Add the name to search for (in the AdvertisedDevice) in the array sensorNames[]
* 2. Create a read_Type(NimBLEClient* pClient, int sensorID) method
* 3. Add the read_Type to the array readSensorData[]
* 4. Create a parse_Type(NimBLEClient* pClient, int sensorID) method
* 5. Add the parse_Type to the array parseSensorData[]
*/

// Array of device names found in AdvertisedDevices
char* sensorNames[] = {
    "LYWSD03MMC",
    "Flower care"
};

/******************************* methods *********************************************/
/***************************** LYWSD03MMC ********************************************/
void read_LYWSD03MMC(NimBLEClient* pClient, int sensorID) {
    // Connect Data service
    NimBLERemoteService* pService = nullptr;
    pService = getService(pClient, NimBLEUUID("ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"));
    if (pService) {
        notified = false;
        NimBLERemoteCharacteristic* pChr = pService->getCharacteristic(NimBLEUUID("ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6"));
        sensors[sensorID].waitForNotify = true;
[... 463 lines cut ...]
        Serial.print(lowestMemory);
        Serial.println();
    }
}

static int loopCnt = 0;
void loop() {
    /** Loop here until we find a device we want to connect to */
    for (int sensorID = 0; sensorID < sensors.size(); sensorID++) {
        // Read the data of a sensor each POLL_TIME
        if (sensors[sensorID].timer == 0 || millis() > sensors[sensorID].timer + POLL_TIME) {
            sensors[sensorID].timer = millis();
            NimBLEDevice::getScan()->stop();
            Serial.print("Loop count= ");
            Serial.println(loopCnt++);
            determineMemory("Start Memory: ",ESP.getFreeHeap());
            processSensorData(sensorID);
            delay(100);
            determineMemory("End Memory: ",ESP.getFreeHeap());
            Serial.print("Free memory: ");
            Serial.println(ESP.getFreeHeap());
        }
    }
}
`

louismulder, 2021-02-24:
Hi @h2zero,

I have tried to insert the code with the Insert Code tags, but the result is still the same.
If you want I can also mail the sketch as attachment. Pls let me know.
regards Louis

h2zero, 2021-02-25:
The code tag in comment doesn't work for multiple lines. For reference in the future you need to do this:

````
```
code 
        code
               code
               }       
        }
}
```
````

Try editing your previous post with those back ticks in place above and below. You can also post the file if you like.

louismulder, 2021-02-26:
[BLE.zip](https://github.com/h2zero/NimBLE-Arduino/files/6049428/BLE.zip)
Hi @h2zero ,

I have uploaded the sketch in the zip file.

h2zero, 2021-02-28:
@louismulder Sorry for the delay, lots of things going on. 

I have tested your code, which is very well done BTW,  I do not see any memory leaks with the simulated devices I used. As stated previously, the use of min/max measuring of the free heap doesn't necessarily indicate a leak. The nature of this library and the RTOS is a fluctuating heap space, the average heap would be a more appropriate measurement. That said I will do some focused testing to be sure.

louismulder, 2021-02-28:
Hi @h2zero , thanks very much for your support and great library.
You may close this issue.

h2zero, 2021-02-28:
Thanks @louismulder, always good to check on these things and sharing the code is very helpful. I ran some tests overnight scanning and connecting/disconnecting and the logs show similar to yours in the min/max values. However, reading the current heap state now vs when it started the free heap is within 200 bytes of the beginning state, which could be the result of containers expanding and not contracting which is normal for c++ containers.

h2zero, 2022-04-05:
Closing a likely resolved in the latest releases.

closed 2022-04-05 by h2zero