h2zero/NimBLE-Arduino #195: iPhone doesn't detect ESP32 as a beacon

ReyWamba, 2021-02-20:
First of all, I want to congratulate you for your work. The amount of memory that is freed is incredible.

But my iPhone doesn't detect my ESP32 as a beacon with your library.

This code works for me with the original ESP32 library, but not with yours:

'void initBLEbeacon() {
  BLEAdvertising* pAdvertising;
  pAdvertising = pServer->getAdvertising();
  pAdvertising->stop();
  // iBeacon
  BLEBeacon myBeacon;
  myBeacon.setManufacturerId(0x4c00);
  myBeacon.setMajor(deviceInfo.major);
  myBeacon.setMinor(deviceInfo.minor);
  myBeacon.setSignalPower(0xc5);
  myBeacon.setProximityUUID(BLEUUID(BEACON_UUID_REV));

  BLEAdvertisementData advertisementData;
  advertisementData.setFlags(0x06);
  advertisementData.setManufacturerData(myBeacon.getData());

  pAdvertising->setAdvertisementData(advertisementData);

  pAdvertising->start();

  esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
  Serial.println("Inicializado beacons");
}`

Is there something i am doing wrong?

Thanks.

h2zero, 2021-02-20:
Hi @ReyWamba, thank you. I don't see anything wrong in the code. Would you be able to share a debug log?

ReyWamba, 2021-02-20:
Hello @h2zero:

I fixed the error. The problem is that the original BLEUUID () function from the ESP32 library returned the UUID string in reverse order, while yours returns it in the correct order.

That is why the iPhone was not detecting the beacon (it must "know" what UUID to look for).

Thank you very much for your speed in answering, and above all, thank you very much for your great work.

h2zero, 2021-02-20:
Sorry for the confusion with that, the NimBLE core handles that data with opposite endian from bluedroid. 

Glad you found the issue!

ReyWamba, 2021-02-21:
Hello again:

In my project, I use the ESP32 simultaneously as a BLE device with a service and as a Beacon. With the original ESP32 library, this was possible (as long as you start the Beacon first and then the service).

However, with your library, activating the Beacon disappears the service. The Beacon is detected and the BLE device is still visible but with zero services. And this happens by starting the service first and then the Beacon or vice versa.

I attach the code for both initialization routines:

`void initBLEservice() {
  BLEAdvertising* pAdvertising;
  pAdvertising = pServer->getAdvertising();
  pAdvertising->stop();

  // Create the BLE Service
  BLEService *pService = pServer->createService(NimBLEUUID(SERVICE_UUID));

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      TO_APP_UUID,
                      NIMBLE_PROPERTY::READ   |
                      NIMBLE_PROPERTY::WRITE  |
                      NIMBLE_PROPERTY::NOTIFY
                    );
  pCharacteristic->setCallbacks(new ToAppCallbacks());
 // pCharacteristic->addDescriptor(new BLE2902());

  pAdvertising->addServiceUUID(NimBLEUUID(SERVICE_UUID));

  // Start the service
  pService->start();

  pAdvertising->start();

  Serial.println("Inicializado service");
}

void initBLEbeacon() {
  BLEAdvertising* pAdvertising;
  pAdvertising = pServer->getAdvertising();
  pAdvertising->stop();
  // iBeacon
  BLEBeacon myBeacon;
  myBeacon.setManufacturerId(0x4c00);
  myBeacon.setMajor(deviceInfo.major);
  myBeacon.setMinor(deviceInfo.minor);
  myBeacon.setSignalPower(0xc5);
  myBeacon.setProximityUUID(BLEUUID(BEACON_UUID_REV));

  BLEAdvertisementData advertisementData;
  advertisementData.setFlags(0x06);
  advertisementData.setManufacturerData(myBeacon.getData());

  pAdvertising->setAdvertisementData(advertisementData);

  pAdvertising->start();

  Serial.println("Inicializado beacons");
}
`

Thanks in advance.

h2zero, 2021-02-21:
Hi @ReyWamba, I'm not sure how this worked in the original library but I suspect it moved the service advertisement to the scan response automatically. That has not been implemented here currently but the solution is simple, just create another instance of BLEAdvertisementData and add your service to it.

```
BLEAdvertisementData responseData;
responseData.setCompleteServices(UUID); or responseData.setPartialServices(UUID);
pAdvertising->setScanResponseData(responseData);
```

Also have a look at this issue: https://github.com/h2zero/NimBLE-Arduino/issues/146 for more info as well

wakwak-koba, 2021-02-21:
I can't test because I don't have an iPhone, recent iOS now require `****_ENC`, at least for using BLE-HID Devices.
https://github.com/h2zero/NimBLE-Arduino/pull/182

> // Create a BLE Characteristic
> pCharacteristic = pService->createCharacteristic(
> TO_APP_UUID,
> NIMBLE_PROPERTY::READ
> NIMBLE_PROPERTY::WRITE
> NIMBLE_PROPERTY::NOTIFY
> );

Could you please try to fix the above part as below and see if the condition improves?

> // Create a BLE Characteristic
> pCharacteristic = pService->createCharacteristic(
> TO_APP_UUID,
> NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::READ_ENC
> NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::WRITE_END
> NIMBLE_PROPERTY::NOTIFY
> );

In the original ESP32 library, there is no explicit ****_ENC for createCharacteristic. 

https://github.com/espressif/arduino-esp32/blob/4b3f5c8ed463b18afe28d132d607bbf164d8eec9/libraries/BLE/src/BLECharacteristic.cpp#L40-L52

https://github.com/espressif/arduino-esp32/blob/4b3f5c8ed463b18afe28d132d607bbf164d8eec9/libraries/BLE/src/BLECharacteristic.cpp#L633-L640

I haven't investigated the internal behavior, but I think `READ_ENC` is also specified when `READ` is specified.

ReyWamba, 2021-02-28:
Hello zero

sorry for my delay in responding. Work keeps me very busy.

I have tried with the following code:

`void initBLEservice() {
  BLEAdvertising* pAdvertising;
  pAdvertising = pServer->getAdvertising();
  pAdvertising->stop();

  // Create the BLE Service
  BLEService *pService = pServer->createService(NimBLEUUID(SERVICE_UUID));

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      TO_APP_UUID,
                      NIMBLE_PROPERTY::READ   |
                      NIMBLE_PROPERTY::WRITE  |
                      NIMBLE_PROPERTY::NOTIFY
                    );
  pCharacteristic->setCallbacks(new ToAppCallbacks());
 // pCharacteristic->addDescriptor(new BLE2902());

  BLEAdvertisementData responseData;
  responseData.setCompleteServices(NimBLEUUID(SERVICE_UUID)); 
  //responseData.setPartialServices(NimBLEUUID(SERVICE_UUID));
  pAdvertising->setScanResponseData(responseData);
  pAdvertising->addServiceUUID(NimBLEUUID(SERVICE_UUID));

  
  // Start the service
  pService->start();

  pAdvertising->start();

  Serial.println("Inicializado service");
}`

Calling initBLEservice first and then initBeacon, the beacons work, but the app still can't detect ESP32 with the 
'centralManager.scanForPeripherals (withServices: [serviceUUID], options: nil)'  function.

Any suggestion?

Thank you very much.

h2zero, 2021-02-28:
I'm not sure how the app code works, might only be looking in advertisement data and not scan response or possibly it is not active scanning. Can you see the esp if you use an app like nRFConnect?

ReyWamba, 2021-03-01:
I have managed to get my app to locate the esp32 and connect to it (by adding a name to BLEAdvertingsementData - responseData.setName ("myName") - and making the app instead of searching for BLEs with services - 'centralManager.scanForPeripherals (withServices: [ serviceUUID], options: nil) '- find all BLEs - 'centralManager (_ central: CBCentralManager, didDiscover peripheral: CBPeripheral, advertisementData: [String: Any], rssi RSSI: NSNumber)' - and filter them by "kCBAdvDataLocalName".

Thanks a lot.

closed 2021-03-01 by ReyWamba