h2zero/NimBLE-Arduino #196: Confusion around use of radio

codewitch-honey-crisis, 2021-02-21:
Sorry, this is more of a request for clarification than an open issue, but I don't know where to post it.

I'm not clear from the documentation when the radio for my ESP32 is engaged.

I remember there being functions somewhere in the NimBLE to turn the radio on and off, but I don't know in which circumstances they are safe to use.

Assuming I am publishing a server, how do I turn the radio on and off safely such that I can either restart the service and characteristics I'm running, or keep them around otherwise until the radio is back on?

I couldn't find an example that had this, at least anywhere obvious, but my thing is battery powered, and despite using BLE, it simulates a connection and pairing paradigm so it doesn't need the radio all the time. When it's not connected i want the radio to be off. when you press the connect button i want the radio and services to be alive again.

Is this even necessary with BLE? Maybe this is all for nothing?

h2zero, 2021-02-21:
Hi @codewitch-honey-crisis, generally the radio is only in use when there is an active connection and or advertising/scanning. If you are not connected to any peers and not advertising or scanning for connections then the radio is not in use.

For your use case as a server you could simply stop advertising/disconnect clients then sleep and wakeup/start advertising on button press (there is an adverting time limit available so you could minimize radio time). Also keep in mind that during connections the radio is only used in bursts with an interval that is determined by the connected client and the amount of time in use is based on how much data needs to be transferred.

Basically as far as the "is this necessarry?" question is concerned, the answer is "it depends" 😄 . If you just need to connect or advertise for a short time every 5-10 minutes the answer is different than if you need to share data every 30 seconds.

codewitch-honey-crisis, 2021-02-21:
Thank you so much! That's really helpful. When my client is connected to the server, the server stops advertising. I don't care about battery on the client as much - worst case it's a phone. Basically I'll probably spend long periods "connected", but not advertising - sending 100bytes/sec of data for usually maybe 1 a minute or 2 but up to 10 minutes at a time, and then idling, but still connected, until somebody (the application or the device) disconnects. 

However, the device can be operated autonomously, and I want the BLE radio to only be active if the user requests it by pushing a "connect" button i have on a device, at which point it starts advertising, and continues to until it finds a connection or a (currently) 15 second timeout occurs - i may change that timeout.

So whenever it's disconnected and not advertising, I want the radio off. It sounds like that's already the case, which answers my second question.

So all I have to do is not advertise, not scan, and if i'm unconnected there's no radio in use. That's awesome. Of course if I'm wrong in how i interpreted this, please let me know. Otherwise, have a great rest of your weekend, and thanks again.

closed 2021-02-21 by codewitch-honey-crisis