h2zero/NimBLE-Arduino #197: Clarification - lifetime of characteristic callbacks

codewitch-honey-crisis, 2021-02-24:
I'm trying to understand the difference in the way callbacks are set and cleaned up by the library.

```
m_pbleServer = NimBLEDevice::createServer();
m_pbleServer->setCallbacks(new BleServerCallbacks(), true);
m_pbleService = pServer->createService(g_svcid);
NimBLECharacteristic *pSessionCharacteristic = g_sessionService->createCharacteristic(
    g_sescharid,
    //          NIMBLE_PROPERTY::READ |
    NIMBLE_PROPERTY::WRITE |
    NIMBLE_PROPERTY::NOTIFY);
m_pbleCharacteristicCallbacks=new BleCharacteristicCallbacks(&m_cmd);
pSessionCharacteristic->setCallbacks(m_pbleCharacteristicCallbacks);
```
See the two setCallback() calls?

One appears to allow me to create a new instance of a callback object, which *the library* will delete on destruction.
The other one appears to require me to handle cleanup myself. There is no parameter to delete the callback object.

I don't understand the discrepancy, and it wouldn't concern me except that now I'm not sure if I'm handling this correctly.

The code above runs - not in a constructor, but in a method that gets called exactly once and almost immediately after construction.

In the destructor of this class, I am calling delete on m_pbleCharacteristicCallbacks if it's non-null.

Would you say that seems like the right thing to do? I'm not exactly clear on where I should be deleting the characteristic callbacks. When is it safe to do so, exactly?

h2zero, 2021-02-24:
>Would you say that seems like the right thing to do? I'm not exactly clear on where I should be deleting the characteristic callbacks. When is it safe to do so, exactly?

You have this correct. You can delete the callback function any time after the characteristic instance has been deleted or after calling it's setCallbacks method with a nullptr. 

I understand the confusion and the addition of a cleanup parameter should be added to the characteristic callback setting. This was not considered when added due to the nature of server characteristics not generally changing once setup.

h2zero, 2022-04-05:
Closing as stale, please reopen if needed.

closed 2022-04-05 by h2zero