h2zero/NimBLE-Arduino #198: Message size vs MTU vs configuration

jshep321, 2021-02-24:
Hi,
I'm trying to send a "long" BLE message to an iOS app and I think I'm failing in my nimble configuration (ESP32).
The iPhone docs state that the MTU max size is 185 bytes.   

1) Is there a way to automatically chunk > 185byte messages and reassemble at the destination?  I've seen some howtos on doing this for e.g. firmware updates, but not explicitly for nimble (so I am assuming default bluedroid stack).
2) Does the setMTU call actually force a change of this?   I set this before and after server start just to be sure.  I then print the MTU of the connection as part of my onConnect callback and it always reads 23.
Server start:
```
  BLEDevice::init(BLEDEVICESTRING);
  BLEDevice::setPower(ESP_PWR_LVL_P7);
  BLEDevice::setMTU(184);   //max of iphone = 185.  max of nimble:  BLE_ATT_MTU_MAX = 527 (in NimBLEDevice.cpp)
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEDevice::setMTU(184);   //max of iphone = 185.  max of nimble: BLE_ATT_MTU_MAX = 527 (in NimBLEDevice.cpp)

```
onConnect callback:
```
   void onConnect(BLEServer* pServer, ble_gap_conn_desc* desc) {
      deviceConnected = true;
      Serial.println("Bluetooth Device Connected.");
      int bleClientMTU = pServer->getPeerMTU(desc->conn_handle); //this shows 23 regardless...
      Serial.print("MTU of client: "); Serial.print(bleClientMTU);
    }
```

The test results are not expected.  I can send >> 23byte payload, but things are not predictable at longer messages.  I've tried sending the .c_str() as well as a hard-limited char array.

Am I missing something obvious?

Thanks!

jshep321, 2021-02-24:
Follow-up:  I can send exactly 181 byte payload with no issue.  182 crashes the iOS app.  Close to 185, but there is perhaps unknown overhead. Gremlins abound.

h2zero, 2021-02-24:
Hi @jshep321, the MTU of 23 you are seeing is due to when that callback is invoked, which is before the MTU exchange happens. There is a 3 byte overhead on ble data for the headers.

jshep321, 2021-02-25:
Thanks @h2zero!  That 3 bytes header explains it.  

Does the server stack have any built in way to handle chunking / reassembling oversized messages?  I could see this as being a very handy feature for my application (json messages), as well as many other payload types.  I guess complexity arises due to the client requiring matching functionality and it having an unknown stack.  Seems like a useful feature to be added to the standard.

jshep321, 2021-02-26:
Thanks.   Discovered this is an iPhone 6s (and earlier) limitation--max MTU of 185.  Documentation from Apple on this is quite poor.  I assume it's actually a Broadcom chipset firmware/HW limitation.

closed 2021-02-26 by jshep321

jshep321, 2021-02-26:
@h2zero 
One more thing:  the max MTU is 527 from NimBLEServer.cpp, but the characteristic provides an error if the message length is >512.  

Is there a purpose in allowing MTU > 515 (512 + 3 overhead bytes)?

h2zero, 2021-02-26:
Max bytes in a characteristic is 512 as per spec. Then we need 3 extra MTU bytes for write overhead, or 5 extra for long write overhead, or 15 extra for signed write overhead (which we can't do yet).

joseitor, 2023-05-19:
I can't sent more than 20 bytes by notification. Can you share your code? Even using NimBLEDevice::setMTU(517), nothing change.

h2zero, 2023-05-19:
Both devices need to have a higher MTU to notify more data.

h2zero, 2023-05-22:
The peer probably does not have a higher MTU capability than the default.