h2zero/NimBLE-Arduino #199: Stopping advertising fails with 524 error if stopped immediately after connection

codewitch-honey-crisis, 2021-02-24:
I probably just need a clarification, but it could be an issue or a feature request, depending:

It appears that I can run the following code successfully if I'm not currently in the middle of being connected:
```
NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
pAdvertising->stop();
```

However, if I try the same thing immediately after connection (but not in the onConnect() callback - after it) it fails with a 524 error code.

Here's my problem:

I need to stop advertising as soon as I end up connected.

How might I accomplish this?

chegewara, 2021-02-24:
If you are being in connecting state then advertising should be already stopped. Try to call stop twice in "normal state" and see if you get the same error.

codewitch-honey-crisis, 2021-02-24:
Well then something else is going on because after I connect, it still advertises. I know this because it's still showing up in client application discovery after connect.

chegewara, 2021-02-24:
I dont know nimble library, maybe advertising is started right after connection established, but i am for 90% sure advertising should be stopped during connection. It is low level api requirements.

h2zero, 2021-02-24:
Advertising does not start automatically after connection unless the connection failed, do you start in the the onConnect callback?

524 error == command disallowed, usually that would mean advertising is not active.

codewitch-honey-crisis, 2021-02-24:
Something came up and I just got time to investigate more. I'm looking into it. It's getting connected, but it's possible that it isn't getting disconnecting like I thought I told it to.

codewitch-honey-crisis, 2021-02-25:
Okay, as best I can figure out, this is what's happening.

1. I'm setting up the server, service and characteristic

2. I push the "connect button" and start advertising, with no timeout ... ->start()
   - (I have my own timeout, in milliseconds that also syncs with an LED that flashes when advertising.)

3. I connect to the application. 
4. The advertising doesn't meet my internal timeout yet, but it connects, so while it's connected i am not advertising (as you said it wouldn't)
5. I close the PC's application in order to disconnect the server and client
6. I run the PC's application again, and even though i have connected and disconnected since i started the first time, it's still advertising whenever i disconnect.

So like, if i don't ->start() with a timeout, it looks like whenever i disconnect again advertising is started once more automatically. I would have thought that once advertising was stopped by connection, you'd have to start it again yourself. In fact, in the sample code I seem to remember that? Like the NimBLE_Server demo.

I must be wrong? Maybe you could weigh in here. Is this what it's supposed to do? It's still possible something's happening in my code that I'm not aware of. It's complicated at this point. And I have to instrument for logging incrementally or i get too much serial spew

h2zero, 2021-02-25:
That makes more sense, when the client disconnects advertising is automatically started. This can be turned off in your server setup by calling 'pServer->advertiseOnDisconnect(false);`

codewitch-honey-crisis, 2021-02-25:
Oh super! Thanks so much, once again! I'm closing this.

closed 2021-02-25 by codewitch-honey-crisis