h2zero/NimBLE-Arduino #200: Crash after write to service characteristic

codewitch-honey-crisis, 2021-02-25:
I'm getting a crash on core 0 (the system core, not the one my code is running on) shortly after my other application writes to the NimBLE characteristic I've exposed.

This remains true even if I do not set any callback for the service characteristic. It does not happen unless I tell the other side to do a write to this characteristic.

If I *do* set a callback, usually the code in the callback will finish running before it crashes. Occasionally it will not.

I can't debug on this one because I can't use a WROVER for this because I need the pins, and for some reason Espressif never released a properly debuggable WROOM to my knowledge.

This is a long shot, but what things could I be doing that would be causing a crash after the other app writes?

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d1d0e  PS      : 0x00060f30  A0      : 0x800d4929  A1      : 0x3ffc9ba0
A2      : 0x3ffca008  A3      : 0x3ffc9fc8  A4      : 0x3ffc9c84  A5      : 0x3ffc187c  
A6      : 0x00000213  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000430  A12     : 0xfeb70988  A13     : 0xfeb70988  
A14     : 0x3ffc2af4  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9
```

h2zero, 2021-02-25:
Hard to say without a decided backtrace but that core dump looks very much like something was trying to write to a nullptr. 

The callbacks execute on core 0 if the originating source is from the BLE stack.

codewitch-honey-crisis, 2021-02-25:
Thanks anyway. I figured you probably wouldn't be able to help here, but I am running out of ideas so I needed to ask at this point. I found when I remove *both* callbacks it works even though the server callbacks simply log - maybe logging from core 0 is doing it or something. I'm going to close this for now, based on the fact that when both callbacks are removed it's not crashing. I'm not sure why it's my callbacks, because they hardly do anything, but apparently it's my callbacks.

closed 2021-02-25 by codewitch-honey-crisis