h2zero/NimBLE-Arduino #201: NimBLEDevice::getClientByID(ClientID)->isConnected() fails on a recently disconnected server

doudar, 2021-02-28:
I'm having an issue that I think may be a bug. 

I'm using NimBLEDevice::getClientByID(myAdvertisedDevice.connectedClientID)->isConnected() to test if a server is connected before reading it's value, and if this is tested within a second of a client disconnecting, I get an exception. This is in my CPS_Factory branch of https://github.com/doudar/SmartSpin2k

Decoding stack results
0x4008f4dc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008f759: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x401058ef: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400e1b69: NimBLEDevice::getClientByID(unsigned short) at .pio\libdeps\esp32doit\NimBLE-Arduino\src\NimBLEDevice.cpp line 234
0x400d5516: bleClientTask(void*) at src/BLE_Client.cpp line 76
0x400906f2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

h2zero, 2021-02-28:
`if (NimBLEDevice::getClientByID(myAdvertisedDevice.connectedClientID)->isConnected());`

This is hazardous code due to nullptr being a possible result.

However that does not explain why the ID did not resolve to a proper client instance, are you saving a previous ID number and passing it to this call? I didn't look too far into the code yet but the instance ID number can change on incremental connection.

doudar, 2021-02-28:
Yes, I'm saving the connection number and peer address on any device connection or reconnect. 

The nullptr is why I was testing if (NimBLEDevice::getClientByID(myAdvertisedDevice.connectedClientID)); previously, but that seems to fail also.

Without hooking up the jtag, my suspicion is that it's got something to do with the disconnect callback happening simultaneously during this check. 

Maybe I'm over analyzing my issue also - what's your preferred method for getting NimBLERemoteCharacteristic.getData() for your previously subscribed characteristics? It would be kinda slick if there was a device{key:value} struct of all subscribed characteristics.

h2zero, 2021-02-28:
With some of the recent changes the getClientByID method has become less useful in this situation (also there is a random assert there that I need to remove). What I do is just use the address of the peer device to find the client instance that already has the database for the device and reconnect it (using false as the parameter). 

>It would be kinda slick if there was a device{key:value} struct of all subscribed characteristics.

This could be implemented just by looping through the client list and getting the vector for each service/characteristic and testing for the UUID.

doudar, 2021-03-01:
I found that my issue was checking the client id (NimBLEDevice::getClientByID) after a disconnect which returns (what looks like) an in initialized value (int 65526). I assumed that the client id would follow the client until it was deleted. Thanks!

h2zero, 2021-03-01:
Yes that would definitely be the issue. That is the macro value BLE_HS_CONN_HANDLE_NONE, since a connection handle of 0 is valid it was made to be 0xffff.

closed 2021-03-01 by doudar