h2zero/NimBLE-Arduino #202: NimBLECharacteristic-> setValue only writes 4 bytes and -> notify doesn't send the data

ReyWamba, 2021-03-02:
When I try to send data from ESP32 to my app, NimBLECharacteristic-> setValue only writes 4 bytes of the string, and the app does not receive the data. However, the ESP32 does receive the data perfectly from the app through the same characteristic.

The function is this:

```
void sendJSONByBLE (JsonObject& json){
  Serial.println();
  Serial.println("sendJSONByBLE");
  String jsonString ="0123456789";
//  json.printTo(jsonString);
  const char* data = jsonString.c_str();

  PrintArg(data);
  pCharacteristic->setValue(data);
  pCharacteristic->notify();
}

```

and the debug info is this:

```
sendJSONByBLE

data = 0123456789
D NimBLECharacteristic: ">> setValue: length = 4, data = 589dfe3f, characteristic UUID = b489981e-a163-394e-98cd-f497368be3f9"
D NimBLECharacteristic: "<< setValue"
D NimBLECharacteristic: ">> notify: length: 4"
D NimBLECharacteristicCallbacks: "onNotify: default"
D NimBLEServer: ">> handleGapEvent:"
D NimBLECharacteristicCallbacks: "onStatus: default"
D NimBLECharacteristic: "<< notify"
exits sendJsonToApp
D NimBLECharacteristicCallbacks: "onWrite: default"
```

The function declaring the characteristic is:

```
void initBLEservice() {
  BLEAdvertising* pAdvertising;
  pAdvertising = pServer->getAdvertising();
  pAdvertising->stop();

  // Create the BLE Service
  BLEService *pService = pServer->createService(NimBLEUUID(SERVICE_UUID));

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      TO_APP_UUID,
                      NIMBLE_PROPERTY::READ   |
                      NIMBLE_PROPERTY::WRITE  |
                      NIMBLE_PROPERTY::NOTIFY
                    );
  pCharacteristic->setCallbacks(new ToAppCallbacks());
 // pCharacteristic->addDescriptor(new BLE2902());

  BLEAdvertisementData responseData;
  responseData.setName("myName");
  responseData.setCompleteServices(NimBLEUUID(SERVICE_UUID)); 
  //responseData.setPartialServices(NimBLEUUID(SERVICE_UUID));
  pAdvertising->setScanResponseData(responseData);
  pAdvertising->addServiceUUID(NimBLEUUID(SERVICE_UUID));

  // Start the service
  pService->start();

  pAdvertising->start();

  Serial.println("Inicializado service");
}

class ToAppCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string value = pCharacteristic->getValue();
      if (value.length() > 0) {
        StaticJsonBuffer <256> jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char *)&value[0]);
        json["by"] = (uint8_t)byBLE;
        packJSON(json);
      }
    }
};
```

h2zero, 2021-03-02:
```
String jsonString ="0123456789";
const char* data = jsonString.c_str();
const char* data = jsonString.c_str();

  PrintArg(data);
  pCharacteristic->setValue(data);
```
You have 2 options to fix the issue here. 
-1 you can change `String jsonString` to `std::string jsonString` and pass that to setValue directly.
-2 you could include the length parameter in the call to  `pCharacteristic->setValue(data);` for instance :`pCharacteristic->setValue(data, jsonString.length());`

btsimonh, 2021-03-02:
Hi,
somehow you are sending the address of the cstr, not the content.
use 
setValue(const uint8_t *data, size_t size)
or 
setValue ( const std::string & value )

personally, I go with the first.....

s
:) h2zero woke up and got there first!

ReyWamba, 2021-03-02:
Thanks!!

My code is directly migrated from the original BLE library, and -> setValue just worked with (const char * data) as the only parameter. I had not queried the API.

Finally the code that works for me is:

```
  const char* data = jsonString.c_str();
  pCharacteristic->setValue((unsigned char *)data, jsonString.length());

```
without casting from (char *) to (unsigned char *) gives me compilation error.

closed 2021-03-02 by ReyWamba

h2zero, 2021-03-03:
> My code is directly migrated from the original BLE library, and -> setValue just worked with (const char * data) as the only parameter. I had not queried the API.

That's quite odd, I'll have to look at the arduino api. I ported this from the snippets library more-so than from arduino. Possibly something to add or at least note. Thanks for the info, glad it's working for you!

ReyWamba, 2021-03-03:
Thanks to you for your great work.

nicolosponziello, 2021-08-20:
Hi, I am experiencing the same behaviour with the latest version. 
My code:
```
String json = "json string";
characteristic->setValue((unsigned char*) json.c_str(), json.length());
```

But when I read the characteristic on the phone I only get an array of 4 bytes. Any idea?

btsimonh, 2021-08-20:
just an idea...  try (const uint_8 *) instead of (unsigned char *).
It may be a matter of the way the compiler is choosing which setValue() function to use from the class, so the lack of const may cause it to choose the wrong one?
(also check the .h file to see EXACTLY how the function is defined, and match it)

nicolosponziello, 2021-08-20:
I tried also converting the String to std::string but always the same behaviour

btsimonh, 2021-08-20:
stick a debug output in the function in the library - the one you hope is being called - to see if it is.... it's got to be calling the wrong function...

sivar2311, 2021-08-20:
You can use a `std::string` directly

```C++
  std::string json = "json string";
  characteristic->setValue(json);
```

or alternatively convert your `String` to a `std::string`

```C++
  String json = "json string";
  characteristic->setValue(std::string(json.c_str()));
```

nicolosponziello, 2021-08-23:
I solved by passing
```
setValue(*(json.c_str()));
```

nicolosponziello, 2021-08-23:
Apparently, I solved with:
```
String json = String("value");
characteristic->setValue(*(json.c_str()));
```